evcc-io/evcc #2233: Charging Restart mit Heidelberg Energy Control und Dacia Spring

chrissooo, 2022-01-07:
**Describe the bug**

Da ich aktuell noch keine PC haben, haeb ein Grid mit folegnden EInstellungen gefaked:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
<- paste here

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
<- paste here

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

chrissooo, 2022-01-07:
Ja, sorr ... habe es auf versehen verschickt wo ich noch nicht fertig war .... soll ich editieren oder neu machen? Ich habe logs usw.

chrissooo, 2022-01-07:
So, jetzt ... danke!

andig, 2022-01-07:
Renault hat kein Refresh API. Vllt. löst ein neuer Login das aus. Kannst Du mal 

    evcc vehicle

starten wenn der SoC gerade nicht passt und schauen ob _das_ Dir den richtigen Wert ausgibt?

chrissooo, 2022-01-07:
Hi,

evcc vehicles liefert die richtigen Daten. Ladevorgang wird dadurch aber nicht gestartet. Das habe ich nämlich auch schon mir überlegt und testweise so konfiguriert dass die Fahrzeigdaten auch nur bei verbunden abgerufen werden statt nur bei laden. Hat nichts gebracht.

DIenst neustart von evcc weckt Dacia auf und beginnt wieder mit laden. Gerade nochmal getestet. Hängt das vielleicht mit der Ansteuerung der Heidelberg Energie Control zusammen? Das kommt mir als einzigstes in Frage dass beim Neustart der LAadebefehl anderst ist wie wenn man auf Sofort klickt?

andig, 2022-01-07:
> DIenst neustart von evcc weckt Dacia auf und beginnt wieder mit laden. Gerade nochmal getestet. Hängt das vielleicht mit der Ansteuerung der Heidelberg Energie Control zusammen? Das kommt mir als einzigstes in Frage dass beim Neustart der LAadebefehl anderst ist wie wenn man auf Sofort klickt?

Mach doch mal ein Log von Neustart vs. Sofort und wir schauen mal was auf dem Modbus los ist.

premultiply, 2022-01-07:
Ist oben angehängt.

chrissooo, 2022-01-07:
Die logfiles sind im ersten Post. Das eine logfiles ist wenn das sofort versagt und das andere ist der neustart

premultiply, 2022-01-08:
Das Aufwecken des Fahrzeug nach dem Neustart von evcc ist ursächlich darin begründet dass die Heidelberg Wallboxen nach 15 Sekunden ohne Kommunikation in den Status F (=Error) wechseln. Dadurch sollen Fehlfunktionen bei ausgefallenem Lastmanagment verhindert werden. Anschließend wird die Kommunikation nach der Initialisierung durch wieder evcc wieder aufgenommen und die Boxen kehren wieder in den Normalbetrieb zurück. Der Zustand F bewirkt normgemäß einen -12V Pegel auf CP (also 0 Volt nach der fahrzeugseitigen Diode = Kabel getrennt) und weckt damit das Fahrzeug auf weil dies die anschließend Reaktivierung des CP-Signals als Aufwachtrigger erkennt.
Dieser Zustand ist bei der Heidelberg-Box aber anders als bei anderen Controllern nicht automatisch oder manuell abseits der vollständigen Einstellung der Buskommunikation bis zum Timeout nicht herstellbar (zumindest nicht dokumentiert).
Abseits einer ursächlichen Problemlösung seitens Dacia/Renault in der Fahrzeugsoftware und/oder der Implementierung einer automatischen CP-Signalisierung seitens Heidelberg als durchaus branchenüblicher Workaround bleibt für evcc hier nur noch eine mögliche Triggerung mittels Vehicle-API als Workaround.

andig, 2022-01-08:
Evcc hat keine Wakeup API. Was wir hatten war Ladung start/stop, die wurde jedoch aufgebaut und seitdem nicht mehr vermisst, siehe https://github.com/evcc-io/evcc/discussions/1929

andig, 2022-01-08:
Ich mache hier mal zu. Die Grunddiskussion dazu sollten wir in #1929 weiter führen.

Danke @premultiply für die Analyse!

closed 2022-01-08 by andig

chrissooo, 2022-01-08:
@premultiply ja, auch nochmal Danke von mir ... gestern bis 1:30 zusammen analysiert :D ich bin halb erfroren in der Garage :)

@andig, #2237 kann man testen? oder ist das nur vorbereitend?

chrissooo, 2022-01-10:
Ein kleines Update dazu. Ich habe sowohl mit Dacia / Renault wie auch Heidelberg Kontakt. Heidelberg hat dann wohl nach längerem Dialog die Problematik verstanden, antwortet aber folgendermaßen:

_"einen undokumentierten Modbus-Befehl gibt es dazu nicht. Eine Änderung der Software ist aktuell nicht vorgesehen. Die einzige Möglichkeit das Fahrzeug zu „wecken“ ist es über den Stecker erneut zu kontaktieren."_