evcc-io/evcc #2244: 1p3p switching stops when detected phases = 2

texperiri, 2022-01-08:
**Describe the bug**
Switching from "now" mode to "minpv" mode correctly switches from 3p to 1p, but does not switch back to 3p - when selecting "now" mode.

**To Reproduce**
Steps to reproduce the behavior:
1. System is in "now" mode with 3 phases active and switched to "minpv" mode - with no/small pv power available.
In "minpv" the phases are switched from 3p to 1p (after 10 minutes)
2. Switch back to "now" -> phases stay at "1p" and are not scaled up to 3p.

Detail:
it seems that the charger - after being "disabled" and phase-switched reports 2 phases for a short time - this seems to lead to a bad system state from which switching to 3p is no longer possible.

Sidenote:
i detected this issue on a feature branch of mine where i develop the integration of an external 1p3p switch (pull request is on its way)
As far as i can assess the issue it seems to me that the problem is not part of my enhancements, but part of the existing loadpoint.go code.

```
[lp-1  ] DEBUG 2022/01/07 12:17:07 charge currents: [4.51 1.9 0.458]A
[lp-1  ] DEBUG 2022/01/07 12:17:07 detected phases: 2p [4.51 1.9 0.458]A
...
[lp-1  ] DEBUG 2022/01/07 12:17:16 max charge current: 0A = 0A + -2.63A (1209W @ 2p)
[lp-1  ] WARN 2022/01/07 12:17:16 inconsistent phases: 1p configured < 2p observed active, updating internal state to 2p
```

**Expected behavior**
When switching back to "now" the phases should scale up to 3p

**EVCC details:**
Show output of <code>evcc -v</code>:
```
see docker

```
<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2022/01/08 21:21:16 evcc 0.0.1-alpha (HEAD)
[main  ] INFO 2022/01/08 21:21:16 using config file evcc.yaml
[RutIp4] DEBUG 2022/01/08 21:21:17 url: udp://192.168.15.18:30303, out1p3p: 2, outReadbackCtrl: 1, outReadbackSts: 3, outWallboxEnable: 4, responseTimeoutSec: 2, relaisSwitchTimeMs: 100
[lp-1  ] WARN 2022/01/08 21:21:20 loadpoint.onDisconnect is deprecated and will be removed in a future release. Use loadpoint.resetOnDisconnect instead.
[lp-1  ] DEBUG 2022/01/08 21:21:20 setPhases: 1
config
======

grid: GridMeter
---------------
Power:  1709W
Energy: 0.0kWh

pv: PvMeter
-----------
Power: 0W

battery: BatteryMeter
---------------------
Power: -0W
SoC:   0%

loadpoint 1
===========

charger: KeContactP30
---------------------
Power:          0W
Energy:         1077.5kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false
Charged:        12.3kWh
Identifier:     0000000000000000
Diagnostic dump:
{ID:100 Serial:22378882 SessionID:0 CurrHW:20000 EStart:10652308 EPres:122570 Started:2022-01-07 12:30:08.000 Ended:2022-01-08 13:49:56.000 StartedS:1641558608 EndedS:1641649796 Reason:1 TimeQ:3 RFIDTag:0000000000000000 RFIDClass:00000000000000000000 Sec:4284407}
[... 1808 lines cut ...]
[cache ] DEBUG 2022/01/07 12:22:36 chargedEnergy: 1652
[cache ] DEBUG 2022/01/07 12:22:36 chargeDuration: 20m0s
[lp-1  ] DEBUG 2022/01/07 12:22:36 charger status: C
[cache ] DEBUG 2022/01/07 12:22:36 connected: true
[cache ] DEBUG 2022/01/07 12:22:36 charging: true
[cache ] DEBUG 2022/01/07 12:22:36 enabled: true
[lp-1  ] WARN 2022/01/07 12:22:36 vehicle soc: Get "https://mobileapi.apps.emea.vwapps.io/vehicles/*******************/status": dial tcp: lookup mobileapi.apps.emea.vwapps.io: Try again (ignored by estimator)
[lp-1  ] DEBUG 2022/01/07 12:22:36 soc estimated: 69.80% (vehicle: 69.00%)
[lp-1  ] DEBUG 2022/01/07 12:22:36 vehicle soc: 69%
[lp-1  ] WARN 2022/01/07 12:22:36 updating remaining time failed: Get "https://mobileapi.apps.emea.vwapps.io/vehicles/*******************/status": dial tcp: lookup mobileapi.apps.emea.vwapps.io: Try again
[cache ] DEBUG 2022/01/07 12:22:36 vehicleSoC: 69
[cache ] DEBUG 2022/01/07 12:22:36 chargeRemainingDuration: 1h34m46s
[cache ] DEBUG 2022/01/07 12:22:36 chargeRemainingEnergy: 6575.78888888889
[lp-1  ] DEBUG 2022/01/07 12:22:36 charge power: 4163W
[cache ] DEBUG 2022/01/07 12:22:36 chargePower: 4163.172
[lp-1  ] DEBUG 2022/01/07 12:22:36 charge currents: [19.5 0.003 0]A
[lp-1  ] DEBUG 2022/01/07 12:22:36 detected phases: 1p [19.5 0.003 0]A
[cache ] DEBUG 2022/01/07 12:22:36 chargeCurrents: [19.544 0.003 0]
[cache ] DEBUG 2022/01/07 12:22:36 activePhases: 1
[savings] DEBUG 2022/01/07 12:22:36 1.7kWh charged since 9h52m30s
[savings] DEBUG 2022/01/07 12:22:36 0.1kWh own energy (4.1%)
[cache ] DEBUG 2022/01/07 12:22:36 homePower: 2466.8280000000004
[site  ] DEBUG 2022/01/07 12:22:46 ----

</pre></details>

texperiri, 2022-01-08:
The issue can be reproduces with the simulation program that comes with the following pull-request: https://github.com/evcc-io/evcc/pull/2245

andig, 2022-01-09:
> it seems that the charger - after being "disabled" and phase-switched reports 2 phases for a short time - this seems to lead to a bad system state from which switching to 3p is no longer possible.

That is exactly https://github.com/evcc-io/evcc/issues/2230. I‘ll still leave this open as it should switch up to 3p nonetheless. Might also consider nich switching up to 2p in the first place but only consider 1p and 2p as valid states.

andig, 2022-01-09:
Corrected error message in https://github.com/evcc-io/evcc/commit/445bafcb8ee8844c7bdefbd7e1b9549dcf9c5b38, the rest is a duplicate of #2230.

closed 2022-01-09 by andig