evcc-io/evcc #2262: PSA: Keine Benachrichtigung bei Ende des Ladevorgangs

rickcrash, 2022-01-11:
Hardware: Citroën ëC4 an ABL eMH-1

Ich habe die Benachrichtigung per E-Mail eingerichtet. Der entsprechende Abschnitt in der evcc.yaml sieht wie folgt aus:

messaging:
  events:
     start: # Ladevorgang startet
       title: ëC4 Laden gestartet
       msg: Laden gestartet im "${mode}" Modus
     stop: # Ladevorgang beendet
       title: ëC4 Laden beendet
       msg: Laden beendet. "${energy:%.1f}" kWh geladen in "${duration}"
     connect: # Fahrzeug angeschlossen
       title: ëC4 Fahrzeug angeschlossen
       msg: Fahrzeug angeschlossen
     disconnect: # Fahrzeug getrennt
       title: ëC4 Fahrzeug getrennt
       msg: Ladekabel getrennt

Bei Verbindung des Fahrzeuges mit der Wallbox erhalte ich eine E-Mail mit dem erwarteten Inhalt. Ebenso bei Start des Ladevorganges. (Diese E-Mail wird erst ausgelöst, wenn ich die App des Herstellers aufrufe und aktualisiere, aber das ist hier OT).

Bei Abschluss des Ladevorganges erhalte ich keine Benachrichtigung.

Bei Trennung des Fahrezuges vom Ladekabel erhalte ich wieder eine Benachrichtigung.

**Expected behavior**
Benachrichtigung bei Beeindigung des Ladevorgangs

evcc version 0.80 (cfb174f9)

evcc läuft auf einem dietpi-Server (debian); kein docker

premultiply, 2022-01-11:
Log?

rickcrash, 2022-01-11:
Hier die Ausgabe von 

`evcc --log debug:`

[main  ] INFO 2022/01/11 21:37:44 evcc 0.80 (cfb174f9)
[main  ] INFO 2022/01/11 21:37:44 using config file /root/evcc.yaml
[main  ] INFO 2022/01/11 21:37:44 listening at 0.0.0.0:7070
[cache ] DEBUG 2022/01/11 21:38:00 sponsor: rickcrash
[cache ] DEBUG 2022/01/11 21:38:00 currency: EUR
[cache ] DEBUG 2022/01/11 21:38:00 savingsSince: 10.760573ms
[cache ] DEBUG 2022/01/11 21:38:00 title: ABL eMH-1
[cache ] DEBUG 2022/01/11 21:38:00 minCurrent: 6
[cache ] DEBUG 2022/01/11 21:38:00 maxCurrent: 16
[cache ] DEBUG 2022/01/11 21:38:00 phases: 3
[cache ] DEBUG 2022/01/11 21:38:00 activePhases: 3
[cache ] DEBUG 2022/01/11 21:38:00 hasVehicle: true
[lp-1  ] INFO 2022/01/11 21:38:00 vehicle updated: unknown -> Citroën ëC4
[cache ] DEBUG 2022/01/11 21:38:00 mode: now
[cache ] DEBUG 2022/01/11 21:38:00 targetSoC: 100
[cache ] DEBUG 2022/01/11 21:38:00 minSoC: 0
[cache ] DEBUG 2022/01/11 21:38:00 vehiclePresent: true
[cache ] DEBUG 2022/01/11 21:38:00 vehicleTitle: Citroën ëC4
[cache ] DEBUG 2022/01/11 21:38:00 vehicleCapacity: 50
[cache ] DEBUG 2022/01/11 21:38:00 vehicleRange: 0
[cache ] DEBUG 2022/01/11 21:38:00 vehicleOdometer: 0
[lp-1  ] DEBUG 2022/01/11 21:38:00 max charge current: 6A
[site  ] INFO 2022/01/11 21:38:00 site config:
[cache ] DEBUG 2022/01/11 21:38:00 siteTitle: Hauptzentrale
[cache ] DEBUG 2022/01/11 21:38:00 chargeCurrent: 6
[site  ] INFO 2022/01/11 21:38:00   meters:    grid ✓ pv ✗ battery ✗
[site  ] INFO 2022/01/11 21:38:00     grid:    power ✓ energy ✗ currents ✗
[cache ] DEBUG 2022/01/11 21:38:00 gridConfigured: true
[cache ] DEBUG 2022/01/11 21:38:00 pvConfigured: false
[lp-1  ] INFO 2022/01/11 21:38:00 loadpoint 1:
[cache ] DEBUG 2022/01/11 21:38:00 batteryConfigured: false
[lp-1  ] INFO 2022/01/11 21:38:00   mode:      now
[lp-1  ] INFO 2022/01/11 21:38:00   charger:   power ✓ energy ✗ currents ✓ timer ✗
[lp-1  ] INFO 2022/01/11 21:38:00   meters:    charge ✓
[lp-1  ] INFO 2022/01/11 21:38:00     charge:  power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/01/11 21:38:00   vehicles:  ✓
[lp-1  ] INFO 2022/01/11 21:38:00     car 0:   range ✓ finish ✓ status ✓ climate ✓
[cache ] DEBUG 2022/01/11 21:38:00 chargeConfigured: true
[site  ] DEBUG 2022/01/11 21:38:00 ----
[site  ] DEBUG 2022/01/11 21:38:00 grid power: 500W
[site  ] DEBUG 2022/01/11 21:38:00 site power: 500W
[cache ] DEBUG 2022/01/11 21:38:00 gridPower: 500
[cache ] DEBUG 2022/01/11 21:38:00 mode: now
[main  ] FATAL 2022/01/11 21:38:00 listen tcp 0.0.0.0:7070: bind: address already in use

Falls weitere logs hilfreich sind, bitte mitteilen, welchen Befehl genau ich eingeben soll.

andig, 2022-01-11:
Mach bitte win vollständiges Ticket in dem man Dein Problem im Logfile sieht.

closed 2022-01-11 by andig

andig, 2022-01-12:
@rickcrash ich brauche v.a. ein Logfile, in dem das Problem auftritt. In Deinem Fall die Beendigung des Ladevorgangs. Dafür wäre es auch gut, wenn Du 

    log:
      push: trace

setzen könntest damit wir auch die Messages mit sehen. Ich mach das Ticket gerne wieder auf wenn Du ein Logfile hast!

rickcrash, 2022-01-14:
Ich habe jetzt das log-file erstellt.

Der Ladevorgang war um 19:43 Uhr beendet mit push-Nachricht. 

Als Test habe ich jetzt mal die Variablen aus dem Text zum Ladestopp herausgenommen. 

[evcc.c4.log](https://github.com/evcc-io/evcc/files/7872635/evcc.c4.log)

Vielen Dank und Grüße
Rick

andig, 2022-01-14:
> Der Ladevorgang war um 19:43 Uhr beendet mit push-Nachricht.

D.h. also es hat funktioniert? Dann gibts hier nichts zu tun ;)

rickcrash, 2022-01-14:
Nein, leider nicht.

Das log erstellen hat funktioniert.

Der Ladevorgang endet auch bei SoC 100%.

Aber es kommt noch immer keine Nachricht, wenn das Laden endet :-(
und das war das ursprüngliche Problem ;-)

https://github.com/evcc-io/evcc/issues/2262#issuecomment-1013388234

Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

andig, 2022-01-14:
Ok. Dann schauen wir doch mal ins Log:

> invalid message template for stop: wanted: [energy duration], got: [gridPower batteryConfigured savingsEffectivePrice minCurrent maxCurrent chargeRemainingEnergy vehicleSoC chargeConfigured chargedEnergy minSoC connected mode gridConfigured homePower connectedDuration savingsSelfConsumptionPercent activePhases savingsTotalCharged chargeCurrents chargeCurrent chargeDuration tariffFeedIn chargeRemainingDuration title tariffGrid savingsSince siteTitle chargePower targetSoC phases sponsor savingsAmount savingsSelfConsumptionCharged vehicleTitle vehiclePresent charging enabled vehicleRange vehicleOdometer pvConfigured pvRemaining pvAction savingsGridCharged vehicleCapacity currency hasVehicle]

Scheint mir ziemlich eindeutig zu sein ;)

rickcrash, 2022-01-14:
Mir auch.

Abgefragt wurde die Ladedauer und als Antwort kam ein Haufen aller möglicher Informationen. 

Aber wie komme ich jetzt an meine Nachricht mit Angabe der Ladedauer und der geladenen Energiemenge? 🤔

https://github.com/evcc-io/evcc/issues/2262#issuecomment-1013446125

andig, 2022-01-14:
Schau mal in die Meldung. Da stehen auch die verfügbaren Werte. Genauso wie korrekterweise in https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml#L162