evcc-io/evcc #2272: resetOnDisconnect does not reset mode

ExecuteGK, 2022-01-12:
evcc.yaml has been configured as follows:

- title: Vorne
  charger: daheimladen
  mode: off
  vehicle: VW
  resetOnDisconnect: true
  soc:
    target: 80
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 2m # switch charger contactor not more often than this (default 10m)

Logfile for Loadpoint:
an 12 12:39:49 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:39:49 charge currents: [0 0 0]A
Jan 12 12:40:06 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:06 charge power: 0W
Jan 12 12:40:07 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:07 charge currents: [0 0 0]A
Jan 12 12:40:07 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:07 charger status: B
Jan 12 12:40:08 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:08 charge start energy: 173kWh
Jan 12 12:40:08 flowcontrol evcc[46040]: [lp-2  ] INFO 2022/01/12 12:40:08 car connected
Jan 12 12:40:08 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:08 vehicle api refresh
Jan 12 12:40:08 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:08 pv action: inactive
Jan 12 12:40:09 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:09 vehicle soc: 37%
Jan 12 12:40:09 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:40:09 vehicle range: 153km
.
.
.
an 12 12:41:35 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:35 set charge mode: now
Jan 12 12:41:35 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:35 pv action: inactive
Jan 12 12:41:35 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:35 charge power: 0W
Jan 12 12:41:36 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:36 charge currents: [0 0 0]A
Jan 12 12:41:37 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:37 charger status: B
Jan 12 12:41:37 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:37 next soc poll remaining time: 58m30s
Jan 12 12:41:41 flowcontrol evcc[46040]: [lp-2  ] WARN 2022/01/12 12:41:41 charger out of sync: expected disabled, got enabled
Jan 12 12:41:42 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:42 max charge current: 16A
Jan 12 12:41:42 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:42 charger enable
Jan 12 12:41:42 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:42 charge power: 0W
Jan 12 12:41:43 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:43 charge currents: [0 0 0]A
Jan 12 12:41:46 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:46 charge power: 0W
Jan 12 12:41:47 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:47 charge currents: [0 0 0]A
Jan 12 12:41:49 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:49 charger status: C
Jan 12 12:41:50 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:50 charge start energy: 173kWh
Jan 12 12:41:50 flowcontrol evcc[46040]: [lp-2  ] INFO 2022/01/12 12:41:50 start charging ->
Jan 12 12:41:50 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:50 vehicle soc: 37%
Jan 12 12:41:50 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:50 vehicle range: 153km
Jan 12 12:41:51 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:51 charge power: 10100W
Jan 12 12:41:51 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:51 charge currents: [15.2 15.1 15.2]A
Jan 12 12:41:51 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:41:51 detected phases: 3p [15.2 15.1 15.2]A
Jan 12 12:42:06 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:42:06 charge power: 10100W
Jan 12 12:42:07 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:42:07 charge currents: [15.2 15.1 15.2]A
Jan 12 12:42:07 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:42:07 detected phases: 3p [15.2 15.1 15.2]A
Jan 12 12:42:10 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:42:10 charger status: C
Jan 12 12:42:10 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:42:10 next soc poll remaining time: 59m39s
Jan 12 12:42:10 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 12:42:10 vehicle soc: 37%
.
.
.
Jan 12 14:05:08 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:08 vehicle range: 218km
Jan 12 14:05:09 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:09 charge power: 10100W
Jan 12 14:05:09 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:09 charge currents: [15.1 15.1 15.2]A
Jan 12 14:05:09 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:09 detected phases: 3p [15.1 15.1 15.2]A
Jan 12 14:05:26 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:26 charge power: 10100W
Jan 12 14:05:27 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:27 charge currents: [15.1 15.1 15.2]A
Jan 12 14:05:27 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:27 detected phases: 3p [15.1 15.1 15.2]A
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 charger status: A
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 final charge energy: 13.8kWh
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] INFO 2022/01/12 14:05:28 stop charging <-
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 pv timer reset
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 pv action: inactive
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] INFO 2022/01/12 14:05:28 car disconnected
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] WARN 2022/01/12 14:05:28 invalid charge mode: 0
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 set min current: 6
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 set max current: 16
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 set min soc: 0
Jan 12 14:05:28 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:28 set target soc: 80
Jan 12 14:05:29 flowcontrol evcc[46040]: [lp-2  ] WARN 2022/01/12 14:05:29 charger out of sync: expected enabled, got disabled
Jan 12 14:05:29 flowcontrol evcc[46040]: [lp-2  ] ERROR 2022/01/12 14:05:29 charger: charging station refused to start transaction
Jan 12 14:05:30 flowcontrol evcc[46040]: [lp-2  ] DEBUG 2022/01/12 14:05:30 charger disable

I saw a similar behavior when evcc is started with this config. No charge mode is selected.

Regards,

Sebastian

rivengh, 2022-01-12:
Sebastian, can you try to add quotation marks around the mode value and give a short feedback:

`  mode: "off"`

This solves the issue, at least on my side.

andig, 2022-01-12:
> However, this is inconsistent with the documentation. On the other loadpoint, I've set minpv, which works without quotes.

"off" is `false` in Yaml 1.1. This will be fixed in Yaml 1.2. I'll need to check why the designed workaround we have in place doesn't work.

andig, 2022-01-12:
Resetting to "off" is also working in my test. Are you running a current version?

ExecuteGK, 2022-01-12:
Yes, I am running 0.81.

"off" works for me. So if you implemented a workaround, something is wrong there. however, if it's due to be fixed in YAML 1.2, I recommend to just add a known-issues entry in the Docu and leave it until the migration to 1.2 is complete.

Regards,

Sebastian

andig, 2022-01-12:
Let me know at cpuidle@gmx.de if you want to run a remote debug session. Closing here with workaround in place and wfm.

closed 2022-01-12 by andig

ExecuteGK, 2022-01-12:
It's totally fine with the workaround.
The most important thing is that the Loadpoint is secured once I pull the plug. It's accessible in front of my house and I need some sort of authentication since evcc bypasses the rfid auth on the WB due to the design of the daheimladen api.