evcc-io/evcc #2281: Builds/Packaging: Raspberry Pi 1 / Zero / Zero W (armel) Kompatiblität wiederherstellen

premultiply, 2022-01-13:
Related #2276 and #2243

andig, 2022-01-13:
Wir haben Armv5 nie unterstützt. Das ist ein Featurerequest. Wollen wir das unterstützen? Es ist ohne weiteres möglich, selbst zu kompilieren.

premultiply, 2022-01-13:
Es geht um armv6 und das ist wohl nur ein Packagingfehler.

andig, 2022-01-13:
armel laut Titel ist NICHT v6. Armel ist v5 siw.

DerAndereAndi, 2022-01-13:
Genau. `armel` ist `armv5`. Aber Raspbian meint da etwas anderes machen zu müssen, und nimmt für diese Geräte `armel` obwohl da auch `armv6` chips drin sein. :(

D.h. um das zu supporten müssten wir `armv5` hinzufügen und die als `armel` markieren.

andig, 2022-01-13:
> Oder diese Anwender müssen das armv6 binary nehmen und können apt nicht nutzen.

Wäre meine bevorzugte Version. Die Systeme sind nicht mehr aktuell.

premultiply, 2022-01-13:
Ist für Laien kaum nachvollziehbar.
Würde regelmäßige Nachfragen wie die obigen nach sich ziehen und wäre daher IHMO keine gute Option.

DerAndereAndi, 2022-01-13:
Ich schau mal wieviel Aufwand es ist, für Raspberry OS eigene Pakete zu machen, mit den gleichen Binaries

DerAndereAndi, 2022-01-13:
Also:
- es ist kein Packaging Fehler, da nur Raspberry Pi OS armel dafür verwenden will, obwohl es armv6 Geräte sind
- Ich habe fürs erste eine Notiz in die Dokumentation hinzugefügt, dass die Raspberry 1 Modelle die manuelle Installation verwenden müssen
- Ein zweites Packaging ist mit dem Tooling nicht trivial, wenn überhaupt möglich. Bräuchte wohl eher ne komplett eigene Konfiguration, und dann muss man auch in der Doku die Distribution beim `apt` angeben. Das macht es für mich jetzt erst mal zu aufwändig.

D.h. würde das erst mal im Status Quo lassen, in der Doku ist es jetzt erwähnt und damit sollte das gut auffindbar sein. Macht die Installation für diese Leute etwas aufwändiger, aber da ist das Potenzial wohl dann eher die Doku noch zu verbessern :D

premultiply, 2022-01-13:
Ich denke das ist erstmal eine akzeptable Lösung.
Dann kann hier erstmal zu.

closed 2022-01-13 by premultiply