evcc-io/evcc #2286: Heidelberg Energy Control Vehicle Wakeup feature

premultiply, 2022-01-13:
Die Heidelberg Boxen haben von sich aus keine branchenübliche automatische Weckfunktion (vgl. Phoenix Controller) für schlafende Autos über die CP-Signalisierung. Es gibt auch keine Möglichkeit dies bei Bedarf gezielt manuell anzutriggern (vgl. SmartWB).
Jegliche Intervention beim Hersteller dazu dies ggf. firmwareseitig nachzurüsten blieb leider völlig erfolglos obwohl die Hardware die entsprechenden Voraussetzungen mitbringt.

Durch diverse Tests konnte nämlich herausgefunden werden, dass der vorhandene interne Status F bei der Box eine entsprechende CP-Signalisierung (-12V) hervorruft und die Fahrzeuge damit aufwachen wenn man die Box nur in diesen Zustand versetzt.

Wie schon geschrieben gibt es jedoch leider keine Möglichkeit auf direktem Wege in den Status F zu wechseln.
Jedoch haben die Boxen eine Watchdog-Funktion, die die Kommunikation mit dem Modbus-Master überwacht (Factory Default 15 Sekunden) und bei ausbleibender Kommunikation mit dem Master standardmäßig in den Zustand F wechseln.
Das Timeout selbst lässt sich Millisekundengenau von 0 (= deaktiviert) bis 65535ms per Modbus Register konfigurieren.

Achtung: Grundsätzlich nachteilig am Zustand F ist, dass der eigentliche Zustand ob ein Fahrzeug angeschlossen ist nicht gleichzeitig erkannt werden kann.

Die Idee ist nun in einer Wakeup-Funktion gezielt das Timeout der Boxen für 3 Sekunden auszulösen und anschließend die Kommunikation fortzusetzen. Hierzu müsste bei Bedarf das Timeout-Register zunächst auf 1 ms gesetzt werden und nach 3 Sekunden in denen sonst keine Modbus-Kommunikation mit der Box mehr stattfinden darf wieder auf den Ursprungswert zurückgesetzt werden.
Die Box würde für diese Zeit in den Status F wechseln und anschließend wieder in den Normalbetrieb.

Related #2265 /cc @chrissooo

andig, 2022-01-13:
Also: du möchtest Status F für wakeup auslösen?

andig, 2022-01-13:
Wontfix für mich. Bewusst Fehlerzustände herbeizuführen um andere Seiteneffekte auszulösen kann egtl. nicht zielführend sein.

premultiply, 2022-01-13:
Ja, es sollte dazu bei der Heidelberg einen kurzen gezielten und gekaselten Zustand F geben (absolut vergleichbar mit SmartWB) der via dem neuen Wakeup-PR ausgelöst wird bei Fahrzeugen die nicht zuvor schon längst selbst aufgewacht sind.

chrissooo, 2022-01-13:
@andig Grundsätzlich bin ich vollkommen bei dir. Das ist natürlich ein Gemurkse einen Fehler bewusst in Kauf zu nehmen um Work-Around-technisch etwas Hinzuschustern. Ich finde aber man sollte es trotzdem tun. Klar habe ich hier persönliches Interesse und neben der Möglichkeit das ganze über die Renault API zu tun, ist das eine doch zuverlässige (habe ein primitiven Test gemacht) Methode Überschussladung in der Konstellation zu realisieren. Die Renault API, falls das dann mal funktioniert und umgesetzte werden sollte, ist jedoch viel Fehleranfälliger. Hier muss das Auto sicher Empfang haben, und die Renault Server dürfen keine Ausfälle haben. Eine Kombination aus beiden wäre wohl das Beste?

Die Heidelberg wurde neben der ABL wohl oft verkauft, von vielen als sehr gut bewertet, gut beworben und günstig. Deshalb habe ich damals zwei installiert. Das ist natürlich Wahrsagerei, aber ich denke dass viele die Boxen über die Förderung gekauft haben bis jetzt aber oft kein Auto und/oder PV Anlage hatten, so wie ich oder sich mit der Thematik Überschussladung nicht beschäftigt haben. Denke dass in Zukunft öfter diese Problemmstellung aufpoppt. Auch mit anderen Box / Auto Konstellationen. Dann wäre es natürlich cool wenn evcc hier ein optionale schicke Lösung anbieten könnte. Soweit ich das beurteilen kann ist das was der @Chris591 macht doch wohl das Fundament und scheint solide zu sein. Und wenn das sauber, optional konfigurierbar und den Rest nicht beeinflusst, dann würde doch evcc nach wie vor eine solide Software sein und Wohl flexibel wie keine Andere sein. Zumindest wenn es um Anbindung an Hardware geht.

steff393, 2022-01-19:
Ich seh das eher wie @andig :-)
Aber wenn ihr schon die Lebensdauer des Flash testen wollt, dann wäre vielleicht Register 259 die elegantere Methode:
https://github.com/steff393/wbec/wiki/Modbus-Register

premultiply, 2022-01-20:
Hier gibt es nichts zu sehen und hier wurde bereits viel getestet.
Dies ist ein funktionierender Workaround für die mangelhafte Firmware der Box der nur unter bestimmten aber sehr relevanten Bedingungen überhaupt ausgeführt wird. Der Flash hat laut Hersteller zudem Wear-Leveling und soll dies problemlos vertragen.
Danke für den Hinweis mit dem RemoteLock-Register!

chrissooo, 2022-01-21:
@steff393, wie schon @premultiply gesagt hat, das sollte laut Heidelberg Support der Speicher können. Meiner Meinung nach könnte man das auch ohne Register Schreiben tun, indem man bei geschickter timeout Einstellung der Box die dann bewusst in den Timeout schickt. Wenn das hier dann Mal im Master ist, und grundsätzlich die WackeUp Funktion in evcc drin ist, werde ich das für mich persönlich abändern und testen ob das auch ein gangbarer weg wäre.

PR #2287 (2022-01-31): Heidelberg wakeup feature
Requires #2265
Fix #2286
closed 2022-01-31 by premultiply