evcc-io/evcc #2306: Create soc update event

andig, 2022-01-15:
> Und wäre es möglich, ein Event "SoC update" oder so zu definieren, wo eine Message mit dem vom Fahrzeug abgerufenen SoC übermittelt wird?

Einmalig bei auslesen wäre kein Problem? Regelmäßig (alle 10% oder so) schon :O

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/2305#discussioncomment-1975920_

andig, 2022-01-15:
/cc @premultiply wäre das sinnvoll ohne konkrete Verbindung zum Ladefortschritt?

premultiply, 2022-01-15:
Fände ich spontan eine gute Idee. Braucht ja nur zu abonnieren wer es so detailliert mag.
Verstehe dies so, dass dann bei einem jeder SoC-Änderung via Vehicle-API ein abonnierbares Event auftreten würde, korrekt?

junibart, 2022-01-16:
> /cc @premultiply wäre das sinnvoll ohne konkrete Verbindung zum Ladefortschritt?

Ich würde das gern nutzen, weil meine Hersteller-Cloud den SoC-Wert gerne mal nicht oder verzögert rausrückt. Wenn er kurz nach Beginn der Ladung kommt, weiss ich, dass ich das laufen lassen kann. Kommt er nicht, muss ich mich drum kümmern. 
Grüße, Gunnar

andig, 2022-01-16:
@premultiply das gibts nur im Estimator. Ich hab nix elegantes gefunden den physischen Wert einfach rauszuholen. Falls Du eine Idee hast…

andig, 2022-01-18:
Ich denke es wäre einfacher, all 10% eine Nachricht zu schicken, ggf. auch initial. Das würde dann aber auch Estimator beinhalten und nicht nur physikalische Abfragen beim Auto. Wäre das ok?

andig, 2022-01-18:
Wär super wenn das jemand testen könnte.

/cc @DerAndereAndi wäre eine Testanleitung für PRs mit Hilfe des `gh cli` und `go` noch einen Artikel wert?

closed 2022-01-20 by andig