evcc-io/evcc #2311: Alphatec Mini charger support

DerAndereAndi, 2022-01-16:
It would be great if the Alphatec Mini charger could be supported.

A documentation including Modbus details is available here (pages 11-13): https://shop.alphatec-systeme.de/media/pdf/4d/0e/64/MontageanleitungwlFxbRgs4NKK3.pdf

We have a user with the box who might be able to help testing

oldcoolman, 2022-01-16:
Hi, ich habe solche boxen hier und könnte testen.

andig, 2022-01-16:
Kannst du Zugangsdaten an cpuidle@gmx.de schicken?

andig, 2022-01-16:
PR ist da. Mir wird aus der Doku allerdings nicht klar wie der Ladestrom einzustellen ist, der Teil fehlt daher noch.

premultiply, 2022-01-16:
[image]

andig, 2022-01-16:
Kann getestet werden

closed 2022-01-17 by andig