evcc-io/evcc #2314: Timezone issues for target time suggestion

naltatis, 2022-01-16:
https://github.com/evcc-io/evcc/discussions/1433#discussioncomment-1977519

> Eigentlich war 7:00 nur der Defaultwert, für die nächsten Tage wird dann die letzte gesetzte Uhrzeit verwendet. Zumindest war das die Idee, vielleicht aber noch nicht umgesetzt?
 
Bei der Implementierung der Default-Zeit haben wir leider vergessen das Zeitzonenthema zu berücksichtigen. Momentan verwendet die UI die Variable `targetTimeHourSuggestion: Number` um die Vorauswahl zu setzen. Hier laufen wir natürlich in Probleme wenn sich die Zeitzonen zwischen Client und Server unterscheiden.
Mein Vorschlag wäre, dass wir uns die letzte "aktivierte" Zielzeit (Stunde/Minute) nur im Browser merken (z.B. localStorage) und für diese beim nächste Mal wieder anzubieten. Der Wert `targetTimeHourSuggestion` würde dann aus der API verschwinden. Nachteil ist, dass wir dann keinen "Sync" zwischen Devices (Desktop/Phone) haben. Das halte ich aber erstmal für vertretbar.

schenlap, 2022-01-16:
ist das Zeitzonenproblem wirklich nicht einfach lösbar? Es hört sich so einfach an die letzt eingestellt Zeit wieder zu verwenden, gefällt mir auch besser also im Browser Cache.

naltatis, 2022-01-16:
Verstehe noch nicht ganz was du meinst. Schlägst du vor eine targetTimeSuggestion als UTC time über die API bereit zu stellen? Können wir auch machen. Da muss dann allerdings auch die Logik für Tagwechsel und so rein. Also der Wert müsste sich ein Mal am Tag aktualisieren. Finde ich auch ne gute Idee, hält Logik aus der UI, weiß aber nicht ob das einfacher zu bauen ist.

schenlap, 2022-01-16:
Nein, so habe ich das nicht gemeint. Meine Antwort ist nicht technisch zu lesen. Wollte sagen: mir hat es so wie es bisher war besser gefallen als im browser Cache. Der Bug müsste halt noch raus :-).
Ich habe aber keine Ahnung woran es genau scheitert bzw. wer überhaupt den Vorschlag generiert.

naltatis, 2022-01-16:
Der aktuelle Vorschlag (die Stunde) wird hier nach erfolgreicher Zielladung generiert:
https://github.com/evcc-io/evcc/blob/master/core/loadpoint_api.go#L125

Um hier sauber die Zeitzonen zu berücksichtigen müssen wir aber das Format ändern. Am besten in ein Datum (mit Zeitzonenangabe).

premultiply, 2022-01-16:
Der Vorschlag selbst ist ja "einheitslos", genau wie die eingegebene Ziezeit, sprich _jeweilige_ Lokalzeit aus Sicht des Anwenders (am Client).

Bei der Beauftragung und Ausführung durch evcc braucht es dann aber die Umrechnung nach UTC,
für die Anzeige und Bearbeitung im Client-Browser umgekehrt als dessen Lokalzeit.

Ich denke es dürfte dann wohl am einfachsten sein dies direkt im Client passend umzurechnen und im Backend/API nur mit UTC zu arbeiten, oder?

andig, 2022-01-18:
> Bei der Implementierung der Default-Zeit haben wir leider vergessen das Zeitzonenthema zu berücksichtigen. Momentan verwendet die UI die Variable targetTimeHourSuggestion: Number um die Vorauswahl [...]
Mein Vorschlag wäre, dass wir uns die letzte "aktivierte" Zielzeit (Stunde/Minute) nur im Browser merken (z.B. localStorage) 

Wenn die Zeit gesetzt ist passt sie ja eher zum Standort der WB denn zum Standort des Anwenders. Insofern erscheint mir Backend korrekt.

> und für diese beim nächste Mal wieder anzubieten. 

Das Backend "speichert" ja eh nix, so dass BE nur good-will wäre.

> Der Wert targetTimeHourSuggestion würde dann aus der API verschwinden. Nachteil ist, dass wir dann keinen "Sync" zwischen Devices (Desktop/Phone) haben. Das halte ich aber erstmal für vertretbar.

Full ack, solange das nur die Suggestion und nicht die eingestellte Zielzeit betrifft.

andig, 2022-01-25:
> Bei der Implementierung der Default-Zeit haben wir leider vergessen das Zeitzonenthema zu berücksichtigen. Momentan verwendet die UI die Variable targetTimeHourSuggestion: Number um die Vorauswahl [...]

Soweit ich sehe schickt das UI die Zeit in UTC. Das passt also. "7:00" als lokale Zeit im Browser finde ich auch sinnig. Dass jemand die Wallbox von einem anderen Kontinent aus das Zielladen setzen will ist eher unwahrscheinlich.

Verbleibt noch

>> Der Wert targetTimeHourSuggestion würde dann aus der API verschwinden. Nachteil ist, dass wir dann keinen "Sync" zwischen Devices (Desktop/Phone) haben. Das halte ich aber erstmal für vertretbar.

> Full ack, solange das nur die Suggestion und nicht die eingestellte Zielzeit betrifft.

oder zumachen?

naltatis, 2022-01-30:
Ok, ich fasse kurz zusammen und würde das dann bei Gelegenheit so bauen:

- Zeitvorschlagslogik im Backend wird entfernt
- UI merkt sich die zuletzt konfigurierte Uhrzeit (lokale Zeit des Users) und verwendet die, sofern vorhanden, für den Vorschlag für eine spätere Zielzeitkonfiguration
- Beim Setzten einer neuen Zielzeit wird, wie bislang, eine UTC Zeit aus der UI ans BE geschickt. Das BE muss die Lokalzeit des Nutzers nicht kennen.

closed 2022-02-26 by github-actions[bot]