evcc-io/evcc #2332: Vorklimatisierung wird von PV-Timer vorzeitig abgebrochen

premultiply, 2022-01-17:
### Discussed in https://github.com/evcc-io/evcc/discussions/2331

<div type='discussions-op-text'>

<sup>Originally posted by **pv2ev** January 17, 2022</sup>
Hallo zusammen,
Ich habe heute ein seltsames Verhalten bzgl. Vorklimatisierung beobachtet:
- EVCC stand auf „nur PV“, es war zu dieser Zeit jedoch noch dunkel und dementsprechend kein Ertrag von der PV.
- Per App hab ich in meinem Auto (VW ID4) die Standheizung an gemacht
- In EVCC wurde dies erkannt (Symbol), Strom wurde trotz „nur PV“ freigegeben
- nach wenigen Sekunden begann der Timer zu laufen, dass meine GO-E Wallbox gleich mit der Ladung aufhören wird, da nicht genug PV Strom zur Verfügung steht
- Die Ladung/der Stromfluss wurde letztendlich gestoppt.

Der vorletzte Punkt ist zwar aus Sicht des PV-Überschusses korrekt. Müsste jedoch die Standklima nicht dazu führen, dass KEINE Abschaltung des Stroms erfolgt? Ansonsten macht es keinen Sinn für die kurze Zeit die Wallbox aktiv zu schalten.
Leider hab ich in der Doku dazu nichts gefunden, wie sich EVCC bei Vorklimatisierung verhalten soll.

Vielen Dank :-)</div>

premultiply, 2022-01-17:
So sollte das natürlich nicht sein. Für die Dauer der Vorklimatisierung sollte der Strom eigentlich schon anliegen.
Könntest du ein Logfile von dem Vorgang zur Verfügung stellen? Das wäre für die Analyse wichtig. Danke!

andig, 2022-01-17:
Works as designed.

premultiply, 2022-01-17:
Ausgangssituation: PV-Mode, ungenügende Erzeugung.

Hier gibt es meines Erachtens nach im Code zwei Probleme:
1. Der Zustand des aktiven Climaters wird zwar zum Start erkannt und der Strom freigegeben jedoch fehlt anschließend der erneute Check ob der Climater noch weiter aktiv ist. Dies führt dazu, dass die dazu freigegebene Ladung sofort den PV-Abschalttimer einstartet statt auf das Ende der laufenden Vorklimatisierung zu warten. Hier fehlt bei den Abschaltbedingungen die Prüfung ob der Climator (noch) läuft.
2. Das war noch völlig unbemerkt: Nur im PV-Modus wird die **maximale** Leistung freigegeben statt die sonst (vgl. https://github.com/evcc-io/evcc/blob/606869f3806a700c6c4c9ada70e58f490577f7f7/core/loadpoint.go#L1421-L1427) übliche und angemessene **minimale** Leistung. Hier soll ja möglichst nur etwas Landstrom für die Vorklimatisierung bereitgestellt werden und nicht unbedingt der Akku mit Vollstrom weiter über das Target hinaus oder aus Netz/Hausbatterie gefüllt werden.

Generell: Die Climater-Funktion gibt minimalen Ladestrom frei wenn und während via Vehicle-API eine aktive Klimatisierung erkannt wird obwohl die sonstigen Bedingungen wie Lademodus, Überschussituation oder TargerSoC erreicht dies zu diesem Zeitpunkt regulär nicht zulassen würden. Dies stellt somit einen gewissen temporären Overwrite der Abschaltbedingungen dar.

Der Fix #2333 behebt (hoffentlich) beide Probleme.

pv2ev, 2022-01-18:
> Der Fix #2333 behebt (hoffentlich) beide Probleme.

Vielen Dank! Alles aus deinem Post stimmt so, das kann ich so nachvollziehen. Da ich gerade nochmal klimatisiert habe, anbei das Logfile ab dem Zeitpunkt, wo EVCC `climater active: true` erkannt hat. Ebenso sieht man, dass der PV disable timer zu laufen beginn
[log.txt](https://github.com/evcc-io/evcc/files/7887980/log.txt)
.

RTTTC, 2022-01-18:
Looks like it did not stop the power after PV disable timer ran out?

PR #2333 (2022-01-18): Fix: Check for climater active
Fixes #2332
closed 2022-01-18 by andig