evcc-io/evcc #2349: Fiat 500e: Benachrichtigung in Fiat-App bei SoC Abfrage durch EVCC

Wolfling78, 2022-01-20:
Das ist eher ein Luxus-Problem und hat definitiv keine Priorität, aber....:
Beim Fiat 500e spuckt die Fiat-APP bei jeder Aktualisierung durch EVCC einen Hinweis im iPhone aus und das nervt doch etwas.
Der FIAT-Adapter im ioBroker verursacht diese Meldung nicht - gibt es da verschiedene Zugriffswege auf den SoC? Oder hat Fiat in der jüngeren Vergangenheit etwas auf deren Seite angepasst?

premultiply, 2022-01-20:
Kannst du "einen Hinweis" etwas näher spezifizieren?
Für Leute die noch nie im Leben ein Fiat-App gesehen haben ist das sonst etwas kompliziert nachzuvollziehen.

Wolfling78, 2022-01-20:
Ich habe den text angepasst und hier gibt es noch ein Screenshot dazu:
[image]

andig, 2022-01-20:
@Wolfling78 das passiert immer wenn wir eine Statusaktualisierung anfordern. Verhindern kannst Du das, indem Du die Pin in der Konfiguration weglässt. Es gab aber berichte, dass dann ein veralteter SoC angezeigt wird. Kannst Du das bestätigen?

/cc @DerAndereAndi bzgl. Fiat Doku

Wolfling78, 2022-01-20:
@andig ich habe die in der config gelöscht und poll mode auf always geändert, jetzt bekomme ich den Fehler `ERROR 2022/01/20 09:54:49 vehicle soc: unexpected status: 403`
Im ioBroker ist auch die PIN hinterlegt - ob die dort für die SoC-Abfrage benutzt wird, kann ich aber nicht nachvollziehen.

andig, 2022-01-20:
Das kann ich nicht nachvollziehen:
- Zugriff funktioniert in meinen Tests auch ohne Pin
- ioBroker kann ich mir anschauen. Wir aktualisieren den SoC immer wenn er vom Fahrzeug älter als 5min ist. Das kannst Du über `expiry` ändern.
Wenn ich rein schauen soll schick bitte Zugangsdaten an cpuidle@gmx.de

closed 2022-01-20 by andig