evcc-io/evcc #2361: Innogy Box always switches into "out of service mode"

Flohman, 2022-01-20:
**Describe the bug**
I'm using the innogy eBox Pro and I've hopefully configured everything correctly according to this document:
[https://www.innogy-emobility.com/content/dam/revu-global/emobility-solutions/neue-website-feb-2021/downloadcenter/digital-services/eld_instman_modbustcpde.pdf](https://www.innogy-emobility.com/content/dam/revu-global/emobility-solutions/neue-website-feb-2021/downloadcenter/digital-services/eld_instman_modbustcpde.pdf)

The "Out of service" can be found under "LDP1"->Allgemein->Betriebsbereitschaft->außer Betrieb. The box also has the red lights on in that case. I can set the status back to active and start charging the car and the charging current is changed by evcc depending on the solar power as expected. So, I guess the configuration on the Box as well as in evcc is fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure and start evcc
2. Set Box status to active in menu "LDP1"->Allgemein->Betriebsbereitschaft
3. wait a few minutes and the box is in status out of service and the red lights are on it is not needed to charge the car
3.1. The Box changes status to out of service right after unplugging the cable after charging.

So, basically I always have to activate the box in the innogy Web Gui and start charging right away cause otherwise the box deactivates again after a few minutes. I guess it's obvious that this process is annoying... ;)

**Expected behavior**
Box always remains in service

**Log of the innogy box**

19.01.22 15:03 | 151 | modbus_ | I | 1 | LOAD_ADAPTION\|ldp:<1>\|type:<modbus>\|status:<start>\|current:32
19.01.22 15:04 | 151 | modbus_ | I | 1 | Modbus TCP\|state:<Master/Client is connected>
19.01.22 15:04 | 151 | modbus_ | I | 1 | LOAD_ADAPTION\|ldp:<1>\|type:<modbus>\|status:<start>\|current:6
19.01.22 15:04 | 151 | modbus_ | I | 1 | ModbusSlaveHandler::closeSlaveContext slave is inactive
19.01.22 15:04 | 151 | modbus_ | W | 1 | ModbusSlaveHandler::handleRequest connection gone reason: Connection   reset by peer
19.01.22 15:04 | 151 | modbus_ | E | 1 | Modbus: TransactionException mbSlaveHandler.handleRequest   MODBUS_HANDLE_REQUEST_CONNECTION_LOST
19.01.22 15:04 | 151 | modbus_ | I | 1 | Modbus TCP\|state:<Slave/Server start listening to port: 5555>
19.01.22 15:04 | 151 | modbus_ | I | 1 | Modbus TCP\|state:<Master/Client is connected>
19.01.22 15:05 | 151 | modbus_ | E | 1 | Modbus: TransactionException mbSlaveHandler.handleRequest   MODBUS_HANDLE_REQUEST_CONNECTION_LOST
19.01.22 15:05 | 151 | modbus_ | I | 1 | ModbusSlaveHandler::closeSlaveContext slave is inactive
19.01.22 15:05 | 151 | modbus_ | W | 1 | ModbusSlaveHandler::handleRequest connection gone reason: Connection   reset by peer
19.01.22 15:05 | 151 | modbus_ | I | 1 | Modbus TCP\|state:<Slave/Server start listening to port: 5555>
19.01.22 15:05 | 151 | modbus_ | I | 1 | Modbus TCP\|state:<Master/Client is connected>
2022-01-19 15:05:13.002 | 151 | modbus_ | W | 1 | ModbusSlaveHandler::handleRequest connection gone reason: Connection   reset by peer
2022-01-19 15:05:13.004 | 151 | modbus_ | I | 1 | ModbusSlaveHandler::closeSlaveContext slave is inactive
2022-01-19 15:05:13.007 | 151 | modbus_ | E | 1 | Modbus: TransactionException mbSlaveHandler.handleRequest   MODBUS_HANDLE_REQUEST_CONNECTION_LOST
2022-01-19 15:05:13.026 | 151 | modbus_ | I | 1 | Modbus TCP\|state:<Slave/Server start listening to port: 5555>
19.01.22 15:05 | 150 | userdet | I | 1 | USERMODE\|status:<timeout>
2022-01-19 15:06:13.043 | 151 | modbus_ | I | 1 | LOAD_ADAPTION\|ldp:<1>\|type:<modbus>\|status:<start>\|current:32

premultiply, 2022-01-20:
Probier mal zunächst einen höheren `timeout:` Wert in der Charger-Konfig. Sowas wie 60s.

premultiply, 2022-01-20:
Log file?

andig, 2022-01-20:
@Flohman die Box wurde blind implementiert und nie getestet. Könntest du mir unter info@evcc.io mal Remotezugang einrichten?

Flohman, 2022-01-21:
Meine Box hängt leider nicht im Internet.

Eigentlich kann es ja nur passieren, wenn the Funktion Enable mit false aufgerufen wird. Sonst schreibt ja keine Funktion in das entsprechende Register. Ich weiß jetzt aber nicht, in welchem Prozess die Box disabled werden soll...

andig, 2022-01-28:
> The box also has the red lights on in that case. 

@Flohman Ist das denn ein Problem? Wir verwenden tatsächlich Register `1028` um der Box das Laden zu verbieten. Wenn das keine gute Idee ist- welches Register würdest Du vorschlagen? Die Stromregister erlauben alle nur 6A als Minimum...

andig, 2022-01-30:
ping @Flohman hast Du eine Idee?

Flohman, 2022-01-30:
Ich habe es mir jetzt über IOBroker selbst gebaut. Dabei habe ich rausgefunden, dass die Box auf 0A stellt sobald man versucht sie auf unter 6A zu setzen. Also ich setze alle 3 Phasen auf 0A um das Laden zu verhindern.

andig, 2022-01-30:
Ahh, also entgegen Doku?

Flohman, 2022-01-30:
Genau. 😉

andig, 2022-01-30:
@Flohman wärst Du trotz eigener Lösung so nett und würdest https://github.com/evcc-io/evcc/pull/2445 mal testen? Mangels eigener Hardware bin ich auf Unterstützung angewiesen.

Bzgl. der einzeln konfigurierbaren Phasenströme: kann die Box damit Phasenumschaltung realisieren? Weißt Du ob ein zuschalten der Ströme intern durch CP-Unterbrechung abgesichert wird oder ist vorher ein Deaktivieren notwendig?

closed 2022-01-31 by andig

Flohman, 2022-01-31:
Ja, werde ich testen. Hoffe mal, dass sich bald die Sonne zeigt...