evcc-io/evcc #2362: Missing current power information for innogy box

Flohman, 2022-01-20:
**Is your feature request related to a problem? Please describe.**
During the easy setup process evcc states that the box does not provide any power information and asks if I would like to configure  a separate meter for it. What I don't have ;)

But the box has the information according to this documentation: [https://www.innogy-emobility.com/content/dam/revu-global/emobility-solutions/neue-website-feb-2021/downloadcenter/digital-services/eld_instman_modbustcpde.pdf](https://www.innogy-emobility.com/content/dam/revu-global/emobility-solutions/neue-website-feb-2021/downloadcenter/digital-services/eld_instman_modbustcpde.pdf)

This means that the Power is not shown in the EVCC UI for that loadpoint. Thats why I've configured a custom meter reading the info from the Car's API. But this is lacking several minutes behind and therefore not very useful.
But the current modulation works fine and correctly. The only thing is that the UI visualises e.g. the consumption of the household incorrectly. And I guess the calculation of the savings is wrong in that case.

premultiply, 2022-01-20:
`evcc meter` ?

andig, 2022-01-20:
Die Box kann Ströme, sonst nix. Die lesen wir auch aus.

premultiply, 2022-01-20:
Man könnte mit einer Formal (I1+I2+I3) * 230V was grob in VA ausrechnen was grob in der Größenordnung der Wirkleistung liegt.
Ich glaube das ist deutlich besser als nix.

PR #2364 (2022-01-21): Innogy: Add CurrentPower()
Fix #2362
closed 2022-01-21 by andig