evcc-io/evcc #2389: EVCC with Tesla Powerwall

LeStef101, 2022-01-23:
**Describe the bug**
Some info. about my setup:
Im running EVCC on Kubuntu (installed via apt), inverter is a SolarEdge 10k, Battery a Tesla Powerwall 2 and my charger is a go-e.

I set up my yaml with evcc configure and used the powerwall as meter and battery (which it automatically does when choosing PW as pv). I inserted all relevant information and everything seems to works perfectly fine....for about 5 minutes.
It then stops working with the warning/error/log below.

**To Reproduce**
Steps to reproduce the behavior:
1. EVCC on Kubuntu (installed via apt), Tesla Powerwall 2 as meter and battery.
2. Set up yaml with evcc configure and use the powerwall as meter and battery. 
3. Run EVCC as service or via direct application call.
4. After about 5minutes check out the general log (no changes made to default) or the warnings in the browser

**Expected behavior**
Connection to Powerwall (Gateway) should be kept alive for more then 5 minutes - maybe even until manual service stop ;)

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.81 (8b8153dd)
```

Output of std out/log: info :
```
Request failed with status code 500: API request failed http://192.168.178.54:7070/api/health
updating battery meter 0: unexpected status: 403
updating grid meter: unexpected status: 403
updating pv meter 0: unexpected status: 403
```

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>

uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
- type: template
  template: tesla-powerwall
  usage: grid
  host: 192.168.178.***
  password: ******  # Passwort des Benutzers Kunde
  name: grid1
- type: template
  template: tesla-powerwall
  usage: pv
  host: 192.168.178.***
  password: *****  # Passwort des Benutzers Kunde
  name: pv2
- type: template
  template: tesla-powerwall
  usage: battery
  host: 192.168.178.***
  password: *****  # Passwort des Benutzers Kunde
  name: battery3

chargers:
- type: template
  template: go-e
  host: 192.168.178.**
  name: wallbox5

vehicles:
- type: template
  template: bmw
  title: i3
  user: ****@googlemail.com
  password: ****
  capacity: 17.5
  name: ev4

loadpoints:
- title: Wallbox
  charger: wallbox5
  vehicles:
  - ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3

</pre></details>

andig, 2022-01-24:
403 ist ein Authentifizierungsproblem. Gibts da bei der PW einen Timeout nachdem eine erneute Authentifizierung erforderlich ist? Bitte mach mal für die PW ein `trace` log damit die Details erkennbar werden (Achtung: anonymisieren!).

premultiply, 2022-01-24:
@andig HTTP-Response-Code in Textform ausgeben für bessere Verständlichkeit?

andig, 2022-01-24:
Können wir machen. Merkwürdig ist hier aber warum das anfangs gehen sollte und später nicht mehr?

andig, 2022-01-24:
@LeStef101 könntest Du Dich zu Debugzwecken bitte mal unter info@evcc.io melden?

LeStef101, 2022-01-24:
Kann ich machen.

andig, 2022-01-25:
@LeStef101 ist nix angekommen

andig, 2022-02-02:
Closing due to inactivity

LeStef101, 2022-02-02:
Ich habe dir übrigens 2x auf deine Mail geantwortet und bis jetzt keine Antwort bekommen.

andig, 2022-02-02:
S.o.- nix angekommen?

LeStef101, 2022-02-02:
Ok, neuer Versuch :)

andig, 2022-02-03:
@LeStef101 kannst Du den PR testen? Falls nein würde ich den in die nightly mergen.

LeStef101, 2022-02-03:
Gibts zu dem PR schon nen build auf Github (wovon ich nicht ausgehe^^) oder müsste ich das lokal bauen. Wenn dem so ist, würde ich sagen du mergst und ich teste den nightly build morgen. ;)

andig, 2022-02-04:
Hast Du morgen im Nightly

LeStef101, 2022-02-04:
Habe das über das über das unstable apt repo installiert heute Mittag und der Fehler tritt nach wie vor auf.

andig, 2022-02-05:
Welcher Commit steht zu Beginn im Logfile? Ich bin egtl ziemlich sicher, dass das passen sollte.

LeStef101, 2022-02-05:
Version 0.82.1643941072.1643941072 ; Commit be2f15c7

andig, 2022-02-05:
Warum auch immer- aber das ist genau der Commit vorher, siehe https://github.com/evcc-io/evcc/commits/master. Bitte nochmal aktuelles Nightly nehmen

@DerAndereAndi baut das nightly korrekt?

DerAndereAndi, 2022-02-05:
@andig ja, siehe hier: https://cloudsmith.io/~evcc/repos/unstable/packages/

andig, 2022-02-05:
Wär super wenn wir https://github.com/evcc-io/evcc/issues/2405 umsetzen könnten, dann wäre auch aus dem UI erkennbar, welcher Softwarestand das ist. Die Timestamps helfen ja nur beim Upload, aber haben sonst keine Beziehung zur Software. Was meinst Du?

DerAndereAndi, 2022-02-05:
Aus des Timestamp kannst du dir den letzten Commit erschließen, aber klar ist nicht so schön. Wir brauchen hier halt eine hochzählende "Versionsnummer". Ansonsten unter dem Link erstmal den Timestamp suchen, dann weisste die Build time in lesbarer Form und kommst auf den Commit.

Wenn ich nen Kopf dazu hab, kann ich mir das anschauen.

LeStef101, 2022-02-06:
> Bitte nochmal aktuelles Nightly nehmen. Falls aktuelle Nightly nicht geht nimm bitte auch nochmal `user: customer` mit in die Config auf. Das hat sich durch die verwendete Library ggf. geändert. 

Okay, habe eben gerade aktualisiert und lasse es mal laufen - werde berichten.
Andere Frage: Die Sache mit dem abweichenden SoC des Speichers, hast du dazu auch was gemacht bzw meine Frage dazu gelesen?

andig, 2022-02-06:
Ne- welche Frage?

andig, 2022-02-06:
Habs sie in Deiner Mail gefunden. Mach mal bitte ein —log trace für die powerwall, dann sehen wir ob es evtl. Noch einen Anderen API-Wert für den SoC gibt der besser wiederspiegelt was Du erwarten würdest.

LeStef101, 2022-02-06:
Ok, mache ich,

Erste Beobachtung zum aktuellen Build:
Nach einiger Zeit ist der gleiche "connection refused" Fehler wieder aufgetreten, für eine kurze Zeit, dann kam
` updating grid meter: API call to https://192.168.178.85/api/login/Basic returned unexpected status code 429 (\"{\\\"code\\\":429,\\\"error\\\":\\\"Api Limit reached for this endpoint\\\",\\\"message\\\":\\\"API Limit Reached\\\"}\") `
und danach ging es dann wieder. Das ist zumindest was ich in der GUI über die Warnings angezeigt bekommen habe.

andig, 2022-02-06:
Hast Du mal bitte ein vollständiges Log davon, gerne per Mail?

LeStef101, 2022-02-06:
> Hast Du mal bitte ein vollständiges Log davon, gerne per Mail?

Kann ich gerne machen, allerdings kam diese Meldung oben nur genau einmal (beim ersten Mal) und dann nicht mehr. Das Problem mit dem "connection refused" scheint recht zyklisch alle 40min aufzutreten.
Schicke dir das log in so ner Stunde.

andig, 2022-02-06:
Super. Evtl läuft auch die Lib erstmal auf den Fehler und gibt den 1x zurück. Das wär natürlich blöd aber besser als vorher.

andig, 2022-02-06:
Connection refused ist ein Thema der pw, nicht von evcc.

closed 2022-02-06 by andig