evcc-io/evcc #2404: Smart vehicle API: unexpected status: 400

thierolm, 2022-01-25:
Currently the Smart vehicle API throws an error:

```bash
pi@raspberrypi:~ $ evcc -c ./evcc_modbus.yaml -l debug vehicle
[main  ] INFO 2022/01/25 19:30:34 evcc 0.81 (8b8153dd)
[main  ] INFO 2022/01/25 19:30:34 using config file ./evcc_modbus.yaml
.SoC:     internal_error (Unknown error occurred): unexpected status: 400
Range:    internal_error (Unknown error occurred): unexpected status: 400
Odometer: internal_error (Unknown error occurred): unexpected status: 400
Capacity: 17kWh
```

Smart vehicle config in evcc.yaml (No change done, it worked fine for since today)
```yaml
vehicles:
- name: smart
  type: smart
  title: Smart EQ # display name for UI
  capacity: 17.6 # kWh
  user: ***
  password: ***
  vin: W1A4533XXXXXXX9
```

**EVCC details:**
Show output of <code>evcc -c ./evcc_modbus.yaml -l trace vehicle</code> (last: relevant log
```
...
[smart ] TRACE 2022/01/25 19:34:11 POST https://id.mercedes-benz.com/ciam/auth/login/user
[smart ] TRACE 2022/01/25 19:34:11
POST /ciam/auth/login/user HTTP/1.1
Host: id.mercedes-benz.com
User-Agent: Go-http-client/1.1
Content-Length: 32
Accept: application/json
Content-Type: application/json
Cookie: CIAM.IS=
Accept-Encoding: gzip

{"username":"***"}

HTTP/1.1 200
Transfer-Encoding: chunked
Connection: keep-alive
Content-Security-Policy: frame-ancestors 'none'
Content-Type: application/json
Date: Tue, 25 Jan 2022 18:34:11 GMT
Referrer-Policy: strict-origin-when-cross-origin
Set-Cookie: CIAM.IS=; Domain=id.mercedes-benz.com; Path=/ciam/auth; Secure; HttpOnly
Strict-Transport-Security: max-age=31536000; includeSubDomains
X-Content-Type-Options: nosniff
X-Frame-Options: SAMEORIGIN
X-Permitted-Cross-Domain-Policies: none
X-Request-Id: 01c3c094965498f56355cde10fcddf65
X-Xss-Protection: 1; mode=block

2b
{"country":"DE","hasPass":true,"lang":"de"}
0
[smart ] TRACE 2022/01/25 19:34:11 POST https://id.mercedes-benz.com/ciam/auth/login/pass
[smart ] TRACE 2022/01/25 19:34:11
POST /ciam/auth/login/pass HTTP/1.1
Host: id.mercedes-benz.com
User-Agent: Go-http-client/1.1
Content-Length: 79
Accept: application/json
Content-Type: application/json
Cookie: CIAM.IS=; PF=rNB8UKUCndMZ4Ob8x1MroB
Accept-Encoding: gzip

{"username":"***","password":"***","rememberMe":true}

HTTP/1.1 200
Transfer-Encoding: chunked
Connection: keep-alive
Content-Security-Policy: frame-ancestors 'none'
Content-Type: application/json
Date: Tue, 25 Jan 2022 18:34:11 GMT
Referrer-Policy: strict-origin-when-cross-origin
Set-Cookie: CIAM.REMEMBER_ME=wJiZG6mRyJ+D8dLRPFS74; Max-Age=15552000; Expires=Sun, 24-Jul-2022 18:34:11 GMT; Domain=id.mercedes-benz.com; Path=/ciam/auth/login; Secure; HttpOnly
Set-Cookie: CIAM.IS=; Domain=id.mercedes-benz.com; Path=/ciam/auth; Secure; HttpOnly
Strict-Transport-Security: max-age=31536000; includeSubDomains
X-Content-Type-Options: nosniff
X-Frame-Options: SAMEORIGIN
X-Permitted-Cross-Domain-Policies: none
X-Request-Id: cf071d2e5347fffa94c9f85c14fba225
X-Xss-Protection: 1; mode=block

575
{"result":"RESUME2OIDCP","token":"***"}
0
[smart ] TRACE 2022/01/25 19:34:11 POST https://id.mercedes-benz.com/as/IjNjO/resume/as/authorization.ping
[smart ] TRACE 2022/01/25 19:34:11
POST /as/IjNjO/resume/as/authorization.ping HTTP/1.1
Host: id.mercedes-benz.com
User-Agent: Go-http-client/1.1
Content-Length: 1367
Content-Type: application/x-www-form-urlencoded
Cookie: INGRESSCOOKIE=1643135652.216.84.322748|b8a39a984f03992e66be2258e1e9b103; PF=rNB8UKUCndMZ4Ob8x1MroB
Accept-Encoding: gzip

token=

HTTP/1.1 302 Found
Content-Length: 0
Cache-Control: no-cache, no-store
Connection: keep-alive
Content-Security-Policy: frame-ancestors 'none'
Content-Type: text/html;charset=utf-8
Date: Tue, 25 Jan 2022 18:34:11 GMT
Expires: Thu, 01 Jan 1970 00:00:00 GMT
Location: https://oneapp.microservice.smart.com?code=nGDgo1ZmKCrciGPIiKiL3YeebD4TAn2MyacOFGdR&state=hSw9Aulf6xGd
Pragma: no-cache
Referrer-Policy: strict-origin-when-cross-origin
Set-Cookie: ciamMB=; Path=/; Expires=Wed, 26-Jan-2022 18:34:11 GMT; Max-Age=86400; Secure; HttpOnly; SameSite=Lax
Set-Cookie: PF=rNB8UKUCndMZ4Ob8x1MroBqYWVgdPhQN2xrtLYZB6xCC; Path=/; Secure; HttpOnly; SameSite=Lax
X-Content-Type-Options: nosniff
X-Frame-Options: SAMEORIGIN
X-Xss-Protection: 1; mode=block
[smart ] TRACE 2022/01/25 19:34:11 GET https://oneapp.microservice.smart.com/seqc/v0/vehicles/***/init-data?requestedData=BOTH&countryCode=DE&locale=de-DE
[smart ] TRACE 2022/01/25 19:34:13
GET /seqc/v0/vehicles/***/init-data?requestedData=BOTH&countryCode=DE&locale=de-DE HTTP/1.1
Host: oneapp.microservice.smart.com
User-Agent: Go-http-client/1.1
Accept: */*
Accept-Language: de-DE;q=1.0
Authorization: Bearer 
Guid: 280C6B55-F179-4428-88B6-E0CCF5C22A7C
X-Applicationname: 70d89501-938c-4bec-82d0-6abb550b0825
Accept-Encoding: gzip

HTTP/2.0 400 Bad Request
Content-Length: 71
Cache-Control: no-cache, no-store, max-age=0, must-revalidate
Content-Type: application/json;charset=UTF-8
Date: Tue, 25 Jan 2022 18:34:13 GMT
Expires: 0
Pragma: no-cache
Set-Cookie: JSESSIONID=CCF11E71BC09AA2972F724DE0C176B22; Path=/; HttpOnly
X-Content-Type-Options: nosniff
X-Frame-Options: DENY
X-Xss-Protection: 1; mode=block

{"error":"internal_error","error_description":"Unknown error occurred"}
.SoC:     internal_error (Unknown error occurred): unexpected status: 400
Range:    internal_error (Unknown error occurred): unexpected status: 400
Odometer: internal_error (Unknown error occurred): unexpected status: 400
Capacity: 17kWh

```

andig, 2022-01-25:
Klapps mit dem iobroker Smart Adapter?

thierolm, 2022-01-25:
Hallo @andig , @afuxi hat mir gerade in einer Mail mitgeteilt, dass auch die App ein Problem hat.
Lass uns den Issue mal bis morgen parken ....

Screenshot der App:
[image]

andig, 2022-01-26:
Ich mach mal zu- dürfte am Backend liegen...

closed 2022-01-26 by andig