evcc-io/evcc #2408: Statusabfrage B, C bei Funkschaltern optional unabhängig von der Power

hrpv, 2022-01-26:
_Originally posted by @hrpv in https://github.com/evcc-io/evcc/discussions/2381#discussioncomment-2049610_

Weiterführung der ursprünglichen Diskussion hier:
In der jetzigen Implementierung der Statusabfrage kommt nur dann Status C zurück wenn der Schalter eine Mindestlast schaltet die größer als die konfigurierte standbypower ist. Bei Verwendung für eine leistungslose Ansteuerung z.B. Warmwasseranhebung einer Wärmepumpe funktioniert das nicht, es wird unabhängig von der Schalterstellung immer nur der Status B zurückgeliefert. Da kein Wechsel von B nach C stattfindet funktioniert das messaging nicht. Es gibt auch Funkschalter die gar keine Power zurückliefern, z.B. der weitverbreitete Shelly1, da funktioniert die jetzige Abfrage gar nicht.

Vorschlag: Über eine negative standbypower die Statusabfrage so ändern: 
standbypower >= 0 Statusabfrage wie bisher über die Power . 
standbypower < 0: Schalter Offen Status B , Schalter Zu Status C, 
unabhängig von der Power. Wäre für alle stationären Ansteuerungen wie in meinem Fall perfekt.

https://github.com/thierolm
Ich denke wir brauchen eigentlich keinen zusätzlichen Parameter, die Verwendung der standbypower wäre rückwärtdkompatibel: die bisherigen Anwender haben da sicher einen positiven Wert drinstehen.
Man müsste nur die Doku anpassen:
standbypower: xx # treat as charging above this power, if xx is negative the power of the switch is ignored for charge status.

andig, 2022-01-26:
Das müsste so für Shelly und Fritz implementiert und dokumentiert (!) werden. Da würde mir ein zusätzlicher Parameter (`powerless`?) sogar noch besser gefallen weil explizit.

hrpv, 2022-01-26:
Die Dokumentation wäre dann in etwa so?
powerless : true # optional, if set standbypower is ignored and charge state C is set when switch is ON.

thierolm, 2022-01-27:
Werde versuchen übers WE einen entsprechenden CR für shelly, fritz, tasmota und tplink bereitzustellen.

andig, 2022-01-27:
Vielleicht finden wir noch einen besseren Namen für den Parameter oder beide Parameter.

hrpv, 2022-01-27:
Testing von fritzbox und einem shelly1 kann ich dann gerne übernehmen

thierolm, 2022-01-27:
Shelly1 wird aktuell nicht unterstützt, da er keinen Strom messen kann...
Den entsprechenden check müsste ich ausbauen.

@andig Was meinst du dazu?

thierolm, 2022-01-27:
Erste fritzdect Version siehe #2414

PR #2414 (2022-01-31): FritzDect static mode to provide B (disabled) & C (enabled) status by switch state
First implementation of `powerless` parameter for FritzDect charger in order to support simple heat pump control use cases without metering the power and charged energy.

Other switches (shelly, tasmota,tp-link) will follow, as soon as parameter and behaviour are finally aligned/approved by evcc team.

Fix #2408
closed 2022-01-31 by andig