evcc-io/evcc #2410: Nach SHM 2.0 Update kein Zugriff auf Grid-Meter mehr

premultiply, 2022-01-26:
### Discussed in https://github.com/evcc-io/evcc/discussions/2409

<div type='discussions-op-text'>

<sup>Originally posted by **OlafLDr** January 26, 2022</sup>
Hallo,
seit dem Update auf des Sunny Home Managers auf 2.08.5.R vorgestern funktioniert meine evcc-Konfiguration leider nicht mehr.
Ausschnitt aus der evcc.yaml:

```
site:
- title: Zuhause
  meters:
    grid: sma-grid
    pv: sma-pv
  residualPower: 100

meters:
- name: sma-grid
  type: sma
  uri: 192.168.1.118
- name: sma-pv
  type: sma
  uri: 192.168.1.114
  password: xxxxx
```

Also nichts besonderes. Konfiguration per docker-compose. Bis vorgestern ging es ohne Probleme.
Der log sieht typischerweise folgendermaßen aus:

```
[36mevcc_1  |[0m [main  ] INFO 2022/01/26 20:16:29 evcc 0.81 (8b8153dd)
[36mevcc_1  |[0m [main  ] INFO 2022/01/26 20:16:29 using config file /etc/evcc.yaml
[36mevcc_1  |[0m [main  ] INFO 2022/01/26 20:16:29 listening at 0.0.0.0:80
[36mevcc_1  |[0m [main  ] FATAL 2022/01/26 20:16:31 cannot create meter 'sma-grid': cannot create meter 'sma': login failed: no packet received in timeout
[36mevcc_1  |[0m [main  ] INFO 2022/01/26 20:16:35 evcc 0.81 (8b8153dd)
[36mevcc_1  |[0m [main  ] INFO 2022/01/26 20:16:35 using config file /etc/evcc.yaml
[36mevcc_1  |[0m [main  ] INFO 2022/01/26 20:16:35 listening at 0.0.0.0:80
[36mevcc_1  |[0m [main  ] FATAL 2022/01/26 20:16:38 cannot create meter 'sma-grid': cannot create meter 'sma': failed to get device: no ping response for 192.168.1.118
```

Manueller ping zum SHM 2.0 geht, UDP zum Container auch. Irgendeine Idee, woran das liegen könnte?
Danke und Gruß, Olaf
</div>

premultiply, 2022-01-26:
Hier geht auch ohne Docker nun nichts mehr mit der gleichen neuen Firmware:

```
pi@autosmatix:~ $ sudo evcc meter --log trace
[main  ] INFO 2022/01/26 23:10:25 evcc 0.81.1643164502.1643164502 (b0316105)
[main  ] INFO 2022/01/26 23:10:25 using config file /home/pi/evcc.yaml
[sma   ] TRACE 2022/01/26 23:10:25 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:25 send discover package
[sma   ] TRACE 2022/01/26 23:10:25 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 recv 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 new inverter at 192.168.178.90 - Serial=3006946342
[sma   ] TRACE 2022/01/26 23:10:26 login for 192.168.178.90:9522
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send discover package
[sma   ] TRACE 2022/01/26 23:10:26 login for 192.168.178.90:9522
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send discover package
[sma   ] TRACE 2022/01/26 23:10:27 login for 192.168.178.90:9522
[sma   ] TRACE 2022/01/26 23:10:27 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:27 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:27 send discover package
[sma   ] TRACE 2022/01/26 23:10:27 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[main  ] FATAL 2022/01/26 23:10:27 cannot create meter 'grid1': cannot create meter 'template': cannot create meter 'sma': login failed: no packet received in timeout
```

premultiply, 2022-01-26:
Hier geht auch ohne Docker mit der neuen SHM-Firmware nichts mehr:

```
pi@autosmatix:~ $ sudo evcc meter --log trace
[main  ] INFO 2022/01/26 23:10:25 evcc 0.81.1643164502.1643164502 (b0316105)
[main  ] INFO 2022/01/26 23:10:25 using config file /home/pi/evcc.yaml
[sma   ] TRACE 2022/01/26 23:10:25 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:25 send discover package
[sma   ] TRACE 2022/01/26 23:10:25 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 recv 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 new inverter at 192.168.178.90 - Serial=XXXXXXXXXXX
[sma   ] TRACE 2022/01/26 23:10:26 login for 192.168.178.90:9522
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send discover package
[sma   ] TRACE 2022/01/26 23:10:26 login for 192.168.178.90:9522
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:26 send discover package
[sma   ] TRACE 2022/01/26 23:10:27 login for 192.168.178.90:9522
[sma   ] TRACE 2022/01/26 23:10:27 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:27 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2022/01/26 23:10:27 send discover package
[sma   ] TRACE 2022/01/26 23:10:27 send 192.168.178.90: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[main  ] FATAL 2022/01/26 23:10:27 cannot create meter 'grid1': cannot create meter 'template': cannot create meter 'sma': login failed: no packet received in timeout
pi@autosmatix:~ $
```

andig, 2022-01-26:
/cc @bboehmke @premultiply welche Infos bräuchten wir von SMA?

premultiply, 2022-01-26:
Achtung, jetzt kommt es:
Wenn man im SHM die optionalen Unicast-Meteringinfos abschaltet (bei mir zu allen WR - NICHT zu evcc) funktioniert es wieder.
Häääh?

premultiply, 2022-01-26:
Trägt man wieder Unicast-IPs in die SHM-Konfig ein schlägt es wieder fehl.
Was hat denn SMA da gemacht?

premultiply, 2022-01-26:
Der Workaround ist hier nichts einzutragen:
[image]

premultiply, 2022-01-26:
Ich mache hier erstmal zu bis klar ist ob das tiefgreifender neuer SMA-Bug ist oder wir hier noch etwas tun können oder müssen.
Derzeit sieht es nach SMA-Bus aus.

closed 2022-01-26 by premultiply

Hamudulu, 2022-05-06:
Hallo, wo finde ich im SHM 2.0 diese Einstellungen zum raus machen? Ich finds nicht...
Danke schon mal