evcc-io/evcc #2429: cannot create vehicle: form not found

PaddyX, 2022-01-29:
Seid gestern Abend bekomme ich den Status meines Fahrzeugs nicht mehr abgefragt. Ich bin ratlos. Die Skoda App funktioniert mit den gleichen Login-Daten tadellos. Ich habe von evcc 0.74 und die neue Version 0.81 ausprobiert. Gleiches Verhalten.

Irgendeine Idee? Hat VW ggf. die API geändert?

`
./evcc.new -l debug vehicle
[main  ] INFO 2022/01/29 10:53:25 evcc 0.81 (8b8153dd)
[main  ] INFO 2022/01/29 10:53:25 using config file /srv/smarthome/evcc/evcc.yaml
[mqtt  ] INFO 2022/01/29 10:53:25 connecting evcc-1841665444 at tcp://localhost:1883
[mqtt  ] DEBUG 2022/01/29 10:53:25 tcp://localhost:1883 connected
.SoC:     vehicle not available: cannot create vehicle 'enyaq': login failed: form not found
Capacity: 0kWh
`

Anbei das Trace im Anhang. 

[evcc_trace.log](https://github.com/evcc-io/evcc/files/7963427/evcc_trace.log)

thomaskien, 2022-01-29:
bei mir bestehrt das selbe problem, till steinbach hat sein projekt diesbezüglich auch schon gestern angepasst, bei mir läuft auch das aber leider nicht aktuell (zumindest für die von mir benutzten belange). lästig dass VW alle fünf minuten da irgendwas(TM) ändert.

thomaskien, 2022-01-29:
carnet_cli.py ist auch kaputt, vermutlich also auch alle projekte die auf der selben library laufen

PaddyX, 2022-01-29:
Danke fürs schnelle Feedback. Ich hatte gestern Abend evcc auf 0.81 aktualisiert und genau in dem Moment ging nichts mehr.

andig, 2022-01-29:
https://github.com/evcc-io/evcc/pull/2430

andig, 2022-01-29:
> bei mir bestehrt das selbe problem, till steinbach hat sein projekt bezüglich der änderungen bei VW auch schon gestern angepasst,

Wer ist till steinbach? Welches projekt?

andig, 2022-01-29:
> carnet_cli.py ist auch kaputt, vermutlich also auch alle projekte die auf der selben library laufen

Nein. Alle Projekte die sich bei VW anmelden wollen.

closed 2022-01-29 by andig

F-L-0, 2022-01-29:
@andig 
> Wer ist till steinbach? Welches projekt?

https://github.com/tillsteinbach/VWsFriend

andig, 2022-01-29:
Kannte ich nicht, danke 🙏🏻