evcc-io/evcc #2442: availableVersion no longer updated via MQTT

blenni, 2022-01-30:
**Describe the bug**
Until v0.79 availableVersion was provided through MQTT and I used it to send some notification that a new version of evcc is available.

Was it removed on purpose?

andig, 2022-01-30:
M.E. hat sich da nix geändert.

emcepe, 2022-11-18:
Mir fällt das auch auf, seit einigen Monaten wird availableVersion nur sporadisch aktualisiert, gleiches gilt für die GUI, die kein Update angezeigt hat, weder dass v0.106.5 noch v0.107.0 verfügbar war.
Ich hab's nur durch Zufall gesehen, dass v0.107.0 verfügbar ist und dann aktualisiert. Aktuell zeigt availableVersion an, dass v0.106.4 verfügbar ist.

Irgendwo scheint da noch ein Bug zu sein :)

andig, 2022-11-20:
@naltatis 

Die “avialableVersion” (im Socket API bisher unterdrückt) scheint es nicht mehr ins UI in den Footer zu schaffen. Auch wenn ich die Unterdrückung raus nehme taucht die neue Version im Footer nicht auf. Mir ist grad auch nicht klar, wie es die Release Notes eigentlich ins Gui schaffen falls ein Update vorhanden ist. Kann aber auch sein, dass wir das ohnehin nur bei Gokrazy hatten?

andig, 2022-11-20:
@blenni @emcepe ich kann das tatsächlich nicht nachvollziehen. Aber Achtung: auf neues Release wird erst (!) nach und dann alle nach Programmstart gecheckt. Das soll verhindern, dass laufende Neustarts das Github Rate Limit strapazieren. In meinen Tests (wenn ich das auf "sofort" ändere) wird auch die neue Version zuverlässig erkannt.

Ich sehe keinen aktuellen Fehler!

emcepe, 2022-11-20:
Hi @andig,

> auf neues Release wird erst (!) nach und dann alle nach Programmstart gecheckt.

was heißt dieser Satz? :)

Ich verstehe ihn nicht.

andig, 2022-11-20:
Ähhhh, da ist was verschwunden- korrigiert

closed 2022-11-20 by andig

blenni, 2022-11-20:
Ich kann auch keine Probleme mehr erkennen, nur das eine Mal bei Meldung im Januar.

Jetzt funktionierts bei mir problemlos, auch z.B. von 106.5 auf 107

emcepe, 2022-11-20:
Ok, das versteht man nun :)

Solche Bugs sind "toll" ;-( ... sehr merkwürdig.

Wie gesagt, die availableVersion stand bei mir seit Release von 0.106.4 auf 0.106.4, auch als .5 schon draußen war, auch als ich auf 0.107.0 aktualisiert habe. Erst heute Nacht um 6:00 Uhr kam die Info, dass 0.107.1 verfügbar ist, passt in dem Fall also (6h nach Restart), da ich jede Nacht um 0:00 Uhr einen Neustart von evcc mache bzw. machen lasse.

Was wird da dann Code-intern geprüft? URL? Evtl. aktualisiert sich diese nicht immer wie gewünscht?
Alles nur Vermutungen.

andig, 2022-11-25:
Ich hoffe, dass es ab nightly tut...

emcepe, 2022-12-07:
ich muss Dich leider enttäuschen, es funktioniert nicht korrekt - benutze aktuell 0.108.3

evcc zeigt bei mir aktuell availableVersion 0.108.0 an (aktuell ist jetzt: 07.12.2022 um 17:03 Uhr)

Restart passiert wie gesagt 1x am Tag nachts um 0 Uhr, insofern hätte evcc nach den jeweiligen 6 Stunden schon irgendwas nach 0.108.0 mitbekommen müssen. Das gleiche gilt für die GUI, da poppen ebenso keine Update Infos auf.

Ich hab' mir jetzt erstmal ein DOIF in FHEM gebaut, was Github abfragt und mich informiert, sobald ein Update am Start ist.

emcepe, 2022-12-07:
ok, mach ich. Melde mich dann zurück mit Log.

emcepe, 2022-12-08:
hmm leider sehe ich im Log nichts was damit zu tun haben könnte, nur solche Sachen:

```
evcc[13687]: [http   ] TRACE 2022/12/08 00:00:23 GET https://raw.githubusercontent.com/neoPix/bluelinky-stamps/master/hyundai-bla.v2.json
evcc[13687]: [hyundai] TRACE 2022/12/08 00:00:24 GET https://eu-account.hyundai.com/auth/realms/euhyundaiidm/login-actions/required-action?execution=hyundai-account-find-link&client_id=bla&tab_id=blub
evcc[13687]: [hyundai] TRACE 2022/12/08 00:00:25 GET https://prd.eu-ccapi.hyundai.com:8080/...
```

emcepe, 2022-12-08:
ok, dann warte ich ab, bis ein neues Release mit dem Fix erscheint und teste dann erneut.

emcepe, 2022-12-09:
könntet ihr das evtl. so erweitern bzw. abändern, daß nach einem Neustart von evcc direkt gesucht wird und dann alle 6 Stunden?

andig, 2022-12-09:
Hab ich geändert. Wenn es Fehler durch Github rate limiting gibt kannst Du `GITHUB_TOKEN` als Environmentvariable oder in einer `.env` Datei angeben.

emcepe, 2022-12-09:
Dankeschön :)

Witzig: heute um 19:11 Uhr kam eine Info via availableVersion, dass 0.109.1 verfügbar ist. Installiert ist 0.109.2 ...

andig, 2022-12-10:
...und gehts jetzt?

emcepe, 2022-12-10:
Ich warte noch aufs nächste Release wo der Fix fürs GitHub Logging mit drin ist :)

emcepe, 2022-12-20:
> Autsch, im nightly behoben. Es braucht `main: trace` um die Requests zu sehen.

```
Dec 20 22:49:52 raspberry-pi-4b evcc[5424]: [main  ] TRACE 2022/12/20 22:49:52 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
Dec 20 22:49:53 raspberry-pi-4b evcc[5424]: [main  ] TRACE 2022/12/20 22:49:53 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/86697662","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/86697662/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/86697662/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.110.0","id":86697662,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"node_id":"RE_kwDODX4bUs4FKua-","tag_name":"0.110.0","target_commitish":"master","name":"0.110.0","draft":false,"prerelease":false,"created_at":"2022-12-20T08:12:56Z","published_at":"2022-12-20T20:37:30Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/88938214","id":88938214,"node_id":"RA_kwDODX4bUs4FTRbm","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"text/plain; charset=utf-8","state":"uploaded","size":754,"download_count":0,"created_at":"2022-12-20T20:49:27Z","updated_at":"2022-12-20T20:49:27Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.110.0/checksums.txt"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/88938211","id":88938211,"node_id":"RA_kwDODX4bUs4FTRbj","name":"evcc_0.110.0_amd64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","size":17504754,"download_count":0,"created_at":"2022-12-20T20:49:26Z","updated_at":"2022-12-20T20:49:27Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.110.0/evcc_0.110.0_amd64.deb"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/88938212","id":88938212,"node_id":"RA_kwDODX4bUs4FTRbk","name":"evcc_0.110.0_arm64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","size":15986060,"download [... cut ...] ts_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"application/gzip","state":"uploaded","size":16838980,"download_count":0,"created_at":"2022-12-20T20:49:23Z","updated_at":"2022-12-20T20:49:24Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.110.0/evcc_0.110.0_linux_
```

mehr kommt da nicht. availableVersion aktualisiert sich nicht.