evcc-io/evcc #2466: Incorrect charge phase indication

mclane, 2022-02-02:
**Describe the bug**
Charge phase indicator indicates 2P or 3P even if my car is configured and can charge only with 1P
 
**To Reproduce**
start evcc
set PV only mode
connect Kia eSoul (with 1p charger)
issue appears specifically often when PV power is changing quickly (clouds, like today in Bavaria)

**Expected behavior**
Phase indicator stays at 1P

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.82 (fbd0d655)

```  
evcc config file:
```uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: trace
levels:
  core: debug
  lp-1: debug

# mqtt message broker
mqtt:
  broker: 192.168.0.38:1883
  topic: evcc # root topic for publishing, set empty to disable
  # user: guest
  # password: guest

# influx database (v1)
influx:
  url: http://localhost:8086
  database: evcc
  user: evcc
  password: evcc

# energy cost
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.31
  feedin:
    type: fixed
    price: 0.14

# meters
meters:
- name: grid
  type: custom
  power: # power reading
    source: mqtt # use mqtt
    topic: sdm630/sdm1-1/Power # mqtt topic
    timeout: 10s # don't use older values
- name: pv
  type: custom
  power:
    source: mqtt # use mqtt
    topic: sdm630/sunspec3-1/Power # mqtt topic
    timeout: 10s # don't use older values
- name: charge
  type: custom
  power: # power reading
    source: mqtt # use mqtt
    topic: sdm630/sdm722-3/Power # mqtt topic
    timeout: 10s # don't use older values
  energy: # energy reading
    source: mqtt # use mqtt
    topic: sdm630/sdm722-3/Sum # mqtt topic
    timeout: 10s # don't use older values

chargers:
- name: phoenix
  type: phoenix-em-eth # Charger with Phoenix Contact controller
  uri: 192.168.0.225:502 # ModBus address
  id: 180

vehicles:
- name: kia
  type: kia
  title: eSoul
  user: uli.schrey@web.de # user
  password: x3TB57ka! # password
  capacity: 64 # kWh
#  cache: 60m
#- name: kia
#  type: custom
#  title: eSoul
#  capacity: 64
#  soc:
#    source: script
#    cmd: /bin/sh -c "echo -1" 
#    source: mqtt # use mqtt
#    topic: evcc/test/soc # mqtt topic
#    timeout: 40s # don't use older values
  
site:
  title: Altenburg # display name for UI
  meters:
    grid: grid # grid meter
    pv: pv # pv meter
    
loadpoints:
- title: Garage # display name for UI
  charger: phoenix # charger
  meters:
    charge: charge # charge meter
  vehicle: kia
  mode: pv
  soc:
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    min: 0
    target: 100
    #alwaysUpdate: false # set true to update vehicle soc even when disconnected
    estimate: true # set true to interpolate between api updates
  onDisconnect:
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge fully
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 20 # maximum charge current (default 16A)

```

andig, 2022-02-02:
Das Passwort Deines Autos ist verbrannt- undbedingt ändern!

mclane, 2022-02-02:
Verdammt! Danke für den Hinweis, da hab ich gepennt. Passwort ist geändert und oben gelöscht. Was fehlt noch an Infos?

andig, 2022-02-02:
Das passende Logfile damit ich den Auslöser finden kann- bitte ab Start.

mclane, 2022-02-02:
siehe syslog anbei. Erstes Auftreten um 11:14:10 (1p -> 2p)
[syslog.log](https://github.com/evcc-io/evcc/files/7986893/syslog.log)
)

andig, 2022-02-02:
/cc @premultiply das ist wieder das Problem aus https://github.com/evcc-io/evcc/issues/2177#issuecomment-1003709301:

6.0A @ 2784W ergibt 1,96...

@mclane hast Du da irgendwo einen Ladezähler eingebaut? Falls nein ist das im Prinzip zu erwarten...

mclane, 2022-02-02:
Ja, ein Ladezähler ist vorhanden (siehe loadpoints in evcc.yaml). Die Daten laufen über MBMD und MQTT wie alle Zählerdaten (auch die vom PV Wechselrichter); sie sollten deshalb zeitlich halbwegs konsistent sein.

andig, 2022-02-02:
Also wenn Du einen Ladezähler hast (nur Leistung, keine Ströme?) dann frage ich mich wie ich 6A bei 2,8kW anders als 2P interpretieren soll?

andig, 2022-02-02:
Dann können wir zumachen. Es gibt nichts was evcc anhand der vorhandenen Informationen anders machen könnte.

closed 2022-02-02 by andig

premultiply, 2022-02-02:
@andig Ich würde diese „Phasenschätzung“ mittels Leistung gerne ganz loswerden, da sehr fehlerhaft und stattdessen durch ein IMHO besseres Modell ersetzen.
Dazu dann mehr auf der Tonspur.