evcc-io/evcc #2467: TP-Link charger template error

xpcone, 2022-02-02:
Hi,
eine TP-Steckdose lässt sich nicht korrekt mit evcc configure einbinden:

##### erzeugte Config:
```
chargers:
- type: template
  template: tplink
  host: 192.168.X.XX  # IP-Adresse oder Hostname
  standbypower: 15  # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet
  name: wallbox4
```
##### Fehlermeldung:
```
Testing the configuration of TP-LINK Smart Plug [switchable sockets] ...
   Error: Get "192.168.X.XX/cm?cmnd=Status+8&password=&user=": unsupported protocol scheme ""
```

##### funktionierende Config:
```
chargers:
- type: tplink
  uri: 192.168.X.XX  # IP-Adresse oder Hostname
  standbypower: 15  # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet
  name: wallbox4
```

thierolm, 2022-02-02:
Das Template nutzt `host` anstatt `uri` ...

andig, 2022-02-02:
@thierolm der Host kommt ja an. Fehlt evtl „defaultscheme“ in der Implementierung?

andig, 2022-02-02:
Ich versteh die ganze Fehlermeldung nicht. Wo kommt hier überhaupt ein GET Request her? Der TPLink macht doch TCP und nicht HTTP?

andig, 2022-02-02:
@DerAndereAndi ich habs gefunden:

    render: |
      type: tasmota
      uri: {{ .host }}
      standbypower: {{ .standbypower }}  # treat as charging above this power

Tasmota kann nicht passen ;)

closed 2022-02-03 by andig