evcc-io/evcc #2469: Funkschalter, Energie wird nicht von separatem charge meter genommen?

andig, 2022-02-02:
### Discussed in https://github.com/evcc-io/evcc/discussions/2463

<div type='discussions-op-text'>

<sup>Originally posted by **hrpv** February  1, 2022</sup>
@andig 
Beim Nachtesten (0.82.1643682716) #2414 des fritzdect schalters, 
mit standbypower auf -1, state B/C funktioniert jetzt wie vorgesehen, **Danke dafür**, ist mir folgendes aufgefallen:
Die Ladeanzeige bleibt bei 0 Wh stehen, während die Ersparnisanzeige korrekte Werte liefert.
Die Ladezeit springt zwischen realen 25 Minuten und mehreren 1000 h.
Ich habe ja wie schon erwähnt ein eigenes charge meter im  Loadpoint definiert,
 das (z.Zt.) aber nur die Leistung liefert. Kann es sein dass die evcc Software die Energie fälschlicherweise noch vom Funkschalter holt und nicht selber aus der Leistung des charge meters errechnet?
Die Ersparnisanzeige (Das „Sonnen-Popup“) dagegen scheint die Werte korrekt zu berechnen.
Verständnisfrage zum GUI: 
die Anzahl Phasen scheint jetzt dynamisch zwischen 1P … 3P zu springen
je Leistungsaufnahme der WP ist alles zwischen 50 W und 6Kw möglich. 

Update: Hole die Werte für den WP Zähler (charge meter) jetzt über MQTT von vzlogger, liefert jetzt sowohl Leistung als auch Energie. Ändert leider nichts das Problem besteht weiter. **Nur wenn ich einen Heizlüfter direkt in der FritzDect einstecke bewegt sich die Wh Anzeige beim Laden korrekt. d.h. die Energie kommt nicht vom wp meter sondern von vom Funkschalter wie vermutet**
Habe die Modi sofort und nur PV getestet
Habe testweise einen 2kW Heizlüfter statt der WP (WP ausgeschaltet) als fixe Last dran, die Anzeige beim Laden klebt fest bei 0 oder manchmal auch 1 Wh. Mit 2 KW Last hätte ich nach 10 min Laden ca. 330 Wh in der Anzeige erwartet. Die Ersparnis-Anzeige rechnet korrekt, die zeigt nach 14 min 0.5 kWh an (rechnerisch 0.46kwh)
evcc meter `-`c evcc.yaml:
wpmeter
Power:  1991W
Energy: 21473.3kWh  (Zählerdisplay 21473)
mosquitto_sub -t vzlogger/data/chn0/agg | jq .value  21473374.200000003  (liefert Werte in wh)

```yaml
#Wärmepumpe meter, bedient von vzlogger -> mqtt ohne middleware alle 10 s
- name: wpmeter
  type: custom
  power:
    source: mqtt
    broker: 127.0.0.1:1883
    topic: vzlogger/data/chn1/agg
    jq: .value
    timeout: 60s # don't accept values older than timeout
  energy: 
    source: mqtt
    broker: 127.0.0.1:1883
    topic: vzlogger/data/chn0/agg
    jq: .value
    scale: 0.001
    timeout: 60s # don't accept values older than timeout
````

[image]

[image]

</div>

andig, 2022-02-02:
@hrpv hast du bitte ein Logfile in dem ich die Werte sehe? Laut Code scheinen wir den richtigen Zähler zu nehmen, deshalb verstehe ichs nicht!

hrpv, 2022-02-02:
Hallo Andi,
hier ist der Logfile, 10 Minuten Ladezeit Modus Sofortladen Aber: im Moment sind es nur ca. 60 W Last, ich kann das Szenario mit dem Heizlüfter erst in ein paar Tagen wieder darstellen (wohne dort nicht) Interessanterweise sind noch die 0.5 kWh von den Tests mit dem Heizlüfter von heute gespeichert, wie kann ich die zurücksetzen? Browser Cache löschen hat nicht geholfen.
Habe mal die Zählerstände von wpmeter und fritzbox vor start von evcc ausgelesen und kurz danach
Der Energiezähler Fritz Dect hat sich erwartungsgemäss nicht verändert, 
der WP meter hat 34 Wh weitergezählt
Fritz Dect Switch Energy [Wh]: 1393 1393
WP Meter  [Wh] 21474394.9  21474428.9 

[evcc_debug.log](https://github.com/evcc-io/evcc/files/7989784/evcc_debug.log)

[image]

andig, 2022-02-02:
Ich versteh nur Bahnhof. Welcher Zähler hat die richtigen Werte? So sehe ich das im Logfile? Was hat falsche Leistung mit Energie zu tun?

andig, 2022-02-02:
@thierolm ich habe zwei Fehler gefunden:
- im Loadpoint nutzte ich den `api.ChargeRater` der Wallbox wenn sie einen hat (=der Fritzdect) -> mein Fehler
- die FritzDect (und TPLink und...) dürfte das aber gar nicht haben, da sie keine Ladesessions zählt sondern nur Gesamtenergie (=`api.TotalEnergy`). -> könntest Du das anschauen?

hrpv, 2022-02-02:
Der WP Meter hat die richtigen Werte: 60W und 21474394.9 Wh
Die geladene Energie 0.5 kWh ist noch von heute mittag gespeichert als ich einen 2 kW Heizlüfter in den Fritz Dect eingestecjt habe, wie bekomme ich die wieder gelöscht?
 [lp-2  ] DEBUG 2022/02/02 20:45:47 charge power: 60W
[lp-2  ] DEBUG 2022/02/02 20:45:49 charger status: C

andig, 2022-02-02:
Ok, Fehler gefunden. Tricky...

hrpv, 2022-02-02:
@thierolm Sorry wenn ich nerve:  kann das sein, dass die gespeicherten 
0.5 kWh die ich nicht mehr löschen kann direkt aus dem Fritz Dect Schalter ausgelesen werden werden?
	**// select energy value of current day**
besonders transparent ist das nicht gerade....

var _ api.ChargeRater = (*FritzDECT)(nil)

// ChargedEnergy implements the api.ChargeRater interface
func (c *FritzDECT) ChargedEnergy() (float64, error) {
	// fetch basicdevicestats
	resp, err := c.fritzdect.ExecCmd("getbasicdevicestats")
	if err != nil {
		return 0, err
	}

	// unmarshal devicestats
	var stats fritzdect.Devicestats
	if err = xml.Unmarshal([]byte(resp), &stats); err != nil {
		return 0, err
	}

	**// select energy value of current day**
	if len(stats.Energy.Values) == 0 {
		return 0, api.ErrNotAvailable
	}

andig, 2022-02-02:
Warte einfach mal auf die Änderung, jetzt nixht neue Themen aufmachen die nicht dazu gehören ;)

thierolm, 2022-02-03:
Moin, ich bin davon ausgegangen, dass die api Funktion ChargedEnergy, die über den Charger geladene Energy für einen bestimmten Zeitraum sammelt. Im Falle der Smart Switches habe ich dafür die von den Dosen gemessene Tagesenergie genommen. Suggeriert so die Übersetzung ...

@andi hat die Funktionen in seinem PR ja schon auf TotalEnergy renamed ... (Tasmota fehlt noch - ich habe unter der Woche  nicht viel Zeit ...)

andig, 2022-02-03:
@hrpv kannst du morgen im nightly ausprobieren.

closed 2022-02-03 by andig

hrpv, 2022-02-03:
> Moin, ich bin davon ausgegangen, dass die api Funktion ChargedEnergy, die über den Charger geladene Energy für einen bestimmten Zeitraum sammelt. Im Falle der Smart Switches habe ich dafür die von [...]

Alles gut, ich war gestern nur total irritiert beim Logfile erstellen warum trotz Neustart von evcc immer noch 0.5 kWh bei geladene Energie stand. Habe mir jetzt mal die tages xml zeile von meinem Fritz Dect Switch geholt, die 549 Wh von gestern waren die paar Minuten Heizlüfter. Heute morgen steht wieder 0 da. Eigentlich ne clevere Idee , hat jeder smarte Schalter so eine Statistik (ausser Shelly1)?
```xml
<stats count="31" grid="86400">0,549,0,1,1,0,...</stats>
```

hrpv, 2022-02-04:
Nightly (0.82.1643941072) Anzeige "Geladen" sieht gut aus 👍 
ich danke euch! Telegram Message jetzt auch ok: Finished "Wärmepumpe" charging **0.4**kWh in 7m38s.
Was geblieben ist die springende Anzeige bei "Dauer" zwischen Ladezeit und vermutlich Restzeit. Kann das sein dass da was in der GUI verrutscht ist?
[image]
[image]

andig, 2022-02-04:
Auf die Dauer kann ich mir keinen Reim machen. Kannst Du bitte nochmal neues Issue mit Logfile- level trace für FritzDECT und Zähler- machen?

thierolm, 2022-02-04:
Was hast du denn als Capacity für die Wärmepumpe definiert?
Das Problem ist doch nicht die springende Dauer, sondern einmal 99Wh und dann 0,2kWh (200Wh) als Teiler durch die Capacity ...
Woher kommen die beiden unterschiedlichen Werte im Wechsel?

andig, 2022-02-04:
Neues Thema!

hrpv, 2022-02-04:
> Was hast du denn als Capacity für die Wärmepumpe definiert? Das Problem ist doch nicht die springende Dauer, sondern einmal 99Wh und dann 0,2kWh (200Wh) als Teiler durch die Capacity ... Woher kommen die beiden unterschiedlichen Werte im Wechsel?

Lass dich nicht von den 99Wh und 0.2 Wh irritieren, die Screenshot sind ca. 2 Minuten auseinander. Es ist gar nicht so einfach die Shots zu kriegen weil die meiste Zeit steht die Stundenanzeige (rot) da und dann blitzt für nur für wenige Sekunden die echte Dauer (grün). Evcc war neu gestartet alles war Anfang 0, bei Dauer und Restzeit  `---`

Dummy-Auto für die Wärmepumpe
```yaml
- name: wpauto             # Konfigurationsname
  type: custom
  title: WP # display name for UI
  capacity: 100 # byttery capacity (kWh)
  onIdentify: # set defaults when vehicle is identified
    minSoC: 20 # charge to at least 20% independent of charge mode
    targetSoC: 95 # limit charge to 90%
  soc:  # battery soc (%)
     source: script # plugin type
     cmd: /bin/sh -c 'echo 35'
     timeout: 5s
```
LoadPoint:
```yaml
- title: Wärmepumpe # display name for UI
##  charger: wpcustom # charger
  charger: wpcharger
  mode: pv
  resetOnDisconnect: true # set defaults when vehicle disconnects
  meters:
    charge: wpmeter
  vehicle: wpauto
  soc:
    poll:
      mode: charging
      interval: 10m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: false # set true to interpolate between api updates
  enable: # pv mode enable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 300 # maximum import power (W)
  guardduration: 20m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 7 # maximum charge current (default 16A)
  phases: 2 # ev phases (default 3)
```