evcc-io/evcc #2510: charger out of sync: expected enabled, got disabled // Easee Wallbox

tmy0, 2022-02-06:
### Discussed in https://github.com/evcc-io/evcc/discussions/2477

Hallo,
wie in der discussion schon beschrieben, kommt es vor, dass meine WB das Auto im Modus min+pv irgendwann nicht mehr lädt. Ich habe ein ausführliches Log erstellt:
[evcc.log](https://github.com/evcc-io/evcc/files/8010234/evcc.log)

Hier sind mir ein paar Dinge aufgefallen, die für mich unplausibel sind.
Wie kann es sein, dass die grid power: 6143W ist und charge power: 9853 soweit auseindander gehen. Zumal zuvor auch nicht so viel Strom produziert wurde. Hier der Bereich im Log dazu.

Feb 06 11:03:20 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:20 grid power: -3930W
Feb 06 11:03:20 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:20 pv power: 5257W
Feb 06 11:03:20 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:20 site power: -3930W
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 charge power: 158W
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 charge currents: [0.01 0.01 0.009]A
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 charger status: C
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] INFO 2022/02/06 11:03:20 start charging ->
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 vehicle soc: 47%
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 vehicle range: 114km
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 vehicle odometer: 2865km
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 max charge current: 19.1A = 2.01A + 17.1A (-3930W @ 1p)
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 charge power: 158W
Feb 06 11:03:20 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:20 charge currents: [0.01 0.01 0.009]A
Feb 06 11:03:30 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:30 ----
Feb 06 11:03:30 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:30 ----
**Feb 06 11:03:30 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:30 grid power: 6143W
Feb 06 11:03:30 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:30 pv power: 5218W
Feb 06 11:03:30 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:30 site power: 6143W
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 charge power: 9853W**
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 charge currents: [14.8 14.2 14.1]A
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 detected phases: 3p [14.8 14.2 14.1]A
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 charger status: C
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 next soc poll remaining time: 59m49s
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 vehicle soc: 47%
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 vehicle range: 114km
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 vehicle odometer: 2865km
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 max charge current: 7.1A = 16A + -8.9A (6143W @ 3p)
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 available power below 3p min threshold of 4140W
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 start phase disable timer: 0s
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 phase action: scale1p in 0s
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 phase disable timer elapsed
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 charger disable
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 pv timer elapse
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 pv action: inactive
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 switched phases: 1p @ 3710W
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 climater active: false, target temp: 20.0°C, outside temp: 20.0°C
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 charge power: 9853W
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 charge currents: [14.8 14.2 14.1]A
Feb 06 11:03:30 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:03:30 detected phases: 3p [14.8 14.2 14.1]A
Feb 06 11:03:40 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:03:40 ----

Warum wurde nach 11:15:10 nicht weiter geladen? Warum wurde trotz des Lademodus min+pv das Laden immer wieder gestoppt?

closed 2022-02-06 by andig

tmy0, 2022-02-06:
Hi, warum wird der issue geschlossen? Welche Infos werden benötigt?

andig, 2022-02-06:
Alle angefragten Informationen werden benötigt, deshalb werden sie angefragt. Dann würde es total helfen zu jeder deiner Fragen den Zeitpunkt zu nennen.