evcc-io/evcc #2519: Easee wallbox stoppt Ladung nach Phasenumschaltung

hanzoh, 2022-02-07:
**Describe the bug**
Das Laden startet bei "Min+PV" 3-phasig, da genug Überschuss vorhanden.
Dann sinkt der PV Output und es erfolgt ein Wechsel auf 1P.
Es erscheint die Warnung ```ignoring inconsistent phases: 1p < 3p observed active``` und die Ladung wird nicht wieder gestartet.

Zum Log unten folgende Zeiten, wann der Modus umgestellt wurde:

12:19:35 STOP
12:20:35 Min+PV

**To Reproduce**
Steps to reproduce the behavior:
1. Starte Ladung mit Min+PV
2. PV Output sinkt unter 3 kW
3. Wallbox bekommt Anweisung 3P->1P
4. Wallbox stoppt
5. Warnung ```ignoring inconsistent phases: 1p < 3p observed active``` erscheint

**Expected behavior**
Nach dem Stopp sollte die Ladung mit 1P fortgesetzt werden.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.82 (fbd0d655)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
config
======

grid: grid3
-----------
Power: -31W

pv 0: pv1
---------
Power: 2097W

battery 0: battery2
-------------------
Power: -277W
SoC:   17%

loadpoint 1
===========

charger: wallbox5
-----------------
Power:          1312W
Current L1..L3: 5.71A 0.008A 0.008A
Charge status:  C
Enabled:        true
Charged:        1.1kWh

vehicle 0
---------
SoC:            58%
Charge status:  C
Range:          189km
Odometer:       not available
Finish time:    2022-02-08 07:29:00 +0100 CET
Climate active: false
Outside temp:   19.5°C
Target temp:    19.5°C
Capacity:       58kWh
OnIdentified:   TargetSoC:100, MinCurrent:6, MaxCurrent:16
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: trace #info
levels:
  modbus: debug

meters:
- type: custom
  name: pv1 
  power:
    source: calc
    add:
    - source: modbus
      model: sunspec
      value: 160:1:DCW # SunSpec Model 160 MPP string 1 DCW
      uri: 192.168.178.46:1502
      id: 71
    - source: modbus
      model: sunspec
      value: 160:2:DCW # SunSpec Model 160 MPP string 2 DCW
      uri: 192.168.178.46:1502
      id: 71
- type: template
  template: kostal-plenticore
  usage: battery 
  modbus: tcpip 
  name: battery2 
  id: 71 
  host: 192.168.178.46 
  port: 1502
- type: template
  template: kostal-ksem-inverter
  usage: grid 
  modbus: tcpip 
  name: grid3 
  id: 71 
  host: 192.168.178.46 
  port: 1502

chargers:
- type: template
  template: easee
  user: xxx
  password: xxx 
  charger: xxx
  name: wallbox5

vehicles:
- type: template
  template: id
  title: ID.3 
  user: xxx
  password: xxx
  vin: xxx
  capacity: 58 
  targetSoC: 100 
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  vehicles:
  - ev4
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  meters:
    grid: grid3
    pvs:
    - pv1
    batteries:
    - battery2
  bufferSoC: 5

sponsortoken: xxx

</pre></details>

<details><summary>Show evcc log output with <code>--log trace</code>:</summary><br><pre>
https://pastebin.com/P3NgWFX0

</pre></details>

andig, 2022-02-07:
Große logs bitte als Datei, Präfixe der Zeilen entfernen damit man etwas sehen kann.

hanzoh, 2022-02-07:
Ich habe es im Kommentar oben angepasst

andig, 2022-02-07:
Bei Dir wird auf ebene Circuit geschaltet, das scheint noch nicht zu funktionieren wie gedacht. Ich sehe was passiert, kann Dir aber nicht erklären, warum die Easee so reagiert (/cc @naltatis):

Bei minpv abschalten und 1p setzen:

    [lp-1  ] DEBUG 2022/02/07 16:09:28 set charge mode: minpv
    [lp-1  ] DEBUG 2022/02/07 16:09:29 phase disable timer elapsed
    [easee ] TRACE 2022/02/07 16:09:29 POST https://api.easee.cloud/api/chargers/xxx/commands/pause_charging
    [easee ] TRACE 2022/02/07 16:09:29 POST https://api.easee.cloud/api/sites/164512/circuits/157936/settings
    {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":0,"dynamicCircuitCurrentP3":0}
    [lp-1  ] DEBUG 2022/02/07 16:09:29 switched phases: 1p @ -2720W

Strom geht auf Null:

    [easee ] TRACE 2022/02/07 16:09:29 ProductUpdate xxx: DYNAMIC_CHARGER_CURRENT 0
    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: DYNAMIC_CIRCUIT_CURRENT_P2 0
    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: DYNAMIC_CIRCUIT_CURRENT_P1 16
    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: DYNAMIC_CIRCUIT_CURRENT_P3 0
    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L2 0

Der Grund sagt mir nix?

    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: REASON_FOR_NO_CURRENT 0052
    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L3 0
    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: OUTPUT_CURRENT 0
    [easee ] TRACE 2022/02/07 16:09:30 ProductUpdate xxx: CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 0

Stromo runter setzen:

    [easee ] TRACE 2022/02/07 16:09:33 POST https://api.easee.cloud/api/chargers/xxx/settings
    {"dynamicChargerCurrent":6}

Ladung wieder beginnen:

    [easee ] TRACE 2022/02/07 16:09:33 POST https://api.easee.cloud/api/chargers/xxx/commands/resume_charging
    [easee ] TRACE 2022/02/07 16:09:34 ProductUpdate xxx: LEDMODE 0023

Der Grund sagt mir nix?

    [easee ] TRACE 2022/02/07 16:09:34 ProductUpdate xxx: REASON_FOR_NO_CURRENT 0008
    [easee ] TRACE 2022/02/07 16:09:34 ProductUpdate xxx: DYNAMIC_CHARGER_CURRENT 6
    [easee ] TRACE 2022/02/07 16:09:35 ProductUpdate xxx: REASON_FOR_NO_CURRENT 0006
    [easee ] TRACE 2022/02/07 16:09:35 POST https://api.easee.cloud/api/chargers/xxx/settings
    {"dynamicChargerCurrent":6}

Das Problem: sie lädt nicht mehr, vmtl wg. `REASON_FOR_NO_CURRENT`- aber was bedeuten die?

Sinnig wäre, den Flow mal per Postman von Hand auszuprobieren um rauszufinden was da passiert.

andig, 2022-02-07:
Hier gibts die Gründe: https://github.com/fondberg/easee_hass/blob/master/custom_components/easee/const.py#L638-L657:

```
   52: RNC_MAX_DYNAMIC_CHARGER_CURRENT_TOO_LOW,
    8: RNC_NO_CURRENT_REQUEST_RECEIVED,
    6: RNC_WAITING_IN_FULLY,
```

andig, 2022-02-07:
Mein Vorschlag: frag bitte mal den Support was `REASON_FOR_NO_CURRENT = RNC_WAITING_IN_FULLY ` bedeutet nach einer 3p1p Umschaltung.

hanzoh, 2022-02-07:
Ich habe Easee mal über deren Kontaktformular geschrieben.
Nach einer Suche nach "waiting_in_fully easee" habe ich folgendes gefunden:

https://developer.easee.cloud/changelog/ocpp-14
WaitingInFully = 6, // charged queue (charger assumes one of: EV uses delayed charging, EV charging complete)

Ist also doch der ID.3 Schuld?

andig, 2022-02-07:
Keine Ahnung. Woher/wie sollte der Charger das wissen? Woran macht er das fest?

/cc @premultiply

premultiply, 2022-02-07:
Auf gut Deutsch: Das Fahrzeug fordert keinen Strom an.
Ist es eingeschlafen oder im Fehlerzustand und die Box bekommt es nicht geweckt?

andig, 2022-02-07:
> Auf gut Deutsch: Das Fahrzeug fordert keinen Strom an.
> Ist es eingeschlafen oder im Fehlerzustand und die Box bekommt es nicht geweckt?

Ich frag mich warum das früher nie passiert ist und ob die ID überhaupt dafür anfällig sind? Zusammen mit #2525 könnte man eher vermuten, dass es bei Easee eine Änderung gab?

hanzoh, 2022-02-09:
Gerade kam die Antwort vom easee Support:

`das Auto wartet in der Warteschlange.Wenn es mehr Energie benötigt, kann es später wieder aufgeladen werden.

Das kann passieren, wenn der Besitzer einen Ladeplan in das Auto programmiert hat.Es kann aber auch einfach nur voll geladen sein.`

premultiply, 2022-02-09:
Muss man das ggf. auf Status B2 (enabled und Status B) mappen?

andig, 2022-02-10:
> Muss man das ggf. auf Status B2 (enabled und Status B) mappen?

Gleicher Punkt wie in der Diskussion. Das Auto lädt nicht- bei uns ist das B. Warum sollte das genau bei der Phasenumschaltung passieren? Hier muss irgendwas anderes nicht stimmen.

hanzoh, 2022-02-10:
Sag Bescheid wenn ich nochmal etwas testen soll. Das ganze läuft bei den Schwiegereltern, am Wochenende könnte ich mir das nochmal anschauen.
Kann man evtl. über die VW API auslesen, ob und wann das Auto eingeschlafen ist?

andig, 2022-02-10:
Ich hab keine Idee was wir da tun sollen. Schau doch mal *nach* der Umschaltung wenn das Auto nicht mehr lädt in die Easee App und ins API und in die BMW App ob da irgendwelche spannenden Informationen drin sind. Aus Sicht Logfile ist alles korrekt.

andig, 2022-02-10:
Wenn beide Schritte erledigt sind könnten wir auch mal probieren ob sich am Verhalten etwas ändert wenn die Umschaltung am Charger anstatt am Circuit erfolgt. Dafür müssten ich aber noch was ändern.

Waldmensch1, 2022-02-13:
Das genau gleiche Verhalten habe ich heute bei meiner Go-E gesehen, mit angeschlossenem Enyaq, allerdings im PV Modus. Der EVCC schaltete von 3 auf 1 phasig und lud dann nicht mehr. Da ich nur über VPN verbunden war, habe ich im UI auf minPV geschaltet. Dadurch hat sich EVCC gefangen und die Ladung auf 1P aktiviert. Dann habe ich wieder auf PV geschaltet was dann ganz normal weiterlief. Habe dazu einen Beitrag im Forum geschrieben. Werde bei nächster Gelegenheit Log beschaffen.

Eventuell liegt der Fehler also nicht bei der easy, oder der Go-E hat den Gleichen.

andig, 2022-02-13:
Dann würde mich auch hier interessieren was die GoE app sagt. Die Easee war anscheinend freigeschaltet. Es bleibt mysteriös…

andig, 2022-02-27:
@Waldmensch1 Dein Fehler bzgl. Go-E sollte durch die in https://github.com/evcc-io/evcc/pull/2613 beschriebenen Anpassungen adressiert werden. Gerne dort testen, wird auch zeitnah in die Nightly Releases überführt.

Zum Easee Fehler habe ich weiter keine Idee- warten auf Easee Support und Diskussion in https://github.com/evcc-io/evcc/discussions/2504

closed 2022-02-27 by andig

andig, 2022-03-04:
Wollt ihr mal die nightly Version ausprobieren? Da sind viele Fehler der Phasenumschaltung behoben (und hoffentlich keine neuen dazu gekommen).

Waldmensch1, 2022-03-06:
> Wollt ihr mal die nightly Version ausprobieren? Da sind viele Fehler der Phasenumschaltung behoben (und hoffentlich keine neuen dazu gekommen).

Ich habe grad mal meinen Fork mit dem Master (stand 6.3. 10:00) gemerged und gebaut. Ich gehe mal davon aus, dass der Master die Quelle für die Nightlies ist. Läuft soweit. Mit Phasenumschaltung wird aber wetterbedingt heute nichts passieren. Die nächste Woche soll sonniger werden. Bin nur leider, aufgrund der aktuellen Geschehnisse, beruflich stark eingebunden.

hanzoh, 2022-03-07:
Ich hab heute Mittag die 0.85 installiert und der ID.3 lädt gerade mit Min+PV 3-phasig bei ca. 6kW Überschuss.
SoC target habe ich auf 100% gestellt, damit er noch ca. 2,5h braucht, bis er voll ist.
Bis dahin sollte die PV unter 4,2kW liefern, sodass er dann auf 1-phasig wechseln muss.
Was mir vorhin aufgefallen war, wenn jemand mit einem Durchlauferhitzer Wasser zapft, ist jeglicher Überschuss natürlich futsch, dann regelt evcc auch direkt runter, ein paar Sekunden später dann wieder zurück.
Macht es in solch einer Konstellation Sinn irgendeinen Parameter zu verstellen, sodass nicht sofort nachgeregelt wird?

premultiply, 2022-03-07:
Nein, die schnelle Regelung ist völlig unproblematisch.
Nur das dauernde komplett an- und abschalten wird vermieden um die Mechanik zu schonen.

hanzoh, 2022-03-16:
Sorry für die Verspätung, bin leider erst heute wieder dazu gekommen mit Schwiegervater zu testen.

Die Problematik besteht leider noch, ich konnte ein neues Logfile ziehen mit id auf WARN, easee auf TRACE und sonst DEBUG.

- ID.3 war bei 70% SoC mit Target 80% und an der Easee angeschlossen.
- Update auf 0.86 gemacht und evcc wieder gestartet (mit default Min+PV)
- Es passiert erstmal nichts 
- VW App sagt, dass das Laden gestartet werden kann, d.h. ohne zutun in dieser App passiert sonst nichts, also via App gestartet
- Easee legt von alleine nicht los, also via evcc auf Stop, dann auf Sofort gestellt (15:10:19)
- Es wird mit 11kW geladen
- Um 15:11:58 auf Min+PV umgestellt, Ladevorgang wird angehalten und startet nicht wieder von allein
- VW App sagt Ladevorgang fehlgeschlagen, bitte prüfen Sie das Kabel
- Via VW App wieder den Ladevorgang gestartet
- Von alleine passiert nichts, also nochmal in evcc auf Stop, dann auf Min+PV gestellt (15:16:59)
- Jetzt lädt er einphasig, da momentan nur 1,9kW vom Dach kommen und 0,7 vom Haus verbraucht werde
[evcc.log](https://github.com/evcc-io/evcc/files/8263277/evcc.log)

hanzoh, 2022-03-16:
Ich habe noch einen weiteren Test durchgeführt, um zu sehen wie das Auto sich verhält, wenn die Wallbox keinen Strom mehr hat:
- Um 16:24 FI Wallbox ausgeschaltet
- evcc bekommt von der easee Cloud steigende dynamicChargerCurrent Werte?
- FI wieder eingeschaltet, nach einiger Zeit kommen neue Werte über die API (16:26:20)
- ID.3 lädt aber nicht von alleine weiter
- Um 16:29:05 dann in evcc auf Sofort gestellt, Ladung startet erfolgreich 3-phasig mit 11 kW
- Jetzt kommt's: Um 16:29:41 wieder auf Min+PV gestellt, Wallbox stellt auf 1-phasig um, Auto lädt mit 1,4 kW weiter
- Wiederholung: Um 16:31:11 wieder auf Sofort gestellt, Wallbox stellt auf 3-phasig, Auto lädt mit 11 kW weiter
- 16:31:51 wieder auf Min+PV gestellt, Wallbox will auf 1-phasig stellen, Auto hört auf zu laden

[evcc.log](https://github.com/evcc-io/evcc/files/8264860/evcc.log)

PR #2613 (2022-02-28): Redesign phase handling
Fix #2519, #2525, #2638

Dieser PR implementiert vieles neu was mit Phasennutzung, -messung und -umschaltung zu tun hat:
- konfigurierte, gemessene und erwartete Phasenwerte sind nun klar logisch voneinander getrennt
- unbekannte Phasenkonfigurationen und Schaltzustände werden wenn nötig konservativ als 3p Fahrzeug/Anschluss interpretiert - dadurch startet z. B. die Ladung im PV-Modus im ungünstigsten Fall erst mit höheren Überschüssen als idealerweise nötig (siehe dazu auch nächster Punkt!)
- Konfiguration der `phases` am Ladepunkt und am Fahrzeug optimiert die Funktio [... cut]