evcc-io/evcc #2520: Nach evcc start fängt die Wallbox an zu laden, obwohl die Bedingungen nicht stimmen

andig, 2022-02-07:
### Discussed in https://github.com/evcc-io/evcc/discussions/2516

<div type='discussions-op-text'>

<sup>Originally posted by **Waldmensch1** February  7, 2022</sup>
Wenn ich evcc starte, wird fast grundsätzlich die Ladung gestartet, gefühlt egal wie die Rahmenbedingungen sind. Ich muss dann erstmal Stopp klicken und dann wieder PV. Dann scheint es normal zu laufen außer das die Skoda Cloud rumzuspacken scheint. Das funktionierte gestern noch, hat mit dem Startup Problem aber sicher nichts zu tun. Ich habe unten alles drangepackt, was mir zur Lösungsfindung nötig scheint. Falls was fehlt, einfach bescheid geben

Log vom Startvorgang, wo mit dem Laden begonnen wird.

```
pi@raspberrypi-evcc:~ $ sudo evcc
[main  ] INFO 2022/02/07 09:00:52 evcc 0.82 (fbd0d655)
[main  ] INFO 2022/02/07 09:00:52 using config file /etc/evcc.yaml
[main  ] INFO 2022/02/07 09:00:52 listening at 0.0.0.0:7070
[mqtt  ] INFO 2022/02/07 09:00:53 connecting evcc-351415490 at tcp://192.168.178.46:1883
[mqtt  ] DEBUG 2022/02/07 09:00:53 tcp://192.168.178.46:1883 connected
[lp-1  ] WARN 2022/02/07 09:00:57 ignoring phases config (3p) for switchable charger
[site  ] INFO 2022/02/07 09:00:57 site config:
[site  ] INFO 2022/02/07 09:00:57   meters:      grid ? pv ? battery ?
[site  ] INFO 2022/02/07 09:00:57     grid:      power ? energy ? currents ?
[site  ] INFO 2022/02/07 09:00:57     pv 0:      power ? energy ? currents ?
[site  ] INFO 2022/02/07 09:00:57     pv 1:      power ? energy ? currents ?
[site  ] INFO 2022/02/07 09:00:57     pv 2:      power ? energy ? currents ?
[site  ] INFO 2022/02/07 09:00:57     battery 0: power ? energy ? currents ? soc ?
[lp-1  ] INFO 2022/02/07 09:00:57 loadpoint 1:
[lp-1  ] INFO 2022/02/07 09:00:57   mode:        pv
[lp-1  ] INFO 2022/02/07 09:00:57   charger:     power ? energy ? currents ? phases ?
[lp-1  ] INFO 2022/02/07 09:00:57   meters:      charge ?
[lp-1  ] INFO 2022/02/07 09:00:57     charge:    power ? energy ? currents ?
[lp-1  ] INFO 2022/02/07 09:00:57   vehicles:    ?
[lp-1  ] INFO 2022/02/07 09:00:57     vehicle 0: range ? finish ? status ? climate ? providerLogin ?
[lp-1  ] INFO 2022/02/07 09:00:57 vehicle updated: unknown -> Enyaq
[site  ] DEBUG 2022/02/07 09:00:57 ----
[site  ] DEBUG 2022/02/07 09:00:57 grid power: -59W
[site  ] DEBUG 2022/02/07 09:00:57 pv power: 868W
[site  ] DEBUG 2022/02/07 09:00:57 battery power: 2W
[site  ] DEBUG 2022/02/07 09:00:57 grid currents: [1.5 1.24 1.38]A
[site  ] DEBUG 2022/02/07 09:00:57 battery soc 0: 5%
[site  ] DEBUG 2022/02/07 09:00:57 site power: -357W
[lp-1  ] DEBUG 2022/02/07 09:00:57 charge power: 0W
[lp-1  ] DEBUG 2022/02/07 09:00:57 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/07 09:00:57 charger status: B
[lp-1  ] ERROR 2022/02/07 09:01:07 vehicle soc: Get "https://api.connect.skoda-auto.cz/api/v1/charging/TMBJC7xx/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] DEBUG 2022/02/07 09:01:07 max charge current: +InfA = 0A + +InfA (-357W @ 0p)
[lp-1  ] DEBUG 2022/02/07 09:01:07 site power -357W < enable threshold 0W
[lp-1  ] DEBUG 2022/02/07 09:01:07 pv enable timer start: 0s
[lp-1  ] DEBUG 2022/02/07 09:01:07 pv action: enable in 0s
[lp-1  ] DEBUG 2022/02/07 09:01:07 pv enable timer elapsed
[lp-1  ] DEBUG 2022/02/07 09:01:08 max charge current: 6A
[lp-1  ] DEBUG 2022/02/07 09:01:08 charger enable
[lp-1  ] DEBUG 2022/02/07 09:01:08 charge power: 0W
[lp-1  ] DEBUG 2022/02/07 09:01:08 charge currents: [0 0 0]A
[site  ] DEBUG 2022/02/07 09:01:08 ----
[site  ] DEBUG 2022/02/07 09:01:08 grid power: 8W
[site  ] DEBUG 2022/02/07 09:01:08 pv power: 841W
[site  ] DEBUG 2022/02/07 09:01:08 battery power: -6W
[site  ] DEBUG 2022/02/07 09:01:08 grid currents: [1.58 1.21 1.45]A
[site  ] DEBUG 2022/02/07 09:01:08 battery soc 0: 5%
[site  ] DEBUG 2022/02/07 09:01:08 giving priority to battery charging at soc: 5
[site  ] DEBUG 2022/02/07 09:01:08 site power: -292W
[lp-1  ] DEBUG 2022/02/07 09:01:08 charge power: 0W
[lp-1  ] DEBUG 2022/02/07 09:01:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/07 09:01:08 charger status: B
[lp-1  ] DEBUG 2022/02/07 09:01:09 next soc poll remaining time: 59m48s
[lp-1  ] DEBUG 2022/02/07 09:01:09 max charge current: +InfA = 0A + +InfA (-292W @ 0p)
[lp-1  ] DEBUG 2022/02/07 09:01:09 pv timer reset
[lp-1  ] DEBUG 2022/02/07 09:01:09 pv action: inactive
[lp-1  ] DEBUG 2022/02/07 09:01:09 max charge current: 16A
[lp-1  ] DEBUG 2022/02/07 09:01:09 charge power: 0W
[lp-1  ] DEBUG 2022/02/07 09:01:09 charge currents: [0 0 0]A
[... 374 lines cut ...]
site:
  title: Stadtweg
  residualPower: -300
  prioritySoC: 70 # Hausbatterie bekommt bis zum SoC 70% Priorität beim laden
  bufferSoC: 80 # Hausbatterie wird oberhalb SoC 80% als Puffer genutzt
  meters:
    grid: sma-em
    pvs:
    - pv-tripower
    - pv-growatt
    - pv-garten
    batteries:
    - solarwatt

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.264 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.13 # [currency]/kWh

```</div>

andig, 2022-02-07:
@waldmensch1 kannst Du das ausprobieren?

andig, 2022-02-08:
Du brauchst kein npm. Einfach `make` reicht wie's in der Doku steht.

Waldmensch1, 2022-02-08:
es gibt scheinbar kein make auf meinem Windows System

```
PS C:\Users\tommy\Documents\Node_Projects\evcc\evcc> make
make : Die Benennung "make" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines ausführbaren Programms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt
ist (sofern enthalten), und wiederholen Sie den Vorgang.
In Zeile:1 Zeichen:1
+ make
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (make:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS C:\Users\tommy\Documents\Node_Projects\evcc\evcc>
```

andig, 2022-02-08:
dann hast Du kein make installiert. Probier einfach

    go run main.go

wenn Du den PR sauber ausgecheckt hast.

Waldmensch1, 2022-02-08:
@andig japp, rennt, nachdem er das halbe github runtergeladen hat :D

Der Wechselrichter stellt sich allerdings aktuell tot. Ich mache an der Stelle morgen nach Sonnenaufgang weiter. Die Inf sind allerdings schon mal weg

```
go: downloading github.com/klauspost/compress v1.14.1
go: downloading github.com/mattn/go-colorable v0.1.12
go: downloading github.com/fatih/color v1.13.0
go: downloading github.com/onsi/ginkgo v1.16.5
go: downloading github.com/onsi/gomega v1.17.0
go: downloading github.com/go-playground/universal-translator v0.18.0
go: downloading github.com/leodido/go-urn v1.2.1
go: downloading github.com/go-playground/locales v0.14.0
go: downloading github.com/jmespath/go-jmespath v0.4.0
[main  ] INFO 2022/02/08 19:15:32 evcc 0.0.1-alpha (HEAD)
[main  ] INFO 2022/02/08 19:15:32 using config file C:\Users\tommy\Documents\Node_Projects\evcc\evcc\evcc.yaml
[main  ] INFO 2022/02/08 19:15:32 listening at 0.0.0.0:7070
[mqtt  ] INFO 2022/02/08 19:15:32 connecting evcc-2047964124 at tcp://192.168.178.46:1883
[mqtt  ] DEBUG 2022/02/08 19:15:32 tcp://192.168.178.46:1883 connected
[lp-1  ] WARN 2022/02/08 19:15:36 ignoring phases config (3p) for switchable charger
[site  ] INFO 2022/02/08 19:15:36 site config:
[site  ] INFO 2022/02/08 19:15:36   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/02/08 19:15:36     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/02/08 19:15:36     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/02/08 19:15:36     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/02/08 19:15:36     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/02/08 19:15:36     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/02/08 19:15:36 loadpoint 1:
[lp-1  ] INFO 2022/02/08 19:15:36   mode:        pv
[lp-1  ] INFO 2022/02/08 19:15:36   charger:     power ✓ energy ✓ currents ✓ phases ✓
[lp-1  ] INFO 2022/02/08 19:15:36   meters:      charge ✓
[lp-1  ] INFO 2022/02/08 19:15:36     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/02/08 19:15:36   vehicles:    ✓
[lp-1  ] INFO 2022/02/08 19:15:36     vehicle 0: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/02/08 19:15:36 vehicle updated: unknown -> Enyaq
[site  ] DEBUG 2022/02/08 19:15:36 ----
[site  ] DEBUG 2022/02/08 19:15:36 grid power: 898W
[mqtt  ] DEBUG 2022/02/08 19:15:36 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:15:47 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:15:57 stadtweg/energy/tripower wait for initial value
[site  ] ERROR 2022/02/08 19:16:07 updating pv meter 0: stadtweg/energy/tripower outdated: 10s
[site  ] DEBUG 2022/02/08 19:16:07 pv power: 0W
[site  ] DEBUG 2022/02/08 19:16:07 battery power: 2W
[site  ] DEBUG 2022/02/08 19:16:07 grid currents: [2.18 1.09 1.98]A
[site  ] DEBUG 2022/02/08 19:16:07 battery soc 0: 5%
[site  ] DEBUG 2022/02/08 19:16:07 site power: 900W
[lp-1  ] DEBUG 2022/02/08 19:16:07 charge power: 0W
[lp-1  ] DEBUG 2022/02/08 19:16:07 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/08 19:16:08 charger status: B
[lp-1  ] DEBUG 2022/02/08 19:16:08 vehicle soc: 62%
[lp-1  ] DEBUG 2022/02/08 19:16:08 vehicle range: 207km
[lp-1  ] DEBUG 2022/02/08 19:16:08 assuming initial phase state: 3p
[lp-1  ] DEBUG 2022/02/08 19:16:08 available power below 3p min threshold of 4140W
[lp-1  ] DEBUG 2022/02/08 19:16:08 phase action: scale1p in 0s
[lp-1  ] DEBUG 2022/02/08 19:16:08 phase disable timer elapsed
[lp-1  ] DEBUG 2022/02/08 19:16:08 pv timer elapse
[lp-1  ] DEBUG 2022/02/08 19:16:08 pv action: inactive
[lp-1  ] DEBUG 2022/02/08 19:16:08 switched phases: 1p @ -900W
[lp-1  ] DEBUG 2022/02/08 19:16:09 charge power: 0W
[lp-1  ] DEBUG 2022/02/08 19:16:09 charge currents: [0 0 0]A
[site  ] DEBUG 2022/02/08 19:16:09 ----
[site  ] DEBUG 2022/02/08 19:16:09 grid power: 912W
[mqtt  ] DEBUG 2022/02/08 19:16:09 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:16:19 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:16:29 stadtweg/energy/tripower wait for initial value
[site  ] ERROR 2022/02/08 19:16:39 updating pv meter 0: stadtweg/energy/tripower outdated: 10s
[site  ] DEBUG 2022/02/08 19:16:40 pv power: 0W
[site  ] DEBUG 2022/02/08 19:16:40 battery power: 2W
[site  ] DEBUG 2022/02/08 19:16:40 grid currents: [1.96 1.1 1.98]A
[site  ] DEBUG 2022/02/08 19:16:40 battery soc 0: 5%
[... 65 lines cut ...]
[site  ] DEBUG 2022/02/08 19:18:16 ----
[site  ] DEBUG 2022/02/08 19:18:16 grid power: 1039W
[mqtt  ] DEBUG 2022/02/08 19:18:16 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:18:26 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:18:37 stadtweg/energy/tripower wait for initial value
[site  ] ERROR 2022/02/08 19:18:47 updating pv meter 0: stadtweg/energy/tripower outdated: 10s
[site  ] DEBUG 2022/02/08 19:18:47 pv power: 0W
[site  ] DEBUG 2022/02/08 19:18:47 battery power: -3W
[site  ] DEBUG 2022/02/08 19:18:47 grid currents: [2.1 1.08 2.63]A
[site  ] DEBUG 2022/02/08 19:18:47 battery soc 0: 5%
[site  ] DEBUG 2022/02/08 19:18:47 giving priority to battery charging at soc: 5
[site  ] DEBUG 2022/02/08 19:18:47 site power: 1039W
[lp-1  ] DEBUG 2022/02/08 19:18:47 charge power: 0W
[lp-1  ] DEBUG 2022/02/08 19:18:47 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/08 19:18:47 charger status: B
[lp-1  ] DEBUG 2022/02/08 19:18:47 next soc poll remaining time: 57m20s
[... cut ...] ge current: 0A = 0A + -4.52A (1039W @ 1p)
[lp-1  ] DEBUG 2022/02/08 19:18:48 charge power: 0W
[lp-1  ] DEBUG 2022/02/08 19:18:48 charge currents: [0 0 0]A
[site  ] DEBUG 2022/02/08 19:18:48 ----
[site  ] DEBUG 2022/02/08 19:18:48 grid power: 1042W
[mqtt  ] DEBUG 2022/02/08 19:18:48 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:18:58 stadtweg/energy/tripower wait for initial value
[mqtt  ] DEBUG 2022/02/08 19:19:08 stadtweg/energy/tripower wait for initial value
```

Waldmensch1, 2022-02-09:
@andig Also ich habe jetzt mehre Neustarts gemacht - die Inf treten nicht mehr auf und der go-E hält die Füße still. Ich denke mal, Du hast es gefixt

```
PS C:\Users\tommy\Documents\Node_Projects\evcc\evcc> C:\Programme\Go\bin\go run main.go
[main  ] INFO 2022/02/09 09:48:06 evcc 0.0.1-alpha (HEAD)
[main  ] INFO 2022/02/09 09:48:06 using config file C:\Users\tommy\Documents\Node_Projects\evcc\evcc\evcc.yaml
[main  ] INFO 2022/02/09 09:48:06 listening at 0.0.0.0:7070
[mqtt  ] INFO 2022/02/09 09:48:07 connecting evcc-1643182329 at tcp://192.168.178.46:1883
[mqtt  ] DEBUG 2022/02/09 09:48:07 tcp://192.168.178.46:1883 connected
[lp-1  ] WARN 2022/02/09 09:48:09 ignoring phases config (3p) for switchable charger
[site  ] INFO 2022/02/09 09:48:09 site config:
[site  ] INFO 2022/02/09 09:48:09   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/02/09 09:48:09     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/02/09 09:48:09     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/02/09 09:48:09     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/02/09 09:48:09     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/02/09 09:48:09     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/02/09 09:48:09 loadpoint 1:
[lp-1  ] INFO 2022/02/09 09:48:09   mode:        pv
[lp-1  ] INFO 2022/02/09 09:48:09   charger:     power ✓ energy ✓ currents ✓ phases ✓
[lp-1  ] INFO 2022/02/09 09:48:09   meters:      charge ✓
[lp-1  ] INFO 2022/02/09 09:48:09     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/02/09 09:48:09   vehicles:    ✓
[lp-1  ] INFO 2022/02/09 09:48:09     vehicle 0: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/02/09 09:48:09 vehicle updated: unknown -> Enyaq
[site  ] DEBUG 2022/02/09 09:48:09 ----
[site  ] DEBUG 2022/02/09 09:48:09 grid power: 294W
[site  ] DEBUG 2022/02/09 09:48:09 pv power: 532W
[mqtt  ] DEBUG 2022/02/09 09:48:09 stadtweg/energy/solarwatt wait for initial value
[site  ] DEBUG 2022/02/09 09:48:10 battery power: -5W
[site  ] DEBUG 2022/02/09 09:48:10 grid currents: [1.7 0.647 1.52]A
[site  ] DEBUG 2022/02/09 09:48:10 battery soc 0: 5%
[site  ] DEBUG 2022/02/09 09:48:10 giving priority to battery charging at soc: 5
[site  ] DEBUG 2022/02/09 09:48:10 site power: 594W
[lp-1  ] DEBUG 2022/02/09 09:48:10 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:10 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/09 09:48:11 charger status: B
[lp-1  ] DEBUG 2022/02/09 09:48:11 vehicle soc: 59%
[lp-1  ] DEBUG 2022/02/09 09:48:11 vehicle range: 197km
[lp-1  ] DEBUG 2022/02/09 09:48:11 assuming initial phase state: 3p
[lp-1  ] DEBUG 2022/02/09 09:48:11 available power below 3p min threshold of 4140W
[lp-1  ] DEBUG 2022/02/09 09:48:11 phase action: scale1p in 0s
[lp-1  ] DEBUG 2022/02/09 09:48:11 phase disable timer elapsed
[lp-1  ] DEBUG 2022/02/09 09:48:11 pv timer elapse
[lp-1  ] DEBUG 2022/02/09 09:48:11 pv action: inactive
[lp-1  ] DEBUG 2022/02/09 09:48:11 switched phases: 1p @ -594W
[lp-1  ] DEBUG 2022/02/09 09:48:12 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:12 charge currents: [0 0 0]A
[site  ] DEBUG 2022/02/09 09:48:19 ----
[site  ] DEBUG 2022/02/09 09:48:19 grid power: 291W
[site  ] DEBUG 2022/02/09 09:48:19 pv power: 536W
[site  ] DEBUG 2022/02/09 09:48:19 battery power: 2W
[site  ] DEBUG 2022/02/09 09:48:19 grid currents: [1.64 0.649 1.45]A
[site  ] DEBUG 2022/02/09 09:48:19 battery soc 0: 5%
[site  ] DEBUG 2022/02/09 09:48:19 site power: 593W
[lp-1  ] DEBUG 2022/02/09 09:48:20 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:20 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/09 09:48:20 charger status: B
[lp-1  ] DEBUG 2022/02/09 09:48:20 next soc poll remaining time: 59m50s
[lp-1  ] DEBUG 2022/02/09 09:48:20 max charge current: 0A = 0A + -2.58A (593W @ 1p)
[lp-1  ] DEBUG 2022/02/09 09:48:20 pv [enable] timer reset
[lp-1  ] DEBUG 2022/02/09 09:48:20 pv action: inactive
[lp-1  ] DEBUG 2022/02/09 09:48:21 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:21 charge currents: [0 0 0]A
[site  ] DEBUG 2022/02/09 09:48:29 ----
[site  ] DEBUG 2022/02/09 09:48:29 grid power: 281W
[site  ] DEBUG 2022/02/09 09:48:29 pv power: 541W
[site  ] DEBUG 2022/02/09 09:48:29 battery power: -2W
[site  ] DEBUG 2022/02/09 09:48:29 grid currents: [1.73 0.662 1.44]A
[site  ] DEBUG 2022/02/09 09:48:29 battery soc 0: 5%
[site  ] DEBUG 2022/02/09 09:48:29 giving priority to battery charging at soc: 5
[site  ] DEBUG 2022/02/09 09:48:29 site power: 581W
[lp-1  ] DEBUG 2022/02/09 09:48:30 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:30 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/09 09:48:30 charger status: B
[lp-1  ] DEBUG 2022/02/09 09:48:30 next soc poll remaining time: 59m40s
[lp-1  ] DEBUG 2022/02/09 09:48:30 max charge current: 0A = 0A + -2.53A (581W @ 1p)
[lp-1  ] DEBUG 2022/02/09 09:48:30 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:30 charge currents: [0 0 0]A
[site  ] DEBUG 2022/02/09 09:48:39 ----
[site  ] DEBUG 2022/02/09 09:48:39 grid power: 295W
[site  ] DEBUG 2022/02/09 09:48:39 pv power: 540W
[site  ] DEBUG 2022/02/09 09:48:39 battery power: 0W
[site  ] DEBUG 2022/02/09 09:48:39 grid currents: [1.76 0.662 1.48]A
[site  ] DEBUG 2022/02/09 09:48:39 battery soc 0: 5%
[site  ] DEBUG 2022/02/09 09:48:39 site power: 595W
[lp-1  ] DEBUG 2022/02/09 09:48:40 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:40 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/02/09 09:48:40 charger status: B
[lp-1  ] DEBUG 2022/02/09 09:48:40 next soc poll remaining time: 59m30s
[lp-1  ] DEBUG 2022/02/09 09:48:40 max charge current: 0A = 0A + -2.59A (595W @ 1p)
[lp-1  ] DEBUG 2022/02/09 09:48:40 charge power: 0W
[lp-1  ] DEBUG 2022/02/09 09:48:41 charge currents: [0 0 0]A
```

closed 2022-02-09 by andig