evcc-io/evcc #2524: Nieder- und Hochtarif

rediculum, 2022-02-07:
In der Schweiz existiert im Stromnetz ein Nieder- und Hochtarif (Zone1 & Zone2). Diese sind zeitlich und preislich unterschiedlich. Hier ein Beispiel des Kantons Aargau: https://www.aew.ch/kundenservice/informationen/preiszonen

Feature Request: Möglichkeit anstatt einen fixen Strombetrag, zwei unterschiedliche zu definieren mit Zeitspanne.

andig, 2022-02-10:
Das wäre sowas wie ein etwas komplizierterer "fixed" Tarif. Letzterer findet sich in https://github.com/evcc-io/evcc/blob/master/tariff/fixed.go. Der müsste jetzt um die Kalenderzonen erweitert werden.

PR welcome!

andig, 2022-02-10:
Idee: die Zonendefinition könnte man sehr schön einfach mittels https://pkg.go.dev/github.com/robfig/cron parsen.

rediculum, 2022-02-10:
Hi Andreas

PR kann ich leider nicht liefern. GoLang hab ich _null_ Ahnung. Ich wollte dich lediglich darüber informieren, dass es bei uns sowas gibt und der Tarif genauer berechnet werden könnte. Man kann natürlich einfach den Durchschnitt der zwei Tarife als fixed eintragen und gut ist. Bleibt dann halt ungenau. Ich überlass es dir, so etwas zu implementieren. Aber der nächste Schweizer wird kommen ;)

BlueSnake830, 2022-02-17:
Gibt es auch in Deutschland: Wallboxen dürfen z.B. an dem Wärmepumpenzähler angeschlossen werden, sofern vorhanden. Dieser hat auch HT/NT.

andig, 2022-04-30:
Die Implementierungsidee- falls es jemand versuchen will- wäre einen "Stapel" von time ranges zu definieren (der sortiert sein muss). Jeder Eintrag im Stapel wird geprüft ob er den fraglichen Zeitpunkt enthält. Der letzte Match definiert den aktuellen Preis. 

Die Einträge selbst bräuchten initial mal folgende Information, grob analog `cron`:

- Wochentag (`0-6 or SUN-SAT`, evtl. auch eher europäische Notation)
- Stunde (`0-23`)
- Minute (`0-59`)
- Preis

Leere Einträge bedeuten "komplette Zeitspanne". Der Zonentarif wäre dann

```
- # Zone 2 
  price: X
- # Zone 1
  valid:
    dow: mon-fri
    hour: 7-20
  price: Y
- # Zone 1
  valid: 
    dow: sat
    hour: 7-13
  price: Y
```  

Das sollte relativ einfach zu implementieren sein, gerne versuchen!

andig, 2022-12-13:
Reopened on behalf of https://github.com/evcc-io/evcc/discussions/3530#discussioncomment-4383182

closed 2022-12-29 by andig