evcc-io/evcc #2525: charger out of sync: expected enabled, got disabled // Easee Wallbox

tmy0, 2022-02-07:
**Describe the bug**
Im Modus min+pv wird das Laden stoppt.

**To Reproduce**
Auto an Wallbox angeschlossen. Ladung beginnt. Irgendwann wird das Laden gestoppt. 
Den genauen Zeitpunkt kann ich nicht reproduzieren. Im Allgemeinen ist das Laden im Modus min+pv schon öfter gestoppt worden, obwohl der SoC noch nicht bei 100% war. 

**Expected behavior**
Es wird im Modus min+pv so lange geladen, bis das Kabel entfernt wird oder SoC 100% erreicht wurde.

**EVCC details:**
evcc version 0.82 (fbd0d655)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
Unsupported Config Type ""

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>

uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: error

meters:
- name: grid1
  type: custom
  power:
    source: modbus
    model: sunspec
    id: 1
    uri: 192.168.178.66:1502
    timeout: 3s
    subdevice: 1
    value: 203:W # sunspec 3-phase meter power reading
    scale: -1
- type: template
  template: solaredge-inverter
  id: 1
  host: 192.168.178.66  # IP-Adresse oder Hostname
  port: 1502  # Port
  id: 1
  port: 1502  # Port
  usage: pv
  modbus: tcpip
  timeout: 3s
  name: pv2

chargers:
- type: template
  template: easee
  user:  # Benutzername um auf das Gerät zuzugreifen
  password:   # Passwort des Benutzerkontos
  charger: 
  name: wallbox4

vehicles:
- type: template
  template: opel
  title: Ginger  # Wird in der Benutzeroberfläche angezeigt
  user:   # Benutzername um auf das Gerät zuzugreifen
  password: # Passwort des Benutzerkontos
  capacity: 50  # Akku-Kapazität in kWh
  name: ev3

loadpoints:
- title: Carport
[... 1933 lines cut ...]
Feb 06 11:23:40 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:40 charge currents: [0.009 0.01 0.01]A
Feb 06 11:23:50 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:23:50 ----
Feb 06 11:23:50 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:23:50 grid power: -3342W
Feb 06 11:23:50 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:23:50 pv power: 4628W
Feb 06 11:23:50 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:23:50 site power: -3342W
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 charge power: 0W
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 charge currents: [0.009 0.01 0.01]A
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 charger status: B
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 next soc poll remaining time: 51m30s
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 max charge current: 14.5A = 0A + 14.5A (-3342W @ 1p)
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 max charge current: 14A
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 charge power: 0W
Feb 06 11:23:50 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:23:50 charge currents: [0.009 0.01 0.01]A
Feb 06 11:24:00 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:24:00 ----
Feb 06 11:24:00 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:24:00 grid power: -3237W
Feb 06 11:24:00 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:24:00 pv power: 4654W
Feb 06 11:24:00 raspberrypi evcc[26090]: [site  ] DEBUG 2022/02/06 11:24:00 site power: -3237W
Feb 06 11:24:00 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:24:00 charge power: 0W
Feb 06 11:24:00 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:24:00 charge currents: [0.009 0.01 0.01]A
Feb 06 11:24:00 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:24:00 charger status: B
Feb 06 11:24:00 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:24:00 next soc poll remaining time: 51m20s
Feb 06 11:24:00 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:24:00 max charge current: 14.1A = 0A + 14.1A (-3237W @ 1p)
Feb 06 11:24:00 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:24:00 max charge current: 14A
Feb 06 11:24:00 raspberrypi evcc[26090]: [lp-1  ] DEBUG 2022/02/06 11:24:00 charge power: 0W
Feb 0

andig, 2022-02-07:
Große logs bitte als Datei, Präfixe der Zeilen entfernen damit man etwas sehen kann.

andig, 2022-02-07:
Bzgl. Ladestop bitte https://github.com/evcc-io/evcc/issues/2519 anschauen. Gleiches Problem? Mangels `trace` nicht erkennbar.

andig, 2022-02-10:
ping @tmy0

tmy0, 2022-02-11:
Ich komme morgen erst wieder zum testen. Ich melde mich.

tmy0, 2022-02-18:
Hi, sorry für die Verspätung. Gestern konnte ich mal wieder laden und der Fehler trat wieder auf. 
Zeit: 12:47:37 stop charging. Danach starten das Laden nicht mehr, obwohl genügend Power vorhanden war. Generell dürfte er im Modus min+pv ja nicht stoppen. Während des kompletten Ladevorgangs, also auch vor 12:47 Uhr stoppte das Laden immer mal wieder. Jedoch startete das Laden dann nach ein paar Minuten wieder. 
Ich kann das Log leider nicht so gut interpretieren. Reicht euch ein gewisser Zeitraum um den Eintrag "stop charging"?
Soll ich nach gewissen Einträgen gucken?

Hier der Trace (12:47:37 stop charging):
Feb 17 12:47:37 raspberrypi evcc[872]: [modbus] TRACE 2022/02/17 12:47:37 203:0:W: 4485
Feb 17 12:47:37 raspberrypi evcc[872]: [site  ] DEBUG 2022/02/17 12:47:37 grid power: -4485W
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 gridPower: -4485
Feb 17 12:47:37 raspberrypi evcc[872]: [modbus] TRACE 2022/02/17 12:47:37 modbus: send 08 14 00 00 00 06 01 03 9c 87 00 32
Feb 17 12:47:37 raspberrypi evcc[872]: [modbus] TRACE 2022/02/17 12:47:37 modbus: recv 08 14 00 00 00 67 01 03 64 09 1c 03 0d 03 07 03 07 ff fe 0f 2a 0f 32 0f 2c 08 c2 08 c7 08 c1 ff ff 14 7a 00 00 13 8a ff fe 14 81 00 00 70 e4 ff fe d9 00 ff fe 03 64 0e e8 00 00 1b d0 ff fd 1d 31 ff ff 14 c9 00 00 80 00 13 6d 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00
Feb 17 12:47:37 raspberrypi evcc[872]: [site  ] DEBUG 2022/02/17 12:47:37 pv power: 5242W
Feb 17 12:47:37 raspberrypi evcc[872]: [site  ] DEBUG 2022/02/17 12:47:37 site power: -4485W
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 charge power: 0W
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 charge currents: [0.009 0.009 0.011]A
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 pvPower: 5242
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/mode: minpv
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/chargePower: 0
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/chargeCurrents: [0.008999999612569809 0.008999999612569809 0.010999999940395355]
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 charger status: B
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/chargeCurrent: 0
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/chargedEnergy: 8761.03591918945
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/chargeDuration: 1h21m19s
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] INFO 2022/02/17 12:47:37 stop charging <-
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/chargeCurrent: 0
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 vehicle soc: 77%
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/connected: true
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/charging: false
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/enabled: false
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/vehicleSoC: 77
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 vehicle range: 224km
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/chargeRemainingDuration: -1ns
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 vehicle odometer: 3211km
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/vehicleRange: 224
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 max charge current: 6.5A = 0A + 6.5A (-4485W @ 3p)
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] WARN 2022/02/17 12:47:37 ignoring inconsistent phases: 1p < 3p observed active
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/vehicleOdometer: 3211.1
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 available power above 3p min threshold of 4140W
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 start phase enable timer: 0s
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/phaseAction: scale3p
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 phase action: scale3p in 0s
Feb 17 12:47:37 raspberrypi evcc[872]: [cache ] TRACE 2022/02/17 12:47:37 lp-1/phaseRemaining: 0s
Feb 17 12:47:37 raspberrypi evcc[872]: [lp-1  ] DEBUG 2022/02/17 12:47:37 phase enable timer elapsed
Feb 17 12:47:37 raspberrypi evcc[872]: [easee ] TRACE 2022/02/17 12:47:37 GET https://api.easee.cloud/api/sites/104296/circuits/101292/settings
Feb 17 12:47:37 raspberrypi evcc[872]: [easee ] TRACE 2022/02/17 12:47:37
Feb 17 12:47:37 raspberrypi evcc[872]: GET /api/sites/104296/circuits/101292/settings HTTP/1.1
Feb 17 12:47:37 raspberrypi evcc[872]: Host: api.easee.cloud
Feb 17 12:47:37 raspberrypi evcc[872]: User-Agent: Go-http-client/1.1
Feb 17 12:47:37 raspberrypi evcc[872]: Accept: application/json
Feb 17 12:47:37 raspberrypi evcc[872]: Authorization: Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJBY2NvdW50SWQiOjk4MDkwLCJVc2VySWQiOjc5NTk0LCJ1bmlxdWVfbmFtZSI6Ijk4MDkwIiwicm9sZSI6IlVzZXIiLCJuYmYiOjE2NDUwODgxOTIsImV4cCI6MTY0NTE3NDU5MiwiaWF0IjoxNjQ1MDg4MTkyfQ.-ZHchvL-Z6jvgt40CsUaDXJfiZdDWhtunx1-daoaH3s
[... cut ...]
Feb 17 12:47:47 raspberrypi evcc[872]: Host: api.easee.cloud
Feb 17 12:47:47 raspberrypi evcc[872]: User-Agent: Go-http-client/1.1
Feb 17 12:47:47 raspberrypi evcc[872]: Content-Length: 0
Feb 17 12:47:47 raspberrypi evcc[872]: Authorization: Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJBY2NvdW50SWQiOjk4MDkwLCJVc2VySWQiOjc5NTk0LCJ1bmlxdWVfbmFtZSI6Ijk4MDkwIiwicm9sZSI6IlVzZXIiLCJuYmYiOjE2NDUwODgxOTIsImV4cCI6MTY0NTE3NDU5MiwiaWF0IjoxNjQ1MDg4MTkyfQ.-ZHchvL-Z6jvgt40CsUaDXJfiZdDWhtunx1-daoaH3s
Feb 17 12:47:47 raspberrypi evcc[872]: Con

andig, 2022-02-20:
Wir werden die komplette Phasenumschaltung redesignen. Aufgrund der Komplexität im Code macht es derzeit nicht mehr soviel Sinn, Einzelprobleme zu lösen. Please stay tuned!

tmy0, 2022-02-20:
ok. dann warte ich auf ein neues release mit Bezug auf #2613. 
Danke.

andig, 2022-02-27:
Dieser Fehler sollte durch die in https://github.com/evcc-io/evcc/pull/2613 beschriebenen Anpassungen adressiert werden. Gerne dort testen, wird auch zeitnah in die Nightly Releases überführt.

closed 2022-02-27 by andig