evcc-io/evcc #2538: UI, Werte für Netzbezug, Eigenverbrauch und Einspeisung fehlen in der PC Ansicht

Waldmensch1, 2022-02-09:
**Describe the bug**
A clear and concise description of what the bug is.

Wenn man das UI auf dem Handy anschaut, stehen neben Netzbezug, Eigenverbrauch und Einspeisung die entsprechenden Werte. Öffnet man das UI im Browser auf dem PC im Chrome, fehlen die Werte. Schiebt man das Fenster auf dem PC zusammen, bis das responsive Design auf die Mobilansicht wechselt erscheinen die Werte plötzlich.

andig, 2022-02-09:
Welcher Browser auf dem PC hat das Problem? Screenshot?

Waldmensch1, 2022-02-09:
Firefox
[image]

Chrome:
[image]

Edge:
[image]

Edge zusammengeschoben:
[image]

Edge noch weiter zusammengeschoben, So wird es auch auf dem Handy dargestellt

[image]

andig, 2022-02-09:
> Wenn man das UI auf dem Handy anschaut, stehen neben Netzbezug, Eigenverbrauch und Einspeisung die entsprechenden Werte. Öffnet man das UI im Browser auf dem PC im Chrome, fehlen die Werte. Schiebt [...]

Ich sehe nicht, wo da jetzt was in Chrome fehlt?

naltatis, 2022-02-09:
Ja, die Werte für Eigenverbrauch, Netzbezug, Einspeisung zeigen wir momentan nur in der mobilen Ansicht explizit in der Legende an. Hintergrund: Im Energiefluss Diagramm werden genau diese Werte ja bereites im Balken dargestellt. Da das Diagramm am größeren Screens IdR genug Platz hat sollten alle relevanten Werte dort ablesbar sein.

Wir arbeiten allerdings gerade an einem neuen Design bei dem wir diese Entscheidung (die ja bei dir auch gerade für Verwirrung sorgt) anpassen. Im neuen Design wird die vollständige Legende auf allen screens mit Werten dargestellt. Hab noch ein paar Wochen Geduld, da wird was kommen ;)

Ich würd diesen Issue aber jetzt erstmal schließen weil das aktuelle Verhalten so funktioniert wie es gedacht war.

closed 2022-02-09 by naltatis

Waldmensch1, 2022-02-09:
@naltatis Supi! Ich würde mich über ein "dark Theme" freuen, falls Ihr Zeit dafür habt :)