evcc-io/evcc #2578: Bluelink app language changed

andig, 2022-02-13:
Aber, immer wenn ich über "evcc vehicle" das Fahrzeug abgefragt habe, ist danach in der Bluelink App auf dem Handy die Sprache von Deutsch auf Englisch geändert. Hat das damit zu tun, dass es momentan nicht funktioniert oder gibt es da ein anderes Problem?

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/2577_

andig, 2022-02-13:
Das ist blöd. Leider müssen wir beim Login eine Sprache setzen. Soweit ich sehe lässt sie sich nicht ermitteln.

andig, 2022-02-14:
@volkerk62 wir könnten die Sprache konfigurierbar machen wenn Dir das hilft.

VolkerK62, 2022-02-14:
Fänd ich nicht schlecht. Wie läuft dann dieses konfigurierbar machen? Default ist englisch und wenn ich was anderes wünsche, muss ich das in der yaml angeben?

andig, 2022-02-14:
Mit dem PR kannst Du 

    language: de

im Auto konfigurieren.

VolkerK62, 2022-02-14:
Sorry für die vermeintlich dumme Frage: Was meinst du mit "PR"?

closed 2022-02-14 by andig

zwiebelxxl, 2022-02-17:
ich habe die 0.83 installiert und die Sprache gesetzt aber es kommt eine Fehlermeldung.

>
vehicles:
        type: template
        template: hyundai
        language: de
        title: IONIQ 5
        user: xxx
        password: xxx
        vin: xxx
        capacity: 58
        name: ev1
>

> 
[main  ] INFO 2022/02/17 11:30:49 evcc 0.83 (7a99de71)
[main  ] INFO 2022/02/17 11:30:49 using config file evcc.yaml
[mqtt  ] INFO 2022/02/17 11:30:49 connecting evcc-1366312613 at tcp://192.168.10.16:1883
[mqtt  ] DEBUG 2022/02/17 11:30:49 tcp://192.168.10.16:1883 connected
.SoC:     vehicle not available: cannot create vehicle 'template': invalid element 'name: language'
Capacity: 0kWh
>

Statusbit, 2022-02-17:
> Mit dem PR kannst Du
> 
> ```
> language: de
> ```
> 
> im Auto konfigurieren.

Also zumindest mit der 0.83 bekommt man da einen Fehler (kann mit language nichts anfangen oder so ähnlich).
Habe es mal auskommentiert ... und kümmere mich darum, dass überhaut geladen wird.

andig, 2022-02-17:
Die Suchfunktion hilft, siehe https://github.com/evcc-io/evcc/issues/2619#issuecomment-1042829356.

Statusbit, 2022-02-17:
Okay danke für die Info ... ich komme von der Suchfunktion aber halt über einen anderen Beitrag.

[von hier](https://github.com/evcc-io/evcc/discussions/2577)