evcc-io/evcc #2619: Bluelink app language changed - not working

zwiebelxxl, 2022-02-17:
ich habe die 0.83 installiert und die Sprache gesetzt aber es kommt eine Fehlermeldung. Vielleicht an der falschen Stelle?

vehicles:
type: template
template: hyundai
language: de
title: IONIQ 5
user: xxx
password: xxx
vin: xxx
capacity: 58
name: ev1

[main ] INFO 2022/02/17 11:30:49 evcc 0.83 (https://github.com/evcc-io/evcc/commit/7a99de719b51e75026aa7297328d2a20210714fb)
[main ] INFO 2022/02/17 11:30:49 using config file evcc.yaml
[mqtt ] INFO 2022/02/17 11:30:49 connecting evcc-1366312613 at tcp://192.168.10.16:1883
[mqtt ] DEBUG 2022/02/17 11:30:49 tcp://192.168.10.16:1883 connected
.SoC: vehicle not available: cannot create vehicle 'template': invalid element 'name: language'
Capacity: 0kWh

andig, 2022-02-17:
Das geht aktuell nicht mit der `template` syntax, Du musst das Auto bitte direkt als 

    type: hyundai

konfigurieren. Die anderen Parameter sollten alle passen.

closed 2022-02-17 by andig

zwiebelxxl, 2022-02-17:
so geht es jetzt!

>
        type: hyundai
        # template: hyundai
>