evcc-io/evcc #2627: New error "MapToStruct: Field not found: MTU" on 0.82 and 0.83

noradtux, 2022-02-17:
**Describe the bug**
Since 0.82 I get a lot of "MapToStruct: Field not found: MTU" messages. Charge control does not work.

**To Reproduce**
Happens immediately after start.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.83 (ca2588a1)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2022/02/17 20:38:49 evcc 0.83 (ca2588a1)
[main  ] INFO 2022/02/17 20:38:49 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/02/17 20:38:49 connecting evcc-1353081476 at tcp://net-svc.lan.tnxip.de:1883
[vehicle] WARN 2022/02/17 20:38:50 cache is deprecated and will be removed in a future release
[lp-1  ] WARN 2022/02/17 20:38:50 poll mode '{connected 20s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2022/02/17 20:38:50 poll interval '20s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
config
======

grid: grid
----------
Power:          910W
Current L1..L3: 3.07A 1.46A 0.12A

pv 0: pv
--------
Power: 0W

battery: battery
----------------
Power: 0W
SoC:   0%

loadpoint 1
===========

charger: nrgkick
----------------
Power:          MapToStruct: Field not found: MTU
Energy:         find device: FlushDevices.RemoveDevice /org/bluez/hci0/dev_74_F6_1C_13_85_7C: Does Not Exist
Current L1..L3: MapToStruct: Field not found: MTU
Charge status:  MapToStruct: Field not found: MTU
Enabled:        MapToStruct: Field not found: MTU
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 20s # control cycle interval

log: info

meters:
- name: grid
  type: custom
  power:
    type: mqtt
    topic: Pwr/meter/MeterPgrid_Tot
    scale: -1
  currents:
  - type: mqtt
    topic: Pwr/meter/MeterIgrid_L1
  - type: mqtt
    topic: Pwr/meter/MeterIgrid_L2
  - type: mqtt
    topic: Pwr/meter/MeterIgrid_L3
- name: pv
  type: custom
  power:
    type: mqtt
    topic: Pwr/inverter_1phase/Pin1
- name: battery
  type: custom
  power:
    type: mqtt
    topic: Pwr/battery/Pba
  soc:
    type: mqtt
    topic: Pwr/battery/Soc

chargers:
- name: nrgkick
  type: nrgkick-bluetooth
  mac: XXXX # BT device MAC address (sudo hcitool lescan)
  pin: XXXX # App PIN number (write protection, ignore leading zeros)

vehicles:
- name: ioniq_ev
  type: custom
  title: Ioniq EV
  capacity: 28 # kWh
  soc:
    type: mqtt
    topic: Car/SOC
  cache: 1m
  onIdentify: # set defaults when vehicle is identified
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%

site:
  title: Home # display name for UI
  meters:
    grid: grid # grid meter
    battery: battery # battery meter
    pvs:
      - pv # list of pv inverters/ meters
  prioritySoC: 90 # give home battery priority up to this soc (empty to disable)
  bufferSoC: 85 # ignore home battery discharge above soc (empty to disable)

loadpoints:
- title: Garage # display name for UI
  charger: nrgkick # charger
  vehicle: ioniq_ev
  mode: pv
  soc:
    poll:
      mode: connected
      interval: 20s
    min: 80 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: true # set true to interpolate between api updates
  resetOnDisconnect: true
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 500 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 32 # maximum charge current (default 16A)

tariffs:
  grid:

mqtt:
  broker: XXXX
  topic: evcc # root topic for publishing, set empty to disable

influx:
  url: XXXX
  database: XXXX
  org: XXXX
  token: XXXX
  interval: 20s

eebus:

messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
[main  ] INFO 2022/02/17 20:32:26 evcc 0.83 (ca2588a1)
[main  ] INFO 2022/02/17 20:32:26 using config file /etc/evcc.yaml
[main  ] INFO 2022/02/17 20:32:26 listening at 0.0.0.0:7070
[mqtt  ] INFO 2022/02/17 20:32:36 connecting evcc-2077752861 at tcp://net-svc.lan.tnxip.de:1883
[mqtt  ] DEBUG 2022/02/17 20:32:36 tcp://net-svc.lan.tnxip.de:1883 connected
[vehicle] WARN 2022/02/17 20:32:37 cache is deprecated and will be removed in a future release
[lp-1  ] WARN 2022/02/17 20:32:37 poll mode '{connected 20s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2022/02/17 20:32:37 poll interval '20s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2022/02/17 20:32:37 site config:
[site  ] INFO 2022/02/17 20:32:37   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/02/17 20:32:37     grid:      power ✓ energy ✗ currents ✓
[site  ] INFO 2022/02/17 20:32:37     pv 0:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/02/17 20:32:37     battery 0: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/02/17 20:32:37 loadpoint 1:
[lp-1  ] INFO 2022/02/17 20:32:37   mode:        pv
[lp-1  ] INFO 2022/02/17 20:32:37   charger:     power ✓ energy ✓ currents ✓ phases ✗
[lp-1  ] INFO 2022/02/17 20:32:37   meters:      charge ✓
[lp-1  ] INFO 2022/02/17 20:32:37     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/02/17 20:32:37   vehicles:    ✓
[lp-1  ] INFO 2022/02/17 20:32:37     vehicle 0: range ✗ finish ✗ status ✗ climate ✗
[lp-1  ] INFO 2022/02/17 20:32:37 vehicle updated: unknown -> Ioniq EV
[lp-1  ] DEBUG 2022/02/17 20:32:37 set charge mode: pv
[lp-1  ] DEBUG 2022/02/17 20:32:37 set target soc: 100
[lp-1  ] ERROR 2022/02/17 20:32:40 charger: MapToStruct: Field not found: MTU
[site  ] DEBUG 2022/02/17 20:32:40 ----
[mqtt  ] DEBUG 2022/02/17 20:32:40 Pwr/meter/MeterPgrid_Tot wait for initial value
[site  ] DEBUG 2022/02/17 20:32:40 grid power: 912W
[site  ] DEBUG 2022/02/17 20:32:40 pv power: 0W
[site  ] DEBUG 2022/02/17 20:32:40 battery power: 0W
[site  ] DEBUG 2022/02/17 20:32:40 grid currents: [3.05 1.48 0.12]A
[site  ] DEBUG 2022/02/17 20:32:40 battery soc 0: 0%
[site  ] DEBUG 2022/02/17 20:32:40 site power: 912W
[lp-1  ] ERROR 2022/02/17 20:32:55 charge meter: MapToStruct: Field not found: MTU
[lp-1  ] ERROR 2022/02/17 20:33:00 charge meter: MapToStruct: Field not found: MTU
[lp-1  ] ERROR 2022/02/17 20:33:06 charger: MapToStruct: Field not found: MTU
[site  ] DEBUG 2022/02/17 20:33:06 ----
[site  ] DEBUG 2022/02/17 20:33:06 grid power: 924W
[site  ] DEBUG 2022/02/17 20:33:06 pv power: 0W
[site  ] DEBUG 2022/02/17 20:33:06 battery power: 0W
[site  ] DEBUG 2022/02/17 20:33:06 grid currents: [3.11 1.48 0.12]A
[site  ] DEBUG 2022/02/17 20:33:06 battery soc 0: 0%
[site  ] DEBUG 2022/02/17 20:33:06 site power: 924W
[lp-1  ] ERROR 2022/02/17 20:33:35 charge meter: MapToStruct: Field not found: MTU
[lp-1  ] ERROR 2022/02/17 20:33:40 charge meter: MapToStruct: Field not found: MTU
[lp-1  ] ERROR 2022/02/17 20:33:45 charger: MapToStruct: Field not found: MTU
</pre></details>

andig, 2022-02-17:
Da stimmt irgendwas mit dem nrgkick-bt nicht. Das müsste sich jemand anschauen der Zugriff auf so ein Teil hat.

noradtux, 2022-02-17:
Ich habe mir gerade mal die loadpoint.go angeschaut. Zumindest eine der Meldungen scheint aus Zeile 1320 zu kommen. Ich habe keinen separaten Zähler für den lp (die Garage) sondern nur den Zähler der site (Haus) und die Daten, die der charger (NRGkick) liefert. Entsprechend ist kein meter für den lp gesetzt.

andig, 2022-02-17:
Die Meldungen kommen vom Charger.

noradtux, 2022-02-18:
Woran es auch liegen mag, ich kann jedenfalls das Projekt bauen und Änderungen testen.

andig, 2022-02-18:
Super. Dann mach mal bitte ein

    go get github.com/muka/go-bluetooth@master

und probier mal obs damit geht.

andig, 2022-02-18:
@noradtux verwendest Du Docker? Falls nein: welche bluez Version hast Du installiert?

noradtux, 2022-02-18:
> Super. Dann mach mal bitte ein
>
> |go get ***@***.*** |
>
> und probier mal obs damit geht.

Keine Änderung im Verhalten.

andig, 2022-02-18:
OK, wie erwartet. Nochmal: welche bluez Version hast Du installiert? Welches OS/Version?

andig, 2022-02-19:
Probier mal bitte Downgrade von bluez auf 5.60- dann wäre zumindest klar woran es liegt. Ich versuche eine Version für 5.63 nachzureichen.

andig, 2022-02-20:
Neuer Versuch: probier mal bitte

     go get -d github.com/muka/go-bluetooth@bluez-5.63

und dann neu bauen.

> Zum Verständnis: war das bei 0.81 nicht der Fall? Oder hast Du bluez aktualisiert?

noradtux, 2022-02-20:
0.81 läuft bestens. Mit deinem Fix scheint 0.83 jetzt auch zu laufen. Die Meldung ist weg und der NGRKick reagiert. Gut, muss ich das BlueZ-Paket doch nicht selbst bauen. Danke :)

andig, 2022-02-20:
Closed in https://github.com/evcc-io/evcc/commit/4823b92dbce5ae5ca17516950e27e5d74fa811db

closed 2022-02-20 by andig