evcc-io/evcc #2628: Invalid scale parameter in Solaredge templates

andig, 2022-02-17:
Ich habe die Konfiguration erneut aufgerufen (evcc configure), mit diesem Ergebnis:

Choose one of the following PV systems, or 'My device is not in this list' if you have none of them

 Choose a PV system: Solaredge Inverter

Please provide the following settings:

 Modbus ID 1
 IP-Address or Hostname 192.168.1.81
 Port 502
 Timeout 3s

Testing the grid configuration of Solaredge Inverter ...
   Error: cannot create meter 'modbus': 1 error(s) decoding:

* '' has invalid keys: scale

 Testing of the grid configuration of Solaredge Inverter failed. Do you want to add it anyway? Yes

Testing the pv configuration of Solaredge Inverter ...

Solaredge Inverter was successfully added.

> Am 17.02.2022 um 21:26 schrieb andig ***@***.***>:
> 
> 
> Und wie hast du jetzt innerhalb von 4min eine Lösung gefunden? Was war die Lösung???
> 
> —
> Reply to this email directly, view it on GitHub <https://github.com/evcc-io/evcc/discussions/2621#discussioncomment-2199971>, or unsubscribe <https://github.com/notifications/unsubscribe-auth/AXYOZAPCXKR5DT6ID7KXUHDU3VKVVANCNFSM5OUQTMWA>.
> Triage notifications on the go with GitHub Mobile for iOS <https://apps.apple.com/app/apple-store/id1477376905?ct=notification-email&mt=8&pt=524675> or Android <https://play.google.com/store/apps/details?id=com.github.android&referrer=utm_campaign%3Dnotification-email%26utm_medium%3Demail%26utm_source%3Dgithub>. 
> You are receiving this because you authored the thread.
>

_Originally posted by @DJPerryCox in https://github.com/evcc-io/evcc/discussions/2621#discussioncomment-2199996_

andig, 2022-02-18:
Vmtl ist https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/solaredge-hybrid.yaml#L36 Das Problem. Hier wird ein `scale` auf einen SunSpec Wert angewandt- ich bin nicht sicher ob das überhaupt vorgesehen ist. Was mich wundert ist, dass wir hier SunSpec und normale Modbus Register in einem Template mischen. Sind die normalen Register überhaupt notwendig wenn der WR SunSpec spricht?

premultiply, 2022-02-18:
Ja, SE mischt da brav.

Und du die Netzeinspeisung hat wohl ein falsches Vorzeichen. Zumindest hatte das jemand reported. Ich kann es selbst nicht prüfen…

andig, 2022-02-18:
Ahh, bingo. Der Fehler ist im Nightly behoben: https://github.com/evcc-io/evcc/pull/2447/files#diff-e3b78373af2864af915083be00578e3ad58a8a71c1bf1d8d51b6842900475b00L26

closed 2022-02-18 by andig