evcc-io/evcc #2632: Mindestladung anhand Reichweite definieren

drivingmx, 2022-02-18:
Ich nutze den Parameter für die Mindestladung um sicherzustellen dass mein Auto immer eine Mindestreichweite darstellen kann.
Also rechne ich überschlagsmäßig wieviel Prozent SOC das ungefähr sein müssen und stelle das in der config ein.

Da sich der Verbrauch im Jahresverlauf bei meinem Auto zwischen 12 und 22 kWh/100km bewegt, führt dieser feste SOC zu starken Abweichungen bei der resultierenden Reichweite. 

Um den Verbrauch immer wieder zu adaptieren müsste ich je nach Wetterlage immer wieder die config anfassen. 

Den Wert immer auf sicher zu stellen, also hoch, würde im Sommer sehr viel Pufferspeicher reservieren, der eigentlich frei bleiben könnte. 

Dieses Problem könnte  evcc lösen indem es neben der mindest SOC Einstellung auch eine Mindestreichweiten Einstellung anbietet. Bei meinem Auto (VW eUp) wird der letzte mittlere Verbrauch bei der Angabe der Restreichweite berücksichtigt. 
Somit kann evcc stoppen wenn die Reichweite reingeladen ist. Im Sommer zB bei 40, im Winter bei 70.

premultiply, 2022-03-03:
Mmmmh. Die Reichweite kann man ja theoretisch über den SoC weiter hochrechnen...
In dem Fall könnte man unter gewissen Umständen am Ziel-SoC-Slider zumindest die ungefähre Reichweite zum SoC anzeigen. Der SoC bleibe dabei aber immer die führende Größe. Sonst müsste man wohl etwas größere Umbauen durchführen.
/cc @naltatis

premultiply, 2022-03-23:
Hilft das so?
https://github.com/evcc-io/evcc/pull/2889#issuecomment-1068448936

drivingmx, 2022-03-23:
Das ist ein sehr guter Anfang.
Für die händisch Justage ist das optimal. Der Dreisatz ist meines Erachtens auch ausreichend.
Für die Automatisierung der Mindestlademenge ist durch diese Erweiterung noch nichts getan.

andig, 2022-04-17:
Ich fänds immer noch eine gute Idee. Zusätzlich wäre es auch Klasse, kWh zu definieren falls die WB einen Zähler hat. Das würde auch bei Fahrzeugen ohne SoC helfen.

drivingmx, 2022-04-17:
Das stimmt.
Die Mindestladung wie sie jetzt ist, ist aus meiner Sicht die größte Schwäche von evcc ohne Anbindung an SOC.
Das würde es rund machen

Headroom42, 2022-05-12:
Hierher verwiesen von @VolkerK62 auch von meiner Seite der Wunsch am Ladepunkt einen festen KW Wert eingeben zu können, der nach erreichen den Charger abschaltet. Gerade für ältere Autos ohne Kommunikationsschnittstelle wichtig, da man sonst immer in mindestens 2 Apps/Websites unterwegs ist.

premultiply, 2022-05-12:
Braucht man die kWh-Einstellung auch bei Fahrzeugen mit SoC oder reichte es dies nur bei Fahrzeugen ohne Livedaten als Rückfallebene zu haben?

Letztendlich könnte man m.M.n. die komplette interne „SoC-Infrastruktur“ nutzen wenn man bei der Zuweisung eines solchen Fahrzeugs (konfiguriert mit Batteriegröße) den Start-SoC immer mit 0% annehmen und dann anhand der geladenen kWh wie gehabt den SoC weiter hochzählen via Estimator.
Somit würde über den SoC-Slider (und MinSoC) indirekt eine kWh-Vorgabe gemacht. Müsste man dann nur in der UI ggf. leicht modifiziert darstellen.

Headroom42, 2022-05-12:
> Braucht man die kWh-Einstellung auch bei Fahrzeugen mit SoC oder reichte es dies nur bei Fahrzeugen ohne Livedaten als Rückfallebene zu haben?

Persönlich würde ich sagen ja, da z.B. mein Ioniq 5 im Carport z.T. keinen Mobilfunkempfang hat und somit nur sporadisch bei Abfrage Daten liefert.

andig, 2022-05-13:
Meeting minutes: Energiemenge wäre ein guter Schritt- sowohl für Begrenzung der Lademenge als auch fürs Zielladen.

Headroom42, 2022-06-08:
Für welches Release ist die Einstellbarkeit der Energiemenge per GUI geplant?

andig, 2022-06-08:
> Für welches Release ist die Einstellbarkeit der Energiemenge per GUI geplant?

Wie immer: wenns fertig ist!

Headroom42, 2022-09-29:
Ich warte immernoch sehnsüchtig auf dieses Feature. Immer selbst den Ur-Ioniq zu monitoren, wann genug kw geladen wurden um 80% zu haben und dann abzuschalten, ist auf Dauer etwas umständlich.
Kann man das einbauen dieses Feature durch eine "Sonderspende" irgendwie beschleunigen?

andig, 2022-09-29:
@Headroom42 Sponsorenwünsche wie Deiner haben Vorrang, wir schauen uns das an!

naltatis, 2022-09-30:
Soo, nu gibts auch eine UI für kWh basierte Limits:
https://github.com/evcc-io/evcc/pull/4639

closed 2022-10-04 by naltatis