evcc-io/evcc #2644: Go-eCharger Status wird 8x pro Sekunde abgefragt

Waldmensch1, 2022-02-20:
**Describe the bug**
Im Trace sieht man, dass in einem EVCC Intervall bis zu 8 Api gleiche Anfragen an den Go.E gesendet werden. Es macht den Eindruck, als ob jeder Wert einzeln abgerufen wird, obwohl mit einem Api Call alle Werte geliefert werden
 
Im Go-E werkelt nur ein ESP8266. Eventuell könnte der etwas überfordet werden, mit so vielen Anfragen

**To Reproduce**
Steps to reproduce the behavior:
1. Starte mit konfiguriertem Go-E EVCC im Trace Modus

**Expected behavior**
Maximal 1-2 API Abfragen (vorher/nachher) beim Setzen von Werten beim Go-E

**EVCC details:**
Show output of <code>evcc -v</code>:
```
<- paste here

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
chargers:
- type: template
  template: go-e-v3
  host: 192.168.178.147  # IP-Adresse oder Hostname
  name: wallbox4

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
[site  ] DEBUG 2022/02/14 13:39:49 ----
[site  ] DEBUG 2022/02/14 13:39:49 grid power: -2030W
[http  ] TRACE 2022/02/14 13:39:49 GET http://192.168.178.155/cm?cmnd=Status%208
[cache ] TRACE 2022/02/14 13:39:49 gridPower: -2030
[http  ] TRACE 2022/02/14 13:39:49 
GET /cm?cmnd=Status%208 HTTP/1.1
Host: 192.168.178.155
User-Agent: Go-http-client/1.1
Accept-Encoding: gzip

HTTP/1.1 200 OK
Connection: close
Transfer-Encoding: chunked
Accept-Ranges: none
Access-Control-Allow-Origin: *
Cache-Control: no-cache, no-store, must-revalidate
Content-Type: application/json
Expires: -1
Pragma: no-cache

ed
{"StatusSNS":{"Time":"2022-02-14T13:39:49","ENERGY":{"TotalStartTime":"2019-03-30T09:09:48","Total":5340.829,"Yesterday":5.699,"Today":3.961,"Power":900,"ApparentPower":900,"ReactivePower":0,"Factor":1.00,"Voltage":214,"Current":4.199}}}
0
[http  ] TRACE 2022/02/14 13:39:49 GET http://192.168.178.193/cm?cmnd=Status%208
[mqtt  ] TRACE 2022/02/14 13:39:49 recv stadtweg/energy/tripower: '{"pwr1":2039,"sw":-2940.847,"pwr2":4979.847}'
[http  ] TRACE 2022/02/14 13:39:49 
GET /cm?cmnd=Status%208 HTTP/1.1
Host: 192.168.178.193
User-Agent: Go-http-client/1.1
Accept-Encoding: gzip

HTTP/1.1 200 OK
Connection: close
Transfer-Encoding: chunked
Accept-Ranges: none
Cache-Control: no-cache, no-store, must-revalidate
[... 161 lines cut ...]
Accept-Encoding: gzip

HTTP/1.1 200 OK
Connection: close
Transfer-Encoding: chunked
Access-Control-Allow-Origin: *
Content-Type: application/json

22a
{"alw":false,"eto":44052,"car":4,"wh":3499.513,"trx":null,"nrg":[233,220,223,0,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"User 1","energy":0,"cardId":true},{"name":"User 2","energy":0,"cardId":false},{"name":"User 3","energy":0,"cardId":false},{"name":"User 4","energy":0,"cardId":false},{"name":"User 5","energy":0,"cardId":false},{"name":"User 6","energy":0,"cardId":false},{"name":"User 7","energy":0,"cardId":false},{"name":"User 8","energy":0,"cardId":false},{"name":"User 9","energy":0,"cardId":false},{"name":"User 10","energy":0,"cardId":false}]}
0

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
<- paste here

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>
<- paste here

</pre></details>

andig, 2022-02-20:
Ist das denn ein Problem?

Waldmensch1, 2022-02-20:
Das weiß ich nicht. Zumindest ist es ungewöhnlich. Gibt es denn einen Grund für die Anfragen Attacke?

andig, 2022-02-20:
Works as designed. Ich mache hier mangels erkennbarem Problem zu. Sicher könnte man auch ein Caching einbauen- ist aber keine Priorität.

closed 2022-02-20 by andig

Waldmensch1, 2022-02-20:
Okay, hab mir selbst was quick'n dirty reingewurstelt

```
var cnt = 0
var s = (*StatusResponse2)(nil)
// Status reads a v1/v2 api response
func (c *LocalAPI) Status() (Response, error) {
	if c.v2 {
		if cnt == 0 {
			fmt.Println("------------------", "Request", "----------------------")
			res := new(StatusResponse2)
			err := c.response("status?filter=alw,car,eto,nrg,wh,trx,cards,fsp", &res)
			s = res
			cnt = 1

			t1 := time.NewTimer(2 * time.Second)
			go func() {
				<-t1.C
				cnt = 0
				fmt.Println("------------------", "Timer expiered", "----------------------")
			}()

			return res, err
		} else {
			fmt.Println("------------------", "Cache", cnt, "----------------------")
			res := s
			cnt += 1
			return res, nil
		}
	}

	res := new(StatusResponse)
	err := c.response("status", &res)
	return res, err
}
```

Log:
```
[site  ] DEBUG 2022/02/20 13:41:22 ----
[site  ] DEBUG 2022/02/20 13:41:22 grid power: 1912W
[site  ] DEBUG 2022/02/20 13:41:22 pv power: 1058W
[cache ] TRACE 2022/02/20 13:41:22 gridPower: 1912
[cache ] TRACE 2022/02/20 13:41:22 pvPower: 1058.4279999999999
[site  ] DEBUG 2022/02/20 13:41:22 battery power: -3W
[cache ] TRACE 2022/02/20 13:41:22 batteryPower: -3.4280000000000004
[site  ] DEBUG 2022/02/20 13:41:22 grid currents: [9.44 1.26 1.72]A
[site  ] DEBUG 2022/02/20 13:41:22 battery soc 0: 19%
[site  ] DEBUG 2022/02/20 13:41:22 giving priority to battery charging at soc: 19
[site  ] DEBUG 2022/02/20 13:41:22 site power: 2412W
------------------ Request ----------------------
[cache ] TRACE 2022/02/20 13:41:22 gridCurrents: [9.436 1.26 1.717]
[cache ] TRACE 2022/02/20 13:41:22 gridEnergy: 26538
[cache ] TRACE 2022/02/20 13:41:22 batterySoC: 19
[cache ] TRACE 2022/02/20 13:41:22 lp-1/mode: pv
[go-e  ] TRACE 2022/02/20 13:41:22 GET http://192.168.178.147/api/status?filter=alw,car,eto,nrg,wh,trx,cards,fsp
[go-e  ] TRACE 2022/02/20 13:41:22
GET /api/status?filter=alw,car,eto,nrg,wh,trx,cards,fsp HTTP/1.1
Host: 192.168.178.147
User-Agent: Go-http-client/1.1
Accept: application/json
Accept-Encoding: gzip

HTTP/1.1 200 OK
Connection: close
Transfer-Encoding: chunked
Access-Control-Allow-Origin: *
Content-Type: application/json

235
[... 16 lines cut ...]
------------------ Cache 5 ----------------------
[lp-1  ] DEBUG 2022/02/20 13:41:22 max charge current: 0A = 0A + -10.5A (2412W @ 1p)
[cache ] TRACE 2022/02/20 13:41:22 lp-1/charging: false
[cache ] TRACE 2022/02/20 13:41:22 lp-1/enabled: false
------------------ Cache 6 ----------------------
[lp-1  ] DEBUG 2022/02/20 13:41:22 charge power: 0W
------------------ Cache 7 ----------------------
[cache ] TRACE 2022/02/20 13:41:22 lp-1/chargePower: 0
[lp-1  ] DEBUG 2022/02/20 13:41:22 charge currents: [0 0 0]A
[cache ] TRACE 2022/02/20 13:41:22 homePower: 2967
[cache ] TRACE 2022/02/20 13:41:22 lp-1/chargeCurrents: [0 0 0]
[mqtt  ] TRACE 2022/02/20 13:41:22 recv stadtweg/energy/sma_em: '{"regard":1925,"surplus":0,"L1":9.369,"L2":1.2530000000000001,"L3":1.353,"energy":26538,"merge":1925}'
[mqtt  ] TRACE 2022/02/20 13:41:23 recv stadtweg/energy/sma_em: '{"regard":2123,"surplus":0,"L1":9.427,"L2":1.246,"L3":1.342,"energy":26538,"merge":2123}'
------------------ Timer expiered ----------------------
[mqtt  ] TRACE 2022/02/20 13:41:24 recv stadtweg/energy/tripower: '{"pwr1":826,"soc":19,"sw":1.7140000000000002,"pwr2":824.286,"pvgarten":79,"pvgrowatt":146}'
[mqtt  ] TRACE 2022/02/20 13:41:24 recv stadtweg/energy/sma_em: '{"regard":2116,"surplus":0,"L1":9.417,"L2":1.241,"L3":1.717,"energy":26538,"merge":2116}'
[mqtt  ] TRACE 2022/02/20 13:41:25 recv stadtweg/energy/sma_em: '{"regard":2118,"surplus":0,"L1":9.413,"L2":1.246,"L3":1.699,"energy":26538,"merge":2118}'
[mqtt  ] TRACE 2022/02/20 13:41:26 recv stadtweg/energy/sma_em: '{"regard":2141,"surplus":0,"L1":9.415000000000001,"L2":1.246,"L3":1.718,"energy":26538,"merge":2141}'
[mqtt  ] TRACE 2022/02/20 13:41:27 recv stadtweg/energy/sma_em: '{"regard":2122,"surplus":0,"L1":9.515,"L2":1.234,"L3":1.718,"energy":26538,"merge":2122}'
[mqtt  ] TRACE 2022/02/20 13:41:28 recv stadtweg/energy/sma_em: '{"regard":2114,"surplus":0,"L1":9.427,"L2":1.234,"L3":1.718,"energy":26538,"merge":2114}'
[mqtt  ] TRACE 2022/02/20 13:41:29 recv stadtweg/energy/tripower: '{"pwr1":827,"soc":19,"sw":-5.142,"pwr2":832.142,"pvgarten":79,"pvgrowatt":146}'
[mqtt  ] TRACE 2022/02/20 13:41:29 recv stadtweg/energy/sma_em: '{"regard":2142,"surplus":0,"L1":9.434000000000001,"L2":1.234,"L3":1.702,"energy":26538,"merge":2142}'
[mqtt  ] TRACE 2022/02/20 13:41:30 recv stadtweg/energy/sma_em: '{"regard":2153,"surplus":0,"L1":9.52,"L2":1.249,"L3":1.707,"energy":26538,"merge":2153}'
[mqtt  ] TRACE 2022/02/20 13:41:31 recv stadtweg/energy/sma_em: '{"regard":2139,"surplus":0,"L1":9.541,"L2":1.25,"L3":1.725,"energy":26538,"merge":2139}'
```