evcc-io/evcc #2651: Support für Corrently / Stromdao Stromanbieter

Hofyyy, 2022-02-20:
**Is your feature request related to a problem? Please describe.**
Nein

**Describe the solution you'd like**
Ich würde gerne in Zukunft zu dem Anbieter Corrently wechseln,
welcher einen Tarif auf Stundengenauer Abrechnung anbietet,
und dort so "sinnvoll" wie möglich im günstigen / grünen Zeitfenster automatisch laden.

**Additional context**
- Das API soll ähnlich oder identisch zu "awattar" sein, welches schon unterstützt ist.
- Corrently.io (2.0.0) API Beschreibung: https://corrently.io/ 

Ich bin mit dem Anbieter in engem Austausch und könnte ggf. einen Testaccount besorgen, wenn nötig.

Viele Grüße

andig, 2022-02-20:
Ich hab da kein Preis Api gefunden. Gibts eins?

Hofyyy, 2022-02-20:
Der Interessante API Call ist glaube ich:
- Marketdata

Hofyyy, 2022-02-20:
Bei denen geht es darum wieviel Ökostrom gerade zu der Stunde in der jeweiligen Postleitzahl unterwegs ist.
Dabei ist interessant das viel Ökostrom == Günstig ist. Es ist also für die Planung gleichbedeutend.

Hofyyy, 2022-02-20:
Sehen kann man das in dem folgenden UI.
- Einfach einmal hier hingehen und eine beliebige Postleitzahl eingeben: https://gruenstromindex.de/

andig, 2022-02-20:
https://corrently.io/#operation/gsiMarketdata sieht gut aus. Inwiefern das als "Preis" auch bei Dir als Kunde ankommt kann ich nicht beurteilen.

Hofyyy, 2022-02-20:
Wenn das als Werbung gewertet wird kann ich das folgende gerne löschen.
Ich bin noch. nicht bei dem Anbieter und prüfe alles was ich finde, aber Sie haben ein Alleinstellungsmerkmal

In der folgenden Grafik sieht man, das man "Punkte" Sammelt um so mehr Grünen Strom man verbraucht. Diesen Bonus kann man in Bäume spenden, in Gutschrift Geld bekommen, oder in eine Solaranlage (Solarpark) umwandeln, die ab dann für immer für einen Arbeitet.

Leider ist der Anbieter noch nicht gut darin das auch in "Realen Prozenten" durchzurechnen und einem an die Hand zu geben. Aber da arbeiten Sie anscheinend dran und sind per Email responsive. Für den Solaranwendungsfall kann ich mehr Auskünfte geben wenn wer möchte.

Das interessant ist nämlich, das man den Grünstrombonus nicht nur ansparen, sondern auch kaufen kann. Ich kann mir also "direkt" Solarzellen in einem Solarpark kaufen, die mir wiederum eine Pauschale KwH pro Jahr zur verfügung stellen. Aktuell geht das für "4,5 Euro" für 1 KwH im Jahr (Dauerhaft). Nachteil: 100% Kundenbindung ab dem Moment, und das Risiko das der Anbieter mal pleite geht.

[image]

andig, 2022-02-20:
Ich mach mal eine Diskussion draus. PR welcome!