evcc-io/evcc #2659: SOC Interpolation

MarkusGH, 2022-02-21:
Vorweg: Kompliment - Spitzenprojekt!

Ich habe hier einen Cayenne E-Hybrid an einem PMCC und betreibe PV Überschussladen.

evcc Version 0.83

Die SOC Interpolation ermittelt bei dieser Konstellation übermäßig optimistische Werte.

Problem dürften die überschaubare Ladeleistung in Verbindung mit dem Eigenverbrauch des Fahrzeugs sein.

Man erreicht bei minimaler Ladeleistung und kalten Außentemperaturen nur eine Effizienz von 70%, bei höherer Ladeleistung wird diese besser.

Vorschlag:

Zusätzliche Parameter:

vehicles.soc.estimate.chargeefficiency # Ersatz für estimator.go : const chargeEfficiency
vehicles.soc.estimate.baselineconsumption # Verbrauch des Fahrzeugs im Lademodus

Default: chargeefficiency 0.9, baselineconsumption 0.0

Alternativ Konfiguration pro Loadpoint (evtl. zusätzlich für unbekannte Fahrzeuge), das wäre besser als nichts, bildet die Realität aber schlechter ab.

andig, 2022-02-21:
Zusätzliche Konfigurationsparameter sind kein Weg den wir gehen wollen. Alternativ Schätzung deaktivieren?

MarkusGH, 2022-02-21:
Euer Projekt, eure Entscheidung.
Aber eine feste chargeEfficiency von 0.9 kann eigentlich nicht der richtige Weg sein, jedes Fahrzeug und jeder Ladepunkt ist anders, spätestens bei DC Laden habe ich auch Verluste in der Wallbox.