evcc-io/evcc #2660: Falscher Charger Status durch falsche Ladeströme  bei PMCC

MarkusGH, 2022-02-21:
evcc version 0.83 (7a99de71)
PMCC (Porsche Mobile Charger Connect)
Cayenne E-Hybrid

Von Zeit zu Zeit meldet der PMCC offenbar falsche Werte für charge power und charge currents was dann entsprechende Folgen für den Ladevorgang hat weil der charger status dann ohne echten Grund auf B oder C wechselt.

Folgeproblem ist dass der Timer dadurch resettet wird und das Webinterface falsche Werte zeigt.

Im angehängten Logfile sieht man sehr schön, wie der charger status viel zu schnell wechselt.
Ob der PMCC hierdurch von evcc An-/Ausgeschaltet wird kann ich aktuell nicht sagen.

==============================================================

evcc.yaml:

uri: 0.0.0.0:7070 # uri for ui
interval: 1s # control cycle interval

log: info
levels:
  porsche: debug
  lp-1: debug

mqtt:
  broker: mqtt.dmstools.de:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.1942 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- type: template
  template: mqtt-meter
  usage: grid
  power: mbmd/solaredge1-1-1/Power
  scalepower: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
  energy: mbmd/solaredge1-1-1/Import
  scaleenergy: 1  # Die Energiemenge wird in kWh erwartet. Falls z.b. Wh geliefert wird, kann dies hier mit dem Wert 0.001 angepasst werden
  currentp1: mbmd/solaredge1-1-1/Current/L1
  currentp2: mbmd/solaredge1-1-1/Current/L2
  currentp3: mbmd/solaredge1-1-1/Current/L3
  timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  name: grid1
- type: template
  template: mqtt-meter
  usage: pv
  power: mbmd/solaredge1-1/Power
  scalepower: 1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
  energy: mbmd/solaredge1-1/Export
  scaleenergy: 1  # Die Energiemenge wird in kWh erwartet. Falls z.b. Wh geliefert wird, kann dies hier mit dem Wert 0.001 angepasst werden
  currentp1: mbmd/solaredge1-1/Current/L1
  currentp2: mbmd/solaredge1-1/Current/L2
  currentp3: mbmd/solaredge1-1/Current/L3
  timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  name: pv1

chargers:
- type: template
  template: pmcc
  ski: <zensiert>
  enforcePVLimits: true  # Verwende PV Laden über eine Ladebegrenzung (wie herkömmliche Wallboxen) bei Fahrzeugen mit Unterstützung von ISO15118
  name: wallbox1

vehicles:
- type: template
  template: porsche
  title: Das Batmobil  # Wird in der Benutzeroberfläche angezeigt
  user: <zensiert>
  password: <zensiert>
  vin: <zensiert>
  capacity: 17.9  # Akku-Kapazität in kWh
  name: ev1

loadpoints:
- title: PMCC
  charger: wallbox1
  vehicles:
  - ev1
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 30m
    min: 0
    target: 100
    estimate: true
  phases: 1
  mincurrent: 2
  maxcurrent: 20
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 5m
  guardduration: 10m

site:
  title: H73
  meters:
    grid: grid1
    pvs:
    - pv1

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----

==============================================================

evcc dump:

[main  ] INFO 2022/02/21 13:34:15 evcc 0.83 (7a99de71)
[main  ] INFO 2022/02/21 13:34:15 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/02/21 13:34:15 connecting evcc-888687057 at tcp://mqtt.dmstools.de:1883
[lp-1  ] WARN 2022/02/21 13:34:32 poll interval '30m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
config
======

grid: grid1
-----------
Power:          3638W
Energy:         7948.5kWh
Current L1..L3: 2A 17.8A 1.4A

pv 0: pv1
---------
Power:          1808W
Energy:         10262.6kWh
Current L1..L3: 2.67A 2.66A 2.64A

loadpoint 1
===========

charger: wallbox1
-----------------
[eebus ] ERROR 2022/02/21 13:35:14 failed to send current limit data:  no eletrical paramaters, measurements, or limits available yet
[eebus ] WARN 2022/02/21 13:35:14 !!  2022-02-21 13:35:14  ev-charger-communication changed from  unknown  to  iec61851
Power:              offline
Current L1..L3:     offline
...............SoC: not available
Charge status:      C
Enabled:            true
Charged:            9.8kWh

vehicle 0
---------
SoC:            61%
Charge status:  C
Range:          22km
Odometer:       1563km
Finish time:    2022-02-21 17:08:00 +0100 CET
Climate active: false
Outside temp:   20.0°C
Target temp:    20.0°C
Capacity:       17kWh

----------------------

Logfile:

[putty.log](https://github.com/evcc-io/evcc/files/8109191/putty.log)

andig, 2022-02-21:
Ohne ins Logfile geschaut zu haben:

> Von Zeit zu Zeit meldet der PMCC offenbar falsche Werte für charge power und charge currents was dann entsprechende Folgen für den Ladevorgang hat weil der charger status dann ohne echten Grund auf B oder C wechselt.

Was sollten wir in diesem Fall tun? Wir müssen uns auf die Meßwerte verlassen können...

MarkusGH, 2022-02-21:
Ich weiß nicht wo genau das Problem liegt, da ich in die eebus Kommunikation keinen Einblick habe.
Ich sehe nur dass die Ladeströme im Log nicht zu stimmen scheinen.
Wo genau das Problem liegt kann ich nicht sagen, eventuell hat ja die eebus Anbindung einen Fehler?
Kann man da evtl. einen Trace aktivieren?

MarkusGH, 2022-02-21:
Passiert auch mit interval: 10s

RTTTC, 2022-02-21:
I would try 60s to eliminate this question.

MarkusGH, 2022-02-21:
I would suggest to address the underlying problem.
I am happy to assist.
Is there any way to get a trace of the eebus communication?

MarkusGH, 2022-02-21:
OK, found it. I suppose it is the topic eebus :-) Will switch it to trace and investigate further.

RTTTC, 2022-02-22:
Could you produce and post a log file using `interval 60s`? Would be interesting to see. 
Thanks!

andig, 2022-02-22:
@RTTTC please don't confuse the discussion. Indepdendent of the interval, currents are always measured, so are powers. Let's please understand first what is actually going on by looking into the logs. Unfortunately the eebus messages are a bit involved...

andig, 2022-02-26:
> Das liegt nicht an falschem Daten, sondern am Handling von EVCC dass es enable() und enabled() nicht in EEBUS und den Wallboxen gibt. D.h. diese müssen irgendwie simuliert werden. Das ist keine Fehler im EEBUS Code. 

@MarkusGH das werden wir leider nicht kurzfristig lösen können da es ein grundsätzliches Redesign erfordert. Ich mache daher hier erstmal zu.

closed 2022-02-26 by andig

MarkusGH, 2022-03-06:
Ich komme leider erst jetzt dazu zu antworten. 

Ich würde Dich bitten das issue wieder zu öffnen, da ich jetzt ein Log liefern kann das m. E. belegt dass es doch ein Problem mit der eebus Kommunikation bzw. dem Handling kurzfristiger Verbindungsstörungen gibt.
Anbei ein Log von heute. [putty.log](https://github.com/evcc-io/evcc/files/8192410/putty.log)

Fahrzeug lädt, zuletzt mit 1P@20A, Kommunikation soweit unauffällig. Um 11:44:17 kommt aus heiterem Himmel die Meldung:
_[eebus ] TRACE 2022/03/06 11:44:17 verifying client ski: bda90ea5fab0a1a4211db2d331b0eadb582b1e83_

Das löst eine ganze Reihe von Problemen aus:

1) In der Folge wird kurz mal ein Ladestrom von 0A angenommen, der charger status wechselt deswegen kurz C->B->C.
2) Der soc estimator sieht wohl bereits kurz nach Beginn des Ladevorgangs bereits _charged energy:  10380 Wh_ und geht deswegen wenig später von einem Ladezustand von 100% aus.
3) (Mutmaßung von mir) Alle Timer die den Ladevorgang steuern werden wohl neu gestartet.

Ob ein Fix ein grundsätzliches Redesign erfordert kann ich nicht beurteilen, ich könnte mir jedoch vorstellen, dass dieses Problem durch ein besseres Handling kurzfristiger Verbindungsstörungen im eebus Code vermieden werden könnte.

andig, 2022-03-07:
@MarkusGH "verifying client ski" bedeutet, dass der Charger eine neue TCP Verbindung zum EEBus Server aufgemacht hat. Als scheint als hätte er diese vorher verloren oder war aus anderen Gründen unglücklich.

> Das löst eine ganze Reihe von Problemen aus

Aussicht evcc ist das ein weiteres Fahrzeug mit der gleichen ID. Das müsste in der internen Logik vmtl anders behandelt werden als bisher.

> Ob ein Fix ein grundsätzliches Redesign erfordert kann ich nicht beurteilen, ich könnte mir jedoch vorstellen, dass dieses Problem durch ein besseres Handling kurzfristiger Verbindungsstörungen im eebus Code vermieden werden könnte.

> Ich würde Dich bitten das issue wieder zu öffnen

Leider habe ich da keine guten Nachrichten. Derzeit haben wir keinen Maintainer für den EEBus Code und werden daher absehbar auch keine Fixes mehr erstellen/ können. Falls das etwas wäre das Dich zum Einstieg interessiert mache ich gerne wieder auf. Anderenfalls muss es leider vorerst bei Wonfix bleiben :(

DerAndereAndi, 2022-03-07:
Das Problem hier kann nur durch ein stabilere Verbindung zur Wallbox gelöst werden, hat nichts mit der Implementierung zu tun.

Die WLAN Verbindung vom PMCC ist leider nicht sehr stabil und es gibt immer wieder Fälle wo dieser zu lange für eine neue Verbindung braucht oder gar keine Verbindung mehr aufbaut, z.b. in Verbindung mit Mesh Netzwerken die Frequenzen ändern. EEBUS erfordert aber einen Ping der jede Sekunde auch beantwortet wird. D.h. jede Form von (kurzzeitigem) Verbindungsverlust ist schlecht.

Lösung generell mit EEBUS: immer eine Kabelgebundene Verbindung zur Wallbox verwenden. Beim PMCC bedeutet das über PLC (Power Line Connection)

MarkusGH, 2022-03-07:
Achja:

> Lösung generell mit EEBUS: immer eine Kabelgebundene Verbindung zur Wallbox verwenden. 
> Beim PMCC bedeutet das über PLC (Power Line Connection)

Kann man für den PMCC einen beliebigen Powerline Adapter verwenden um das Ding ins LAN zu bringen?

DerAndereAndi, 2022-03-08:
Im Prinzip ja. Nur so neumodisches superhyperschnelle Geräte wie die devolo Magic gehen nicht. Die Verbindung läuft sowieso nur mit 10MBit, aufpassen dass die Distanz zum PMCC nicht zu lange ist, der sendet nicht so stark wie übliche Haushalts-PLC-Geräte.

RTTTC, 2022-03-20:
@MarkusGH I am curious - what `interval` do you use lately ?

My (slow) system goes through complete cycle in 4 to 12 seconds using `evcc --log trace`  
And once in a while another modbus TCP system conflict happens, so 10s timeout gets added.
In the end I am running `timeout: 30s` to be on the safe side.

MarkusGH, 2022-03-20:
@RTTTC: Currently I am using 10s in a try to avoid issues with my charger connection. Before that I used 1s (which did work more or less the same). Still investigating...