evcc-io/evcc #2669: configure: Sponsortoken not added to config in advanced mode

andig, 2022-02-21:
Auffällig war noch, dass ich in der EVCC.yaml meinen Sponsoringtoken nicht gefunden habe, obwohl ich ihn während der Konfiguration eingegeben habe.

_Originally posted by @ABBolle in https://github.com/evcc-io/evcc/discussions/2668_

ThiloBaWue, 2022-02-24:
Möchte noch hinzufügen, dass dies Problem auch im "Simple"-Mode auftritt!

andig, 2022-03-02:
Könnt ihr mir mal bitte den Flow zeigen? In meinem Durchlauf im simple mode kommt das Token:

```
Die nächsten Schritte führen durch die Einrichtung einer Konfigurationsdatei für evcc.
Beachte dass dieser Prozess nicht alle möglichen Szenarien berücksichtigen kann.
Durch Drücken von CTRL-C kann der Prozess abgebrochen werden.

ACHTUNG: Diese Funktionalität hat experimentellen Status!
  D.h. es kann möglich sein, dass die hiermit erstellte Konfigurationsdatei
  in einem Update nicht mehr funktionieren könnte und neu erzeugt werden müsste.
  Wir freuen uns auf euer Feedback auf https://github.com/evcc-io/evcc/discussions/

Auf geht`s:

 In welchem Modus soll die Konfiguration durchgeführt werden? Standard Modus (So einfach und schnell wie möglich)

- Hausinstallation einrichten

Wähle eines der folgenden PV Komplettsysteme aus, oder 'Mein Gerät ist nicht in der Liste' falls keines dieser Geräte vorhanden ist

 Wähle ein PV System: Mein Gerät ist nicht in der Liste

 Möchtest du einen Netz-Stromzähler hinzufügen? No

 Möchtest du einen PV Wechselrichter (oder entsprechender Stromzähler) hinzufügen? No

 Möchtest du einen Batterie Wechselrichter (oder entsprechender Stromzähler) hinzufügen? No

 Möchtest du ein Fahrzeug hinzufügen? No

- Ladepunkt(e) einrichten
 Titel des Ladepunktes Garage

- Konfiguration Wallbox

 Wähle eine Wallbox: ABL eMH
-- Sponsorship -----------------------------

Dieses Gerät benötigt ein Sponsorship von evcc. Wie das funktioniert und was ist, findest du hier: https://docs.evcc.io/docs/sponsorship

 Bist du ein Sponsor? Yes
 Bitte gib das Sponsortoken ein [? for help] **********

--------------------------------------------

- Standort einrichten
 Titel des Standortes Mein Zuhause

Die Konfiguration wurde erfolgreich in der Datei evcc.yaml gespeichert

~/htdocs/evcc feature/template-tests 39s
❯ cat evcc.yaml
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: info

site:
  title: Mein Zuhause
  meters:

sponsortoken: ldfkgjdflk
```

andig, 2022-03-03:
Fixed in https://github.com/evcc-io/evcc/commit/ed4fcdffcc812f55e27e61a584db48e7609ddd64

closed 2022-03-03 by andig