evcc-io/evcc #2702: Nissan Leaf pre 2019 (carwings) shows constantly 1% SOC

coolchip, 2022-02-25:
Instead of the current SOC, evcc shows constantly 1%. No matter what the current charging state of the car is. The Nissan Connect EV an the "My Leaf" mobile app are showing the right value.
The remaining range and the activity of the heater are correctly displayed in evcc.

I expect to see the right SOC in the evcc web ui.

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
vehicles:
- type: template
  template: carwings
  title: Nissan Leaf  # Wird in der Benutzeroberfläche angezeigt
  user: xxxx # Benutzername um auf das Gerät zuzugreifen
  password: xxxx # Passwort des Benutzerkontos
  vin: xxxx  # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
  #  minSoC: 25  # Lade sofort mit maximaler Geschwindigkeit bis zu dem angegeben Ladestand, wenn der Lademodus nicht auf 'Aus' steht  # Optional
  targetSoC: 80  # Bis zu welchem Ladestand (SoC) soll das Fahrzeug geladen werden  # Optional
  capacity: 24  # Akku-Kapazität in kWh
  name: ev4
</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
Feb 25 08:17:23 odroid evcc[25321]: [main  ] INFO 2022/02/25 08:17:23 evcc 0.83 (7a99de71)
Feb 25 08:17:23 odroid evcc[25321]: [main  ] INFO 2022/02/25 08:17:23 using config file /etc/evcc.yaml
Feb 25 08:17:23 odroid evcc[25321]: [main  ] INFO 2022/02/25 08:17:23 listening at 0.0.0.0:7070
[...]
Feb 25 08:17:30 odroid evcc[25321]: [lp-1  ] INFO 2022/02/25 08:17:30 vehicle updated: unknown -> Nissan Leaf
[...]
Feb 25 08:18:09 odroid evcc[25321]: [lp-1  ] DEBUG 2022/02/25 08:18:09 vehicle soc: 1%
</pre></details>

andig, 2022-02-25:
Meld Dich mal bitte mit Zugangsdaten bei info@evcc.io

coolchip, 2022-02-25:
Fehler steckt offensichtlich im Modul für carwings

https://github.com/joeshaw/carwings/issues/45

andig, 2022-02-25:
Der ist aber auch wirklich prähistorisch ;)

coolchip, 2022-02-28:
Danke für den schnellen Fix. Jetzt funktioniert alles.  ❤️