evcc-io/evcc #2730: Estimator - Fehlermeldungen im UI bezgl. NaN

Waldmensch1, 2022-02-27:
### Discussed in https://github.com/evcc-io/evcc/discussions/2701

<div type='discussions-op-text'>

<sup>Originally posted by **Waldmensch1** February 25, 2022</sup>
Ich habe gestern mal mit loadpoint.soc.estimate = true gespielt. Über Nacht hatte ich dann diese Fehler im UI. Keine Ahnung, ob estimate dafür ursächlich war, aber der NaN Fall sollte grundsätzlich abgefangen (eventuell auf 0 oder null gesetzt werden) und nicht ans UI gehen. Das mündet sonst in einem invaliden JSON Objekt.

Codebasis: aktueller Master Branch

[image]

Nachtrag: es liegt nicht an estimate. Das habe ich jetzt auf false. Der Fehler kommt trotzdem. Muss also am aktuellen Master liegen. Vehicle wird über TRONITY abgefragt. Funktionierte schon eine ganze Weile reibungslos 

[image]

Nachtrag2: Wenn man den kompletten soc: parameter (mit unterparametern) aus dem Loadpoint entfernt, scheint der Fehler nicht aufzutreten.</div>

premultiply, 2022-02-28:
Ich kann mir auf einen flüchtigen Blick in den Estimator-Code momentan noch nicht vorstellen woher da NaN kommen sollte.
Wie sieht denn die zugehörige Konfig aus?

andig, 2022-02-28:
Hilfreich wäre vllt auch ein komplettes Logfile damit wir sehen wo die ersten NaNs auftauchen.

Waldmensch1, 2022-02-28:
@andig da ist im Trace leider nichts zu sehen, oder ich habe es nicht erwischt, weil es in zeitlich großem Abstand auftritt. (Ca. 1h, s. 1. Screenshot). Ich werde über den Tag den evcc nochmal auf auf dem PC starten und den Trace in ein File loggen, um das vielleicht doch zu schnappen.

@premultiply die configuration ist Standard TRONITY, die funktioniert auch. Auffällig im 2. Screenshot ist, dass es das große JSON Objekt ist. Normal kommen über den Websocket ja Einzelwerte. Die Loadpoint config editiere ich hier gleich noch vom PC aus rein

andig, 2022-02-28:
Kannst Du mal in `/api/status` rein schauen wenn das passiert? Das ist ein kompletter Abzug des ganzen Caches.

premultiply, 2022-02-28:
Und bitte auch die Fahrzeug-Konfig. Z. B. `capacity` ist da relevant.

Waldmensch1, 2022-02-28:
@premultiply capacity ist gar nicht gesetzt. Vieleicht ist es das schon?

```
- name: enyaq
  type: tronity
  title: Enyaq
  credentials:
    id: 299e4759-0732-416c-8984-xxxxxxxxxxx
    secret: eb76070b-1672-4d1f-xxxxxxxxxxxxxxx
  capacity: # optional
  vin: #optional
```

premultiply, 2022-02-28:
Ah, ja, genau. Da wird das herkommen!

Waldmensch1, 2022-02-28:
Solltest Du dann evtl. abfangen, da der Parameter laut Doku optional ist. Ich lass es jetzt erstmal über den Tag laufen mit full trace 2 file. Wobei im Estimator keine TRACE ausgaben sind. Die laufen alle unter DEBUG

premultiply, 2022-02-28:
Jo, das hatte ich nicht auf dem Schirm dass das 0 sein könnte.
Würde dann auch IMHO eigentlich weiter vorne abgefangen gehört, oder @andig?

Waldmensch1, 2022-02-28:
Der Vollständigkeit halber, das NaN kommt direkt nach der Tronity response

```
[lp-1  ] DEBUG 2022/02/28 11:19:22 charger status: B
[cache ] TRACE 2022/02/28 11:19:22 lp-1/chargeDuration: 0s
[cache ] TRACE 2022/02/28 11:19:22 lp-1/connected: true
[cache ] TRACE 2022/02/28 11:19:22 lp-1/charging: false
[cache ] TRACE 2022/02/28 11:19:22 lp-1/enabled: false
[tronity] TRACE 2022/02/28 11:19:22 POST https://api-eu.tronity.io/oauth/authentication
[tronity] TRACE 2022/02/28 11:19:23 {"client_id":"***","client_secret":"***","grant_type":"app"}
--
{"id":"fe544e35-2df8-4f0e-9b92-3d1236a77e85","access_token":"xxxxxxxxxxxxxxxxx","expires_in":3600,"token_type":"bearer"}
[tronity] TRACE 2022/02/28 11:19:23 GET https://api-eu.tronity.io/v1/vehicles/6207aee48xxxxxx0012c430ff/bulk
[tronity] TRACE 2022/02/28 11:19:23 {"odometer":0,"range":168,"level":47,"charging":"Complete","plugged":true,"chargerPower":0,"chargeRemainingTime":null,"latitude":null,"longitude":null,"timestamp":1646043486619}
[lp-1  ] DEBUG 2022/02/28 11:19:23 soc estimated: NaN% (vehicle: 47.00%)
[lp-1  ] DEBUG 2022/02/28 11:19:23 vehicle soc: NaN%
[lp-1  ] DEBUG 2022/02/28 11:19:23 vehicle range: 168km
[cache ] TRACE 2022/02/28 11:19:23 lp-1/vehicleSoC: NaN
[cache ] TRACE 2022/02/28 11:19:23 lp-1/chargeRemainingEnergy: NaN
[cache ] TRACE 2022/02/28 11:19:23 lp-1/vehicleRange: 168
[lp-1  ] DEBUG 2022/02/28 11:19:23 vehicle odometer: 0km
[cache ] TRACE 2022/02/28 11:19:23 lp-1/vehicleOdometer: 0
[lp-1  ] DEBUG 2022/02/28 11:19:23 max charge current: 0A = 0A + -0.609A (140W @ 1p)
[lp-1  ] DEBUG 2022/02/28 11:19:23 charge power: 0W
[lp-1  ] DEBUG 2022/02/28 11:19:23 charge currents: [0 0 0]A
[cache ] TRACE 2022/02/28 11:19:23 lp-1/chargePower: 0
[cache ] TRACE 2022/02/28 11:19:23 lp-1/chargeCurrents: [0 0 0]
[cache ] TRACE 2022/02/28 11:19:23 homePower: 805.0000000000002
[mqtt  ] TRACE 2022/02/28 11:19:25 recv stadtweg/energy/tripower: '{"smaem":{"regard":0,"surplus":405,"L1":1.223,"L2":2.076,"L3":1.4000000000000001,"energy":26588,"merge":-405},"pwr1":560,"soc":15,"sw":-1456.068,"pwr2":2016.068,"pvgarten":227,"pvgrowatt":391}'
[mqtt  ] TRACE 2022/02/28 11:19:30 recv stadtweg/energy/tripower: '{"smaem":{"regard":0,"surplus":414,"L1":1.241,"L2":2.086,"L3":1.363,"energy":26588,"merge":-414},"pwr1":574,"soc":15,"sw":-1479.9940000000001,"pwr2":2053.994,"pvgarten":227,"pvgrowatt":391}'
[site  ] DEBUG 2022/02/28 11:19:32 ----
```

PR #2732 (2022-02-28): Fix NaN if vehicle.capacity is not set
Fixes #2730
commit bf169061 (2022-02-28): Fix NaN if vehicle.capacity is not set (#2732)

Fixes #2730
closed 2022-02-28 by andig