evcc-io/evcc #2733: PV Meter login timeout

vfricke, 2022-02-28:
SMA TRIPOWER inverter is connected via Ethernet / PowerLan to Router. A login timeout is reported as:  
updating pv meter 0: update timeout
login failed: no packet received in timeout 

When SMA TRIPOWER inverter is directly connected via Ethernet to Router there is no timeout reported.

**Expected behavior**
No login timeout when using PowerLAN / powerline communication. 

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.84

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
type: template
  template: sma-inverter
  usage: pv 
  host: 192.168.178.26  # IP-Adresse oder Hostname
  password: xxxxxxx  # Wird nur benötigt, wenn das Standardpasswort geändert wurde
  name: pv2

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
[main  ] INFO 2022/02/28 10:46:55 evcc 0.84
[main  ] INFO 2022/02/28 10:46:55 using config file /home/pi/evcc.yaml
[main  ] INFO 2022/02/28 10:46:55 listening at 0.0.0.0:7070
[site  ] INFO 2022/02/28 10:46:58 site config:
[site  ] INFO 2022/02/28 10:46:58   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/02/28 10:46:58     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/02/28 10:46:58     pv 0:      power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/02/28 10:46:58 loadpoint 1:
[lp-1  ] INFO 2022/02/28 10:46:58   mode:        pv
[lp-1  ] INFO 2022/02/28 10:46:58   charger:     power ✓ energy ✓ currents ✗ phases ✗
[lp-1  ] INFO 2022/02/28 10:46:58   meters:      charge ✓
[lp-1  ] INFO 2022/02/28 10:46:58     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2022/02/28 10:46:58   vehicles:    ✗
[main  ] FATAL 2022/02/28 10:46:59 listen tcp 0.0.0.0:7070: bind: address already in use

</pre></details>

premultiply, 2022-02-28:
> `[main  ] FATAL 2022/02/28 10:46:59 listen tcp 0.0.0.0:7070: bind: address already in use`

Da laufen einfach zwei evcc Prozesse gleichzeitig und kommen sich in die Quere.

closed 2022-02-28 by premultiply

vfricke, 2022-02-28:
Ich habe keine zwei Prozesse gestartet. 

Dies bekomme ich angezeigt: 
```
pi@raspberrypi:~ $ sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-02-28 10:57:02 CET; 4min 25s ago
   Main PID: 764 (evcc)
      Tasks: 8 (limit: 415)
        CPU: 8.814s
     CGroup: /system.slice/evcc.service
             └─764 /usr/local/bin/evcc

Feb 28 10:57:31 raspberrypi evcc[764]: [site  ] INFO 2022/02/28 10:57:31   meters:      grid ✓ pv ✓ battery ✗
Feb 28 10:57:31 raspberrypi evcc[764]: [site  ] INFO 2022/02/28 10:57:31     grid:      power ✓ energy ✓ currents ✓
Feb 28 10:57:31 raspberrypi evcc[764]: [site  ] INFO 2022/02/28 10:57:31     pv 0:      power ✓ energy ✓ currents ✓
Feb 28 10:57:31 raspberrypi evcc[764]: [lp-1  ] INFO 2022/02/28 10:57:31 loadpoint 1:
Feb 28 10:57:31 raspberrypi evcc[764]: [lp-1  ] INFO 2022/02/28 10:57:31   mode:        pv
Feb 28 10:57:31 raspberrypi evcc[764]: [lp-1  ] INFO 2022/02/28 10:57:31   charger:     power ✓ energy ✓ currents ✗ phases ✗
Feb 28 10:57:31 raspberrypi evcc[764]: [lp-1  ] INFO 2022/02/28 10:57:31   meters:      charge ✓
Feb 28 10:57:31 raspberrypi evcc[764]: [lp-1  ] INFO 2022/02/28 10:57:31     charge:    power ✓ energy ✓ currents ✗
Feb 28 10:57:31 raspberrypi evcc[764]: [lp-1  ] INFO 2022/02/28 10:57:31   vehicles:    ✗
Feb 28 10:59:46 raspberrypi evcc[764]: [sma   ] ERROR 2022/02/28 10:59:46 login failed: no packet received in timeout
```

premultiply, 2022-02-28:
`sudo systemctl stop evcc.service`
`evcc --log debug`
sollte Klarheit schaffen.

Wenn diese Meldungen nur gelegentlich auftreten dann ist das Problem eher in deiner Netzwerkinstallation zu suchen.
Ein grundsätzliches Problem in dieser Richtung seitens evcc kann ich jedenfalls ziemlich sicher ausschließen. Einer meiner SMA WR ist auch per PowerLAN angebunden und funktioniert problemlos. Und wenn es bei dir auch ohne PowerLAN funktioniert grenzt dies den Fehler ebenso darauf ein. PowerLAN ist aber eine transparente LAN-Verbindung die aber erhöhtem Paketverlust unterliegt. Da SMA Speedwire über UDP läuft werden verlorere oder defekte Pakete ggf. nicht immer erkannt oder führen zu merkwürdigen Effekten.

Waldmensch1, 2022-02-28:
Ich kann nur von Powerlan in Verbindung mit Speedwire abraten. Habe das selbst alles durch. Wirklich stabil ist nur ein Kabel. Wenn zu umständlich, lieber den Raspberry mit evcc in Richtung Wechselrichter bewegen und dort alles an einen Switch anschließen.

premultiply, 2022-02-28:
Wie gesagt: Hier wirklich gar keine Probleme mit Speedwire und PowerLAN. Funktioniert absolut störungsfrei.

SMA-WR kann man alternativ aber auch problemlos per SunSpec/Modbus abfragen (ID 126).