evcc-io/evcc #2741: feature request: Ladeblockierung

MoritzLerch, 2022-02-28:
Die Möglichkeit das Laden zu stoppen, "Stop"-Modus, hindert bei dauerthafter Speicherung (`resetOnDisconnect: false`) andere/fremde Personen daran, an der eigenen Wallbox zu laden, ist meiner Meinung nach aber nur eine temporäre Lösung.

In meinem Fall ist das die ABL eMH1, die allerdings in Kombination mit einem Schlüsselschalter auch die Möglichkeit besitzt, den Ladepunkt dauerhaft zu sperren -> Indikation durch das Leuchten beider LED's. Dieser Modus ist laut dem Protokoll-Manual allerdings nicht nur durch den Schlüsselschalter, sondern auch durch einen Modbus-Call zu erreichen. Hier die Modbus-Dokumentation: [Schnittstellenbeschreibung_Modbus-ASCII.pdf](https://github.com/evcc-io/evcc/files/8157112/Schnittstellenbeschreibung_Modbus-ASCII.pdf). Auf Seite 7 ist zu sehen, dass auch der gesperrte Modus durch das Versetzen in state E0 hervorgerufen werden kann (auf der letzten Seite sind nochmal die ganzen States vermerkt). 

Mein Wunsch wäre, dass es im Frontend bei den verschiedenen Modi (evtl. neben dem Charger-Namen) ein kleines Schloss gäbe, das es ermöglicht, in diesen Modus zu schalten. Der aktuelle Stand des Modus sollte natürlich dann auch dauerhaft gespeichert bleiben, bis die Wallbox wieder freigeschaltet wird. In [diesem Kommentar](https://github.com/evcc-io/evcc/issues/526#issuecomment-894760964) wurde auch schon einmal dieser Modus beschrieben, so wie ich ihn mir wünschen würde. Andere Wallboxen unterstützen solche gesperrten Zustände sicher auch. 

Für mich wäre das Feature wichtig, um optisch an der Wallbox (speziell bei der eMH1) zu signalisieren, dass keine Ladung möglich ist, damit man das im vorrübergehen schnell überprüfen kann. Momentan blinkt die Wallbox nämlich immer noch ganz normal, obwohl der Ladevorgang deaktiviert wurde. Unsere Wallbox hängt recht zugänglich und da wäre es äußerst praktisch, diese Funktion zu haben.

premultiply, 2022-02-28:
Hallo Moritz,
so wie ich es verstehe möchtest du nach dem Abstecken eines Fahrzeugs die Box automatisch sperren lassen?

Das geht mittels des Default-Reset-Modus `resetOnDisconnect: true` (https://docs.evcc.io/docs/reference/configuration/loadpoints#resetondisconnect) sowie `mode: off` am loadpoint.

Darüber hinaus kannst du auch eurem Fahrzeug eine Default-Action mittels [onIdentify](https://docs.evcc.io/docs/reference/configuration/vehicles/#onidentify) zuweisen und dort mittels `mode` einen angemessenen Default-Lademodus (usw.) zuweisen so dass dieser Ladevorgang trotz der Sperre automatisch beginnt sobald das Fahrzeug erkannt wurde

MoritzLerch, 2022-02-28:
Ich habe mir nochmal den Issue-Thread durchgelesen und da wurde auch erwähnt, dass das Sperren der Box über das Setzen des Stroms auf 0A gemacht wurde. Jedoch gibt es bei der ABL-Wallbox eben auch eine richtige Sperrung der Box, sodass das dann auch mittels der LED's signalisiert wird. Die anderen Dinge, die du erwähnst, hatte ich ja auch schon beschrieben. 

Mein Wunsch wäre es eben, einen extra Button dafür zu bekommen, mit dem man bei Wallboxen, die einen richtigen Sperrmodus besitzen, diese Sperrung durchführen zu können. Also äquivalent zum Schlüsselschalter an der ABL, was ich in den letzten drei Absätzen beschreibe.

premultiply, 2022-02-28:
Da die Box dann in einen Fehlerzustand (E0) wechselt der die sonstige Zustände überlagert stört dieser Modus die sonstigen Abläufe von evcc, insbesondere die Fahrzeugerkennung. Daher nutzen wir diese Funktion nicht.

MoritzLerch, 2022-02-28:
Ok, schade.

Hofyyy, 2022-03-01:
> Ok, schade.
> 
> 

Nur eine idee ob ich es richtig verstehe. 

Bei mir ist in der wallbox Authentifizierung per nfc aktiviert. also im weitesten sinne ist die wallbox für fremde dauerhaft gesperrt. 

Der default modus von mir ist ^min^. wenn mein auto erkannt wird fängt er ohne nfc das laden an, weil sie per modbus tcp das go bekommt  

ist das nicht das was du willst? und  warum kann du deine box nicht dauerhaft an der box sperren?