evcc-io/evcc #2748: Deadlock when charging (ab Release 0.83)

andig, 2022-03-01:
### Discussed in https://github.com/evcc-io/evcc/discussions/2746

<div type='discussions-op-text'>

<sup>Originally posted by **danistst** March  1, 2022</sup>
Dies ist ein fantastisches Projekt, ich würde es unbedingt gerne zum Laufen bekommen!

Die Releases 0.83 und 0.84 führen bei mir dazu, dass sich evcc komplett aufhängt wenn ein Ladevorgang gestartet wird (egal ob im Modus "PV" und durch Überschuss oder manuell durch anwählen das Modus "Now")

Mit Release 0.82 habe ich dieses Problem nicht. Allerdings verhält sich die Phasenumschaltung mit 0.82 nicht wie ich das aufgrund der Beschreibung erwartet hätte und es kommt zu Issues wie ich sie auch anderswo hier in den Diskussionen lese.

Mein Setup:
go-e Wallbox, Skoda Enyaq, PV
Dazu: bestehender Smartfox für Eigenverbrauchsoptimierung der Wärmepumpe und des Warmwasserboilers (habe ich so konfiguriert, dass evcc genügend Zeit bekommt bei Überschuss mit Laden zu beginnen)
Home Assistant (im Moment nur um die PV und Gridwerte vom Smartfox via MQTT für evcc zur Verfügung zu stellen, später möchte ich evcc als Home Assistant Add-on laufen lassen).
evcc läuft aktuell auf einem Windows Rechner bis das Setup funktioniert (dann Migration auf Home Assistant/Rasperry Pi).

Meine Config:
 
```
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

sponsortoken: MEINTOKEN

log: debug

meters:
- name: smartfox_power
  type: custom
  power:
    source: mqtt
    topic: power/smartfox/powertotal
    scale: 1
    timeout: 30s
- name: smartfox_pv
  type: custom
  power:
    source: mqtt
    topic: power/smartfox/pv
    scale: 1
#    timeout: 30s

chargers:
- type: template
  template: go-e-v3
  host: 192.168.178.85  # IP-Adresse oder Hostname
  name: wallbox2

vehicles:
- type: template
  template: enyaq
  title: Enyaq  # Wird in der Benutzeroberfläche angezeigt
  user: "MEINEEMAIL"  # Benutzername um auf das Gerät zuzugreifen
  password: "MEINPASSWORT"  # Passwort des Benutzerkontos
  capacity: 77  # Akku-Kapazität in kWh
  phases: 3  # Die maximale Anzahl der Phasen welche genutzt werden können
  mode: pv  # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll
  minSoC: 25  # Lade sofort mit maximaler Geschwindigkeit bis zu dem angegeben Ladestand, wenn der Lademodus nicht auf 'Aus' steht
  identifiers: # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/guides/vehicles/#erkennung-des-fahrzeugs-an-der-wallbox
  - 
  name: ev1

loadpoints:
- title: go-eCharger
  charger: wallbox2
  vehicles:
  - ev1
  mode: pv
  phases: 3
[... 176 lines cut ...]
[site  ] DEBUG 2022/03/01 11:27:09 ----
[site  ] DEBUG 2022/03/01 11:27:09 grid power: -2820W
[site  ] DEBUG 2022/03/01 11:27:09 pv power: 5499W
[site  ] DEBUG 2022/03/01 11:27:09 site power: -2820W
[lp-1  ] DEBUG 2022/03/01 11:27:09 charge power: 3830W
[lp-1  ] DEBUG 2022/03/01 11:27:09 charge currents: [15.7 0 0]A
[lp-1  ] DEBUG 2022/03/01 11:27:09 charger status: C
[lp-1  ] INFO 2022/03/01 11:27:09 start charging ->
[lp-1  ] DEBUG 2022/03/01 11:27:10 vehicle soc: 48%
[lp-1  ] DEBUG 2022/03/01 11:27:10 vehicle range: 222km
[lp-1  ] DEBUG 2022/03/01 11:27:10 available power above 3p min threshold of 4140W
[lp-1  ] DEBUG 2022/03/01 11:27:10 start phase enable timer: 0s
[lp-1  ] DEBUG 2022/03/01 11:27:10 phase action: scale3p in 0s
[lp-1  ] DEBUG 2022/03/01 11:27:10 phase enable timer elapsed
[lp-1  ] DEBUG 2022/03/01 11:27:10 charger disable
[lp-1  ] DEBUG 2022/03/01 11:27:10 wake-up timer: stop
[lp-1  ] DEBUG 2022/03/01 11:27:10 vehicle phases: 3p
```
Und dann hängt alles. Im Command Prompt werden keine Logs mehr geschrieben. Im Hintergrund passiert auch nichts (d.h. der Status des Chargers ändert sich nicht).

Das einzige was ich noch sehe: Im Frontend (Browser) erscheint die folgende Fehlermeldung

> Request failed with status code 500: API request failed http://192.168.178.91:7070/api/health

</div>

andig, 2022-03-01:
@danistst das sieht nach einem Deadlock aus. Kannst Du mal bitte http://evcc.local:7070/debug/pprof/goroutine?debug=1 im Browser eingeben? Evtl. evcc.local durch Deine IP ersetzen.

danistst, 2022-03-01:
@andig danke, habe ich soeben versucht. ich bekomme einen 404 page not found.
mit lokaler IP. wenn ich in einem anderen fenster das normale UI öffne kann ich den deadlock reproduzieren.

Waldmensch1, 2022-03-02:
@andig Das muss nach meinem PR passiert sein. Als Du den gemerged hast, habe ich das letzte mal den master auf meinen Fork gezogen. Bei mir läuft alles normal. Nach make wird mir die 83 angezeigt. Vielleicht kannst Du so den Auslöser zeitlich eingrenzen. Ich habe auch den go-e V3 und Enyaq (allerdings über TRONITY eingebunden)

andig, 2022-03-02:
@Waldmensch1 guter Tipp, aber ich seh nix. @danistst bitte noch `--profile` beim Start anhängen, aus Sicherheitsgründen ist das standardmäßig nicht aktiv.

danistst, 2022-03-02:
Here we go:

```
goroutine profile: total 35
3 @ 0x20be36 0x201a93 0x234549 0x2bddf2 0x2bf425 0x2c023f 0x3830a9 0x396925 0x48526d 0x40d683 0x40d7dd 0x48ad96 0x23ae21
#	0x234548	internal/poll.runtime_pollWait+0x88	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/netpoll.go:234
#	0x2bddf1	internal/poll.(*pollDesc).wait+0x31	/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_poll_runtime.go:84
#	0x2bf424	internal/poll.execIO+0xe4		/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:175
#	0x2c023e	internal/poll.(*FD).Read+0x25e		/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:441
#	0x3830a8	net.(*netFD).Read+0x28			/opt/hostedtoolcache/go/1.17.6/x64/src/net/fd_posix.go:56
#	0x396924	net.(*conn).Read+0x44			/opt/hostedtoolcache/go/1.17.6/x64/src/net/net.go:183
#	0x48526c	net/http.(*connReader).Read+0x16c	/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:780
#	0x40d682	bufio.(*Reader).fill+0x102		/opt/hostedtoolcache/go/1.17.6/x64/src/bufio/bufio.go:101
#	0x40d7dc	bufio.(*Reader).Peek+0x5c		/opt/hostedtoolcache/go/1.17.6/x64/src/bufio/bufio.go:139
#	0x48ad95	net/http.(*conn).serve+0xc35		/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:1955

2 @ 0x20be36 0x1d69cc 0x1d6438 0x1073f4a 0x23ae21
#	0x1073f49	github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe+0x69	/home/runner/work/evcc/evcc/util/pipe/limiter.go:115

1 @ 0x1dc47c 0x236198 0x537119 0x23ae21
#	0x236197	os/signal.signal_recv+0x97	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/sigqueue.go:169
#	0x537118	os/signal.loop+0x18		/opt/hostedtoolcache/go/1.17.6/x64/src/os/signal/signal_unix.go:24

1 @ 0x20be36 0x1d5b0c 0x1d56bd 0x102c325 0x23ae21
#	0x102c324	github.com/evcc-io/evcc/core.(*Site).loopLoadpoints+0x64	/home/runner/work/evcc/evcc/core/site.go:472

1 @ 0x20be36 0x1d69cc 0x1d63f8 0x10eecbe 0x23ae21
#	0x10eecbd	github.com/evcc-io/evcc/cmd.run.func2+0xbd	/home/runner/work/evcc/evcc/cmd/root.go:251

1 @ 0x20be36 0x1d69cc 0x1d6438 0x10731c5 0x23ae21
#	0x10731c4	github.com/evcc-io/evcc/server/updater.Run+0x2a4	/home/runner/work/evcc/evcc/server/updater/run.go:23

1 @ 0x20be36 0x1d69cc 0x1d6438 0x107330d 0x23ae21
#	0x107330c	github.com/evcc-io/evcc/server/updater.(*watch).watchReleases+0x6c	/home/runner/work/evcc/evcc/server/updater/watch.go:31

1 @ 0x20be36 0x1d69cc 0x1d6438 0x5d3d7e 0x23ae21
#	0x5d3d7d	github.com/evcc-io/evcc/util.(*Cache).Run+0x15d	/home/runner/work/evcc/evcc/util/cache.go:25

1 @ 0x20be36 0x1d69cc 0x1d6438 0x5d7645 0x23ae21
#	0x5d7644	github.com/evcc-io/evcc/util.(*Tee).Run+0x64	/home/runner/work/evcc/evcc/util/tee.go:27

1 @ 0x20be36 0x1d69cc 0x1d6438 0x79fd34 0x23ae21
#	0x79fd33	github.com/eclipse/paho%2emqtt%2egolang.startComms.func2+0x73	/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/net.go:424

1 @ 0x20be36 0x1d69cc 0x1d6438 0x79fdd1 0x23ae21
#	0x79fdd0	github.com/eclipse/paho%2emqtt%2egolang.startComms.func1+0x70	/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/net.go:401

1 @ 0x20be36 0x1d69cc 0x1d6438 0x7a311c 0x23ae21
#	0x7a311b	github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2+0xbb	/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/router.go:170

1 @ 0x20be36 0x1d69cc 0x1d6438 0x7b33d6 0x23ae21
#	0x7b33d5	github.com/evcc-io/evcc/server.(*MQTT).Run+0x595	/home/runner/work/evcc/evcc/server/mqtt.go:135

1 @ 0x20be36 0x1d69cc 0x1d6438 0x7b39d1 0x23ae21
#	0x7b39d0	github.com/evcc-io/evcc/server.(*SocketClient).writePump+0x90	/home/runner/work/evcc/evcc/server/socket.go:43

1 @ 0x20be36 0x1d69cc 0x1d6438 0xcc5e05 0x23ae21
#	0xcc5e04	github.com/evcc-io/evcc/push.(*Hub).Run+0x64	/home/runner/work/evcc/evcc/push/hub.go:55

1 @ 0x20be36 0x201a93 0x234549 0x2bddf2 0x2bf425 0x2c023f 0x3830a9 0x396925 0x2498ba 0x787445 0x787425 0x79d3ea 0x23ae21
#	0x234548	internal/poll.runtime_pollWait+0x88					/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/netpoll.go:234
#	0x2bddf1	internal/poll.(*pollDesc).wait+0x31					/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_poll_runtime.go:84
#	0x2bf424	internal/poll.execIO+0xe4						/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:175
#	0x2c023e	internal/poll.(*FD).Read+0x25e						/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:441
#	0x3830a8	net.(*netFD).Read+0x28							/opt/hostedtoolcache/go/1.17.6/x64/src/net/fd_posix.go:56
#	0x396924	net.(*conn).Read+0x44							/opt/hostedtoolcache/go/1.17.6/x64/src/net/net.go:183
#	0x2498b9	io.ReadAtLeast+0x99							/opt/hostedtoolcache/go/1.17.6/x64/src/io/io.go:328
#	0x787444	io.ReadFull+0x64							/opt/hostedtoolcache/go/1.17.6/x64/src/io/io.go:347
#	0x787424	github.com/eclipse/paho.mqtt.golang/packets.ReadPacket+0x44		/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/packets/packets.go:115
#	0x79d3e9	github.com/eclipse/paho%2emqtt%2egolang.startIncoming.func1+0xe9	/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/net.go:119
[... 96 lines cut ...]
#	0xd1541a	google.golang.org/grpc/internal/transport.(*controlBuffer).get+0x11a	/home/runner/go/pkg/mod/google.golang.org/grpc@v1.43.0/internal/transport/controlbuf.go:407
#	0xd15b04	google.golang.org/grpc/internal/transport.(*loopyWriter).run+0x84	/home/runner/go/pkg/mod/google.golang.org/grpc@v1.43.0/internal/transport/controlbuf.go:534
#	0xd1eaa4	google.golang.org/grpc/internal/transport.newHTTP2Client.func3+0x64	/home/runner/go/pkg/mod/google.golang.org/grpc@v1.43.0/internal/transport/http2_client.go:414

[... cut ...]
#	0x48bb8e	net/http.HandlerFunc.ServeHTTP+0x2e			/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:2047
#	0x48d488	net/http.(*ServeMux).ServeHTTP+0x148			/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:2425
#	0x4d66ae	github.com/gorilla/mux.(*Router).ServeHTTP+0x1ce	/home/runner/go/pkg/mod/github.com/gorilla/mux@v1.8.0/mux.go:210
#	0x48f0fa	net/http.serverHandler.ServeHTTP+0x43a			/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:2879
#	0x48ac67	net/http.(*conn).serve+0xb07				/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:1930
```

danistst, 2022-03-02:
hier was nun erscheint:

Dies ist der Zustand mit 0.84, nachdem ich im Frontend auf "Now" laden geklickt habe und es einen Deadlock gibt.

```
goroutine profile: total 35
3 @ 0x20be36 0x201a93 0x234549 0x2bddf2 0x2bf425 0x2c023f 0x3830a9 0x396925 0x48526d 0x40d683 0x40d7dd 0x48ad96 0x23ae21
#	0x234548	internal/poll.runtime_pollWait+0x88	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/netpoll.go:234
#	0x2bddf1	internal/poll.(*pollDesc).wait+0x31	/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_poll_runtime.go:84
#	0x2bf424	internal/poll.execIO+0xe4		/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:175
#	0x2c023e	internal/poll.(*FD).Read+0x25e		/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:441
#	0x3830a8	net.(*netFD).Read+0x28			/opt/hostedtoolcache/go/1.17.6/x64/src/net/fd_posix.go:56
#	0x396924	net.(*conn).Read+0x44			/opt/hostedtoolcache/go/1.17.6/x64/src/net/net.go:183
#	0x48526c	net/http.(*connReader).Read+0x16c	/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:780
#	0x40d682	bufio.(*Reader).fill+0x102		/opt/hostedtoolcache/go/1.17.6/x64/src/bufio/bufio.go:101
#	0x40d7dc	bufio.(*Reader).Peek+0x5c		/opt/hostedtoolcache/go/1.17.6/x64/src/bufio/bufio.go:139
#	0x48ad95	net/http.(*conn).serve+0xc35		/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:1955

2 @ 0x20be36 0x1d69cc 0x1d6438 0x1073f4a 0x23ae21
#	0x1073f49	github.com/evcc-io/evcc/util/pipe.(*Dropper).pipe+0x69	/home/runner/work/evcc/evcc/util/pipe/limiter.go:115

1 @ 0x1dc47c 0x236198 0x537119 0x23ae21
#	0x236197	os/signal.signal_recv+0x97	/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/sigqueue.go:169
#	0x537118	os/signal.loop+0x18		/opt/hostedtoolcache/go/1.17.6/x64/src/os/signal/signal_unix.go:24

1 @ 0x20be36 0x1d5b0c 0x1d56bd 0x102c325 0x23ae21
#	0x102c324	github.com/evcc-io/evcc/core.(*Site).loopLoadpoints+0x64	/home/runner/work/evcc/evcc/core/site.go:472

1 @ 0x20be36 0x1d69cc 0x1d63f8 0x10eecbe 0x23ae21
#	0x10eecbd	github.com/evcc-io/evcc/cmd.run.func2+0xbd	/home/runner/work/evcc/evcc/cmd/root.go:251

1 @ 0x20be36 0x1d69cc 0x1d6438 0x10731c5 0x23ae21
#	0x10731c4	github.com/evcc-io/evcc/server/updater.Run+0x2a4	/home/runner/work/evcc/evcc/server/updater/run.go:23

1 @ 0x20be36 0x1d69cc 0x1d6438 0x107330d 0x23ae21
#	0x107330c	github.com/evcc-io/evcc/server/updater.(*watch).watchReleases+0x6c	/home/runner/work/evcc/evcc/server/updater/watch.go:31

1 @ 0x20be36 0x1d69cc 0x1d6438 0x5d3d7e 0x23ae21
#	0x5d3d7d	github.com/evcc-io/evcc/util.(*Cache).Run+0x15d	/home/runner/work/evcc/evcc/util/cache.go:25

1 @ 0x20be36 0x1d69cc 0x1d6438 0x5d7645 0x23ae21
#	0x5d7644	github.com/evcc-io/evcc/util.(*Tee).Run+0x64	/home/runner/work/evcc/evcc/util/tee.go:27

1 @ 0x20be36 0x1d69cc 0x1d6438 0x79fd34 0x23ae21
#	0x79fd33	github.com/eclipse/paho%2emqtt%2egolang.startComms.func2+0x73	/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/net.go:424

1 @ 0x20be36 0x1d69cc 0x1d6438 0x79fdd1 0x23ae21
#	0x79fdd0	github.com/eclipse/paho%2emqtt%2egolang.startComms.func1+0x70	/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/net.go:401

1 @ 0x20be36 0x1d69cc 0x1d6438 0x7a311c 0x23ae21
#	0x7a311b	github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2+0xbb	/home/runner/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.3.5/router.go:170

1 @ 0x20be36 0x1d69cc 0x1d6438 0x7b33d6 0x23ae21
#	0x7b33d5	github.com/evcc-io/evcc/server.(*MQTT).Run+0x595	/home/runner/work/evcc/evcc/server/mqtt.go:135

1 @ 0x20be36 0x1d69cc 0x1d6438 0x7b39d1 0x23ae21
#	0x7b39d0	github.com/evcc-io/evcc/server.(*SocketClient).writePump+0x90	/home/runner/work/evcc/evcc/server/socket.go:43

1 @ 0x20be36 0x1d69cc 0x1d6438 0xcc5e05 0x23ae21
#	0xcc5e04	github.com/evcc-io/evcc/push.(*Hub).Run+0x64	/home/runner/work/evcc/evcc/push/hub.go:55

1 @ 0x20be36 0x201a93 0x234549 0x2bddf2 0x2bf425 0x2c023f 0x3830a9 0x396925 0x2498ba 0x787445 0x787425 0x79d3ea 0x23ae21
#	0x234548	internal/poll.runtime_pollWait+0x88					/opt/hostedtoolcache/go/1.17.6/x64/src/runtime/netpoll.go:234
#	0x2bddf1	internal/poll.(*pollDesc).wait+0x31					/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_poll_runtime.go:84
#	0x2bf424	internal/poll.execIO+0xe4						/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:175
#	0x2c023e	internal/poll.(*FD).Read+0x25e						/opt/hostedtoolcache/go/1.17.6/x64/src/internal/poll/fd_windows.go:441
#	0x3830a8	net.(*netFD).Read+0x28							/opt/hostedtoolcache/go/1.17.6/x64/src/net/fd_posix.go:56
#	0x396924	net.(*conn).Read+0x44							/opt/hostedtoolcache/go/1.17.6/x64/src/net/net.go:183
#	0x2498b9	io.ReadAtLeast+0x99							/opt/hostedtoolcache/go/1.17.6/x64/src/io/io.go:328
#	0x787444	io.ReadFull+0x64							/opt/hostedtoolcache/go/1.17.6/x64/src/io/io.go:347
[... 98 lines cut ...]
#	0xd1541a	google.golang.org/grpc/internal/transport.(*controlBuffer).get+0x11a	/home/runner/go/pkg/mod/google.golang.org/grpc@v1.43.0/internal/transport/controlbuf.go:407
#	0xd15b04	google.golang.org/grpc/internal/transport.(*loopyWriter).run+0x84	/home/runner/go/pkg/mod/google.golang.org/grpc@v1.43.0/internal/transport/controlbuf.go:534
#	0xd1eaa4	google.golang.org/grpc/internal/transport.newHTTP2Client.func3+0x64	/home/runner/go/pkg/mod/google.golang.org/grpc@v1.43.0/internal/transport/http2_client.go:414

1 @ 0x20be36 0x21b532 0xd51287 0x23ae21
#	0xd51286	google.golang.org/grpc.(*ccBalancerWrapper).watcher+0xa6	/home/runner/go/pkg/mod/google.golang.org/grpc@v1.43.0/balancer_conn_wrappers.go:77
[... cut ...]
#	0x48bb8e	net/http.HandlerFunc.ServeHTTP+0x2e			/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:2047
#	0x48d488	net/http.(*ServeMux).ServeHTTP+0x148			/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:2425
#	0x4d66ae	github.com/gorilla/mux.(*Router).ServeHTTP+0x1ce	/home/runner/go/pkg/mod/github.com/gorilla/mux@v1.8.0/mux.go:210
#	0x48f0fa	net/http.serverHandler.ServeHTTP+0x43a			/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:2879
#	0x48ac67	net/http.(*conn).serve+0xb07				/opt/hostedtoolcache/go/1.17.6/x64/src/net/http/server.go:1930
```

danistst, 2022-03-02:
Ha! Das war eventuell ein Volltreffer mit MQTT. Ich habe bei den Metern beim MQTT publish retain auf true gestellt und nun scheint es für den Moment zu funktionieren. Wäre ja super wenn das Problem gelöst wäre.

Allerdings ist es schon komisch, dass dies dann zwischen 0.82 und 0.84 einen Unterschied macht. Und ist es Absicht, dass evcc nicht mehr weiterläuft, wenn zu einem gewissen Zeitpunkt grad keine Werte da sind? Wo ich nämlich sicher bin ist, dass im Verlaufe der Zeit immer Werte via MQTT broker published werden. Ich lese die auch anderswo ein.

PS: Falls das mit retain=true das Problem löst, wäre das ein Hinweis in der Doku wert. Oder hab ich den übersehen?

andig, 2022-03-02:
> nd ist es Absicht, dass evcc nicht mehr weiterläuft, wenn zu einem gewissen Zeitpunkt grad keine Werte da sind?

Das liegt an Deiner Config- Du hast oben den Timeout rausgenommen, damit wird auf einen Wert gewartet. Wenn der kam bevor evcc lief dann sehen wir den ohne retained nicht und schon knallt es...

> Oder hab ich den übersehen?

S.o. It depends...

danistst, 2022-03-02:
ahh. danke!
bin allerdings noch nicht so gut, dass ich sowas in einem log sehe. bin so ein stackoverflow-copy-paste-ler :)

danistst, 2022-03-03:
ou mist. ich bin wieder gleich weit. ohne dass ich etwas geändert hätte seit gestern. wieder ein deadlock :-(
was mir auffällt: der log endet im deadlock zustand immer mit der gleichen Zeile 

> [site  ] DEBUG 2022/03/03 22:47:04 grid power: 1183W
[site  ] DEBUG 2022/03/03 22:47:04 pv power: 0W
[site  ] DEBUG 2022/03/03 22:47:04 site power: 1183W
[lp-1  ] DEBUG 2022/03/03 22:47:05 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 22:47:05 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/03/03 22:47:05 charger status: B
[lp-1  ] DEBUG 2022/03/03 22:47:05 next soc poll remaining time: 54m49s
[lp-1  ] DEBUG 2022/03/03 22:47:05 max charge current: 0A = 0A + -5.14A (1183W @ 1p)
[lp-1  ] DEBUG 2022/03/03 22:47:06 set charge mode: now
[lp-1  ] DEBUG 2022/03/03 22:47:06 pv timer elapse
[lp-1  ] DEBUG 2022/03/03 22:47:06 pv action: inactive
[lp-1  ] DEBUG 2022/03/03 22:47:06 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 22:47:06 charge currents: [0 0 0]A
[site  ] DEBUG 2022/03/03 22:47:06 ----
[site  ] DEBUG 2022/03/03 22:47:06 grid power: 1183W
[site  ] DEBUG 2022/03/03 22:47:06 pv power: 0W
[site  ] DEBUG 2022/03/03 22:47:06 site power: 1183W
[lp-1  ] DEBUG 2022/03/03 22:47:06 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 22:47:07 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/03/03 22:47:08 charger status: B
[lp-1  ] DEBUG 2022/03/03 22:47:08 next soc poll remaining time: 54m46s
[lp-1  ] DEBUG 2022/03/03 22:47:08 vehicle phases: 3p

Ich bin auch sicher, dass laufend MQTT Werte für grid und pv reinkommen. Ich sehe die auch, wenn ich in einem zweiten command line window evcc meter starte (während evcc im deadlock ist).

andig, 2022-03-03:
Mhhm. Bitte nochmal das pprof von oben und bitte mqtt: trace.

danistst, 2022-03-03:
Danke! !

> C:\evcc>evcc --profile
[main  ] INFO 2022/03/03 23:41:59 evcc 0.84
[main  ] INFO 2022/03/03 23:41:59 using config file C:\evcc\evcc.yaml
[main  ] INFO 2022/03/03 23:41:59 listening at 0.0.0.0:7070
[mqtt  ] INFO 2022/03/03 23:41:59 connecting evcc-2119469692 at tcp://192.168.178.61:1883
[mqtt  ] DEBUG 2022/03/03 23:41:59 tcp://192.168.178.61:1883 connected
[mqtt  ] TRACE 2022/03/03 23:41:59 recv power/smartfox/powertotal: '1933'
[mqtt  ] TRACE 2022/03/03 23:41:59 recv power/smartfox/pv: '0'
[lp-1  ] WARN 2022/03/03 23:42:03 ignoring phases config (3p) for switchable charger
[site  ] INFO 2022/03/03 23:42:03 site config:
[site  ] INFO 2022/03/03 23:42:03   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/03/03 23:42:03     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/03/03 23:42:03     pv 0:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/03/03 23:42:03 loadpoint 1:
[lp-1  ] INFO 2022/03/03 23:42:03   mode:        pv
[lp-1  ] INFO 2022/03/03 23:42:03   charger:     power ✓ energy ✓ currents ✓ phases ✓
[lp-1  ] INFO 2022/03/03 23:42:03   meters:      charge ✓
[lp-1  ] INFO 2022/03/03 23:42:03     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/03/03 23:42:03   vehicles:    ✓
[lp-1  ] INFO 2022/03/03 23:42:03     vehicle 0: range ✓ finish ✓ status ✓ climate ✗
[lp-1  ] INFO 2022/03/03 23:42:03 vehicle updated: unknown -> Enyaq
[lp-1  ] DEBUG 2022/03/03 23:42:03 set charge mode: pv
[lp-1  ] DEBUG 2022/03/03 23:42:03 set min soc: 25
[lp-1  ] DEBUG 2022/03/03 23:42:03 vehicle phases: 3p
[site  ] DEBUG 2022/03/03 23:42:05 ----
[site  ] DEBUG 2022/03/03 23:42:05 grid power: 1933W
[site  ] DEBUG 2022/03/03 23:42:05 pv power: 0W
[site  ] DEBUG 2022/03/03 23:42:05 site power: 1933W
[lp-1  ] DEBUG 2022/03/03 23:42:05 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 23:42:05 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/03/03 23:42:05 charger status: B
[lp-1  ] DEBUG 2022/03/03 23:42:06 vehicle soc: 64%
[lp-1  ] DEBUG 2022/03/03 23:42:06 vehicle range: 285km
[lp-1  ] DEBUG 2022/03/03 23:42:06 assuming initial phase state: 3p
[lp-1  ] DEBUG 2022/03/03 23:42:06 available power below 3p min threshold of 4140W
[lp-1  ] DEBUG 2022/03/03 23:42:06 phase action: scale1p in 0s
[lp-1  ] DEBUG 2022/03/03 23:42:06 phase disable timer elapsed
[lp-1  ] DEBUG 2022/03/03 23:42:06 pv timer elapse
[lp-1  ] DEBUG 2022/03/03 23:42:06 pv action: inactive
[lp-1  ] DEBUG 2022/03/03 23:42:06 switched phases: 1p @ -1933W
[lp-1  ] DEBUG 2022/03/03 23:42:07 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 23:42:07 charge currents: [0 0 0]A
[site  ] DEBUG 2022/03/03 23:42:07 ----
[site  ] DEBUG 2022/03/03 23:42:07 grid power: 1933W
[site  ] DEBUG 2022/03/03 23:42:07 pv power: 0W
[site  ] DEBUG 2022/03/03 23:42:07 site power: 1933W
[lp-1  ] DEBUG 2022/03/03 23:42:07 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 23:42:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/03/03 23:42:08 charger status: B
[lp-1  ] DEBUG 2022/03/03 23:42:08 next soc poll remaining time: 59m57s
[lp-1  ] DEBUG 2022/03/03 23:42:08 max charge current: 0A = 0A + -8.4A (1933W @ 1p)
[lp-1  ] DEBUG 2022/03/03 23:42:08 pv enable timer reset
[lp-1  ] DEBUG 2022/03/03 23:42:08 pv action: inactive
[lp-1  ] DEBUG 2022/03/03 23:42:09 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 23:42:10 set charge mode: now
[lp-1  ] DEBUG 2022/03/03 23:42:10 pv timer elapse
[lp-1  ] DEBUG 2022/03/03 23:42:10 pv action: inactive
[lp-1  ] DEBUG 2022/03/03 23:42:10 charge currents: [0 0 0]A
[site  ] DEBUG 2022/03/03 23:42:10 ----
[site  ] DEBUG 2022/03/03 23:42:10 grid power: 1933W
[site  ] DEBUG 2022/03/03 23:42:10 pv power: 0W
[site  ] DEBUG 2022/03/03 23:42:10 site power: 1933W
[lp-1  ] DEBUG 2022/03/03 23:42:10 charge power: 0W
[lp-1  ] DEBUG 2022/03/03 23:42:11 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2022/03/03 23:42:11 charger status: B
[mqtt  ] TRACE 2022/03/03 23:42:11 recv power/smartfox/powertotal: '1932'
[mqtt  ] TRACE 2022/03/03 23:42:11 recv power/smartfox/pv: '0'
[lp-1  ] DEBUG 2022/03/03 23:42:12 next soc poll remaining time: 59m53s
[... 193 lines cut ...]
#	0xe8cc72	github.com/evcc-io/evcc/core.(*LoadPoint).SetMode+0x52		/home/runner/work/evcc/evcc/core/loadpoint_api.go:29
#	0x601ec7	github.com/evcc-io/evcc/server.chargeModeHandler.func1+0xa7	/home/runner/work/evcc/evcc/server/http_handler.go:104
#	0x2dbdce	net/http.HandlerFunc.ServeHTTP+0x2e				/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:2047
#	0x43f01c	github.com/gorilla/handlers.(*cors).ServeHTTP+0x77c		/home/runner/go/pkg/mod/github.com/gorilla/handlers@v1.5.1/cors.go:138
#	0x43e4e5	github.com/gorilla/handlers.CompressHandlerLevel.func1+0x665	/home/runner/go/pkg/mod/github.com/gorilla/handlers@v1.5.1/compress.go:141
#	0x2dbdce	net/http.HandlerFunc.ServeHTTP+0x2e				/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:2047
#	0x601693	github.com/evcc-io/evcc/server.jsonHandler.func1+0xf3		/home/runner/work/evcc/evcc/server/http_handler.go:49
#	0x2dbdce	net/http.HandlerFunc.ServeHTTP+0x2e				/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:2047
#	0x3268ee	github.com/gorilla/mux.(*Router).ServeHTTP+0x1ce		/home/runner/go/pkg/mod/github.com/gorilla/mux@v1.8.0/mux.go:210
#	0x2df33a	net/http.serverHandler.ServeHTTP+0x43a				/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:2879
#	0x2daea7	net/http.(*conn).serve+0xb07					/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:1930

[... cut ...]
#	0x2dbdce	net/http.HandlerFunc.ServeHTTP+0x2e			/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:2047
#	0x2dd6c8	net/http.(*ServeMux).ServeHTTP+0x148			/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:2425
#	0x3268ee	github.com/gorilla/mux.(*Router).ServeHTTP+0x1ce	/home/runner/go/pkg/mod/github.com/gorilla/mux@v1.8.0/mux.go:210
#	0x2df33a	net/http.serverHandler.ServeHTTP+0x43a			/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:2879
#	0x2daea7	net/http.(*conn).serve+0xb07				/opt/hostedtoolcache/go/1.17.7/x64/src/net/http/server.go:1930
```

danistst, 2022-03-03:
[image]

Das sehe ich im Frontend.

danistst, 2022-03-03:
hier noch die config.

```
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

sponsortoken: XXXX

log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug
  mqtt: trace

meters:
- name: smartfox_power
  type: custom
  power:
    source: mqtt
    topic: power/smartfox/powertotal
    scale: 1
    timeout: 30s
- name: smartfox_pv
  type: custom
  power:
    source: mqtt
    topic: power/smartfox/pv
    scale: 1
    timeout: 30s

chargers:
- type: template
  template: go-e-v3
  host: 192.168.178.85  # IP-Adresse oder Hostname
  name: wallbox2

vehicles:
- type: template
  template: enyaq
  title: Enyaq  # Wird in der Benutzeroberfläche angezeigt
  user: "XXXX"  # Benutzername um auf das Gerät zuzugreifen
  password: "XXXXX"  # Passwort des Benutzerkontos
  capacity: 77  # Akku-Kapazität in kWh
  phases: 3  # Die maximale Anzahl der Phasen welche genutzt werden können
  mode: pv  # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll
  minSoC: 25  # Lade sofort mit maximaler Geschwindigkeit bis zu dem angegeben Ladestand, wenn der Lademodus nicht auf 'Aus' steht
  identifiers: # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/guides/vehicles/#erkennung-des-fahrzeugs-an-der-wallbox
  - 
  name: ev1

loadpoints:
- title: go-eCharger
  charger: wallbox2
  vehicles:
  - ev1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  guardduration: 0m

# mqtt message broker
mqtt:
  broker: 192.168.178.61:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: XXXXX
  password: "XXXXXX"

site:
  title: XXXXX
  meters: 
    grid: smartfox_power
    pv: smartfox_pv
```

andig, 2022-03-04:
Ok, denke ich sehe es. Muss ich drüber nachdenken. Der Fehler passiert in dem Moment, wo Du den Modus im UI umschaltest, richtig?

andig, 2022-03-04:
Ich bin mir nicht sicher, warum das zu einem Deadlock kommen kann, aber hier warten der Webserver:

    github.com/evcc-io/evcc/core.(*LoadPoint).SetMode+0x52	

und der Ladepunkt gleichzeitig:

```
#	0xe888ce	github.com/evcc-io/evcc/core.(*LoadPoint).setVehiclePhases+0xee		/home/runner/work/evcc/evcc/core/loadpoint.go:992
#	0xe88c6e	github.com/evcc-io/evcc/core.(*LoadPoint).setPhases+0x18e		/home/runner/work/evcc/evcc/core/loadpoint.go:1032
#	0xe88e5e	github.com/evcc-io/evcc/core.(*LoadPoint).scalePhases+0x13e		/home/runner/work/evcc/evcc/core/loadpoint.go:1061
#	0xe88a78	github.com/evcc-io/evcc/core.(*LoadPoint).scalePhasesIfAvailable+0x18	/home/runner/work/evcc/evcc/core/loadpoint.go:1007
#	0xe8c3a8	github.com/evcc-io/evcc/core.(*LoadPoint).Update+0x808			/home/runner/work/evcc/evcc/core/loadpoint.go:1527
```

andig, 2022-03-04:
Könntest Du mal bitte auf die Nightly Version upgraden? Da scheint es den Fehler nicht zu geben.

danistst, 2022-03-04:
Genau, dann wenn ich den Modus umschalte. Egal ob wie evcc UI oder via MQTT (über welches ich evcc via Home Assistant steuern kann).

Das mache ich gerne mit den Nightly. jetzt musst du mir aber kurz helfen wie ich das tue. Ich nehme an ich muss den Code runterladen und bei mir einen Build machen? Wenn ja, wie schaff ich das?

ich hier nur amateur und lade jeweils auf https://github.com/evcc-io/evcc/releases das ZIP für Windows runter.

Aber das mit den nightly wäre insofern auch interessant, dass ich dann beim Testen helfen könnte.

andig, 2022-03-04:
Für Windows musst Du das leider selber machen: Projekt mit git auschecken und

    make build

Falls das nicht klappt meld Dich mal bei cpuidle@gmx.de, dann schick ich Dir eine Binary.

danistst, 2022-03-04:
das habe ich leider nicht fertiggebracht. 

aber dafür habe ich Docker Desktop zum Laufen gebracht. Und mit "docker pull andig/evcc:nightly" läuft das nun :-)

Damit habe ich nun keinen Deadlock mehr. super!

Die Verbindung zum Enyaq läuft nicht. Aber das ist wohl nicht Teil dieses branches oder so.

Und das Timing zwischen evcc Update und MQTT Daten ist nicht mehr ideal. Das "System" braucht länger um sich einzuschwingen (sprich zuerst gibt es ein paar on-off-on-off...). Das könnte aber auch daran liegen, dass das nun in Docker bei mir lokal läuft. 

Also, ich bin schon mal sehr happy und warte auf 0.85, welches ich dann ganz einfach wieder als Home Assistant add-on laufen lasse.

Danke @andig !!!

danistst, 2022-03-04:
Jetzt läuft auch die Enyaq Integration. War eventuell was on Skoda Seite...

closed 2022-03-04 by andig