evcc-io/evcc #2750: Fehlermeldung invalid charge mode 0 bei Rücksetzung auf off

maikheinrich, 2022-03-01:
Hallo zusammen,

evcc meldet bei Disconnect des Autos eine gelbes Dreieck mit folgender Fehlermeldung: "invalid charge mode 0".

in der evcc.yaml ist unter loadpoints folgendes konfiguriert

mode: off
resetOnDisconnect: true

Auch wird beim start von EVCC in der Website kein Lademodus angezeigt. Hier scheint beim starten ebenfalls der Mode off nicht sauber gesetzt zu werden. Konfiguriere ich Testweise mode: pv oder mode: now funktioniert alles wie erwartet.
Bitte keine Diskussion warum ich nachdem anstecken NICHT automatisch starten möchte. Der Ladepunkt soll so gegen unberechtigt Nutzung blockiert werden.

LG Maik

maikheinrich, 2022-03-01:
Hi andig,

welche Infos benötigt ihr noch. Bei diesem kleinen glitch, die kompletten logs zu sammeln ist eventuell ein wenig overkill ?
Können wir hier weitermachen, Oder soll ich erneut ein Ticket öffnen.

LG Maik

andig, 2022-03-01:
Du könntest zumindest mal die Version mitteilen auf die sich dein Fehler bezieht. Die Fragen stellen wir ja nicht aus Langeweile sondern genau um nicht bei jeden Issue rückfragen zu müssen.

maikheinrich, 2022-03-01:
Hallo andig,

ja sorry, da hast Du natürlich recht. Ich benutze die 0.84. Ich wollte in "meiner Eile" noch schnell den Bug melden. Dies ist gründlich daneben gegangen. Ich mache es jetzt nochmals "ordentlich"

LG Maik

andig, 2022-03-01:
Ich fürchte das ist wieder https://github.com/evcc-io/evcc/issues/1136. Alle Workarounds waren erfolglos. Dank Deiner Meldung bin ich aber auf https://github.com/spf13/viper/blob/5f0a660bb4d3c4404be6a39bab58ad29c2319af3/TROUBLESHOOTING.md gestossen, ich werde mal probieren ob das schon einsatzbereit ist. Solange „off“ in Anführungszeichen setzen!

maikheinrich, 2022-03-01:
Hi andig,

danke, das hilft doch schonmal weiter. Eventuell könnte man dies noch in der Anleitung erwähnen. 
https://docs.evcc.io/docs/reference/configuration/loadpoints

Damit der nächste nicht drauf reinfällt.

Schönen Abend und vielen Dank

Maik

closed 2022-03-02 by andig