evcc-io/evcc #2755: Auto voll -> Ladeleistung wird weiter angenommen

tribun83, 2022-03-02:
Wenn mein Auto (MG ZS EV ohne Einbindungsmöglichkeit in EVCC) an der Wallbe Eco 2.0S voll geladen ist und keine Leistung mehr abnimmt denkt EVCC, dass noch Ladeleistung abgenommen wird und auch die kWh bei "Geladen" werden weiter hochgezählt. 
Die Anzeige zur Verteilung Sonne/Netz/Haus stimmt auch nicht (Sonne oben nicht über den Gesamtverbrauch, obwohl Einspeisung). Ich versuche den Screenshot als Kommentar hochzuladen. 

Ich hoffe ich bin bei "Bug" richtig. Hängt wohl nicht primär von meiner Konfiguration ab. Hab ich etwas falsch eingerichtet? Fehler bekannt? Tritt sicher nicht auf, wenn das Auto smart genug ist und mit EVCC kommuniziert. 
Wenn benötigt liefere ich natürlich gerne logs, config, Fernzugang, etc.. 

Wenn im Gesamtsystem "zufällig" so viel Energie "über" ist, wie beim Auto als Ladung angenommen wird könnte EVCC doch annehmen, dass das Auto voll ist und von "lädt" auf "verbunden" (oder gar "voll"?) wechseln.

closed 2022-03-02 by andig

andig, 2022-03-02:
Das Auto teilt mit ob es lädt (Status B vs C). Evcc nutzt die Information, einen Annahme ist nicht notwendig.

RTTTC, 2022-03-02:
Looking at your printscreen it seems that your configuration is incorrect, something is off with `meter` ?

tribun83, 2023-09-23:
> Das Auto teilt mit ob es lädt (Status B vs C). Evcc nutzt die Information, einen Annahme ist nicht notwendig.

Danke, Bei mir funktioniert das leider nicht. Auf welche Daten muss ich da schauen, bzw woran kann es liegen? 
Wallbe Eco 2.0s und Kaco WR sind EVCC bekannt. Das Auto ist voll, aber EVCC meint, dass es weiter geladen wird... 

2023/09/23   13:15:12,stdout,[lp-1  ] DEBUG   2023/09/23 12:15:12 pv charge current: 40.1A = 20A + 20.1A (-4630W @ 1p)
2023/09/23 13:15:12,stdout,[lp-1  ] DEBUG 2023/09/23 12:15:12 charger status:   C
2023/09/23 13:15:12,stdout,[lp-1  ] DEBUG 2023/09/23 12:15:12 charge power:   4600W
2023/09/23 13:15:02,stdout,[lp-1  ] DEBUG 2023/09/23 12:15:02 charge power:   4600W
2023/09/23 13:14:52,stdout,[lp-1  ] DEBUG 2023/09/23 12:14:52 pv charge   current: 51.4A = 20A + 31.4A (-7232W @ 1p)
2023/09/23 13:14:52,stdout,[lp-1  ] DEBUG 2023/09/23 12:14:52 charger status:   C

premultiply, 2023-09-23:
Das Auto signalisiert einfach weiter Energiebedarf, die Box lässt daher das Schütz an (Status C).

Deine Box hat keinen Zähler und es ist offensichtlich auch kein externer Zähler vorhanden daher kann evcc nur die Leistung anhand der freigegebenen Stromstärke (1p, 230V, 20A) schätzen.

tribun83, 2023-09-23:
Verstehe, danke. Hilft es, wenn ich die Daten von meinem SmartMeter EVCC per MQTT bereitstelle?
Ansonsten: Mit dem "Problem" bin ich ja nicht alleine und falls ich nichts übersehe möchte ich folgenden Workaround vorschlagen, falls möglich. Evtl. ja zur lokalen Aktivierung. 
Wenn angenommene Ladeleistung größer als Netzbezug+PV-Ertrag+Batterieleistung = Voll geladen annehmen (charger status: B). Die Logik greift natürlich nicht, wenn der Verbrauch (>1,4kW) im Haus (Backofen oä) auftritt... 

Kann ich das sonst irgenwie elaborieren oder diskutieren, damit es hilfreich ist?