evcc-io/evcc #2797: Slack Einladungslink auf evcc.io ist nicht mehr gültig

premultiply, 2022-03-06:
### Discussed in https://github.com/evcc-io/evcc/discussions/2793

<div type='discussions-op-text'>

<sup>Originally posted by **lorbetzki** March  6, 2022</sup>
Hi,

siehe Betreff. Der Einladungslick zum Slack Worspace ist ungültig. 

Viele Grüße</div>

premultiply, 2022-03-06:
@andig @naltatis Gibt es dauerhafte Links die ich noch nicht gefunden haben?
Bei mir laufen die immer nach 27 Tagen ab. Das ist wenig hilfreich.

naltatis, 2022-03-06:
Hm, ich finde auch keinen Weg. Durch unsere temporäre PRO Mitgliedschaft können wir zeitlich unbegrenzte Gast-Einladelinks erzeugen. Gäste sind aber eingeschränkt und das löst unser Problem nicht wirklich.

But there seems to be a way. At least the guys from single-spa seem to [have found a way](https://github.com/single-spa/single-spa.js.org/pull/502#issue-1057647512) to create a non-expiring but 2000 people limited invite link. Maybe @filoxo can tell us how he did this. 😀

andig, 2022-03-07:
Als temporärer Link sollte dieser funktionieren: https://join.slack.com/t/evccgroup/shared_invite/zt-14yub38w0-7xDdOMo3Wy7AwdrNrAtWsQ

filoxo, 2022-03-08:
hi @naltatis sorry about the lack of response. I didn't see a notification until now. I don't think I did anything special for this link but here are the steps I took:

- within the slack workspace, click on the workspace name in the top-left corner
- click on "Invite people to [workspace name]"
- click on "Edit link settings"
- change "Expiration" to "never expires"

And here is a screenshot of that menu within my Slack workspace. The paragraph at the top is where it calls out that the link will expire after it is used by 2000 people. 

[image]
 
Also, we don't have a paid Slack account so these steps will most likely be the same for all Slack workspaces. Hopefully this helps!

naltatis, 2022-03-08:
[Website](https://github.com/evcc-io/evcc.io/commit/072e7f4df1c4dad674f8399c899765ad07af313e#diff-200abda264c3530e9cdb5d895e25d50bc3ed567079b39fe2ee3c1d1b5d661e09L92) und [Blog](https://github.com/evcc-io/docs/commit/8406e8669a9ab0b9ecb147035cd4bc3d401a3e7e#diff-a776f7090b6bc74b0b04f065b99125acda0e13c1f2c232ed5e840ce9dbe5dcd1L13) sind nun wieder auf dem aktuellen Stand.

closed 2022-03-08 by naltatis