evcc-io/evcc #2812: PMCC lädt in PV Modus auch wenn keine Erzeugung vorhanden ist.

MarkusGH, 2022-03-08:
Ein Ladepunkt mit PMCC, Modus PV aktiv.

Fahrzeug wird 23:54:27 angesteckt.
23:55:12 meldet evcc einen maximalen Ladestrom von 0A:

[lp-1  ] DEBUG 2022/03/07 23:55:12 charger status: B
[eebus ] TRACE 2022/03/07 23:55:12 !! identify: ev communication is IEC61851 which does not support any identification
[eebus ] TRACE 2022/03/07 23:55:12 !! soc: feature not available
[lp-1  ] DEBUG 2022/03/07 23:55:12 max charge current: 0A = 0A + -9.1A (2093W @ 1p)

Dennoch startet 23:55:13 der Ladevorgang:

[eebus ] TRACE 2022/03/07 23:55:12 phases:  1 , charged energy:  0 Wh
[eebus ] TRACE 2022/03/07 23:55:12 current current: L1  1 A, L2  0 A L3  0 A
[eebus ] TRACE 2022/03/07 23:55:12 current power: L1  83 W, L2  0 W L3  0 W
[eebus ] TRACE 2022/03/07 23:55:12 send: notify EVSE:DeviceDiagnosis DeviceDiagnosisHeartbeatData
[eebus ] TRACE 2022/03/07 23:55:12 send: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":5}]},{"addressDestination":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1]},{"feature":5}]},{"msgCounter":104},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"deviceDiagnosisHeartbeatData":[{"heartbeatCounter":55},{"heartbeatTimeout":"PT4S"}]}]]}]}]}}]}
[eebus ] TRACE 2022/03/07 23:55:12 recv: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1]},{"feature":5}]},{"addressDestination":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":5}]},{"msgCounter":601},{"msgCounterReference":104},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":6}]}]]}]}]}}]}
[eebus ] TRACE 2022/03/07 23:55:12 recv: result CEM:DeviceDiagnosis ResultData
[eebus ] TRACE 2022/03/07 23:55:13 !! current power: returning 83.000000
[lp-1  ] DEBUG 2022/03/07 23:55:13 charge power: 83W
[eebus ] TRACE 2022/03/07 23:55:13 !! currents: returning 1.000000, 0.000000, 0.000000,
[lp-1  ] DEBUG 2022/03/07 23:55:13 charge currents: [1 0 0]A
[eebus ] TRACE 2022/03/07 23:55:13 !! charged energy: returning 0.000000
[lp-1  ] DEBUG 2022/03/07 23:55:13 charger status: C
[lp-1  ] INFO 2022/03/07 23:55:13 start charging ->

**Expected behavior**

Ladevorgang startet unter diesen Bedingungen nicht.
Hilfsweise: Ladevorgang wird oder wird mangels vorhandener PV Leistung schnell wieder beendet.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.83 (7a99de71)

```
<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>

[main  ] INFO 2022/03/08 10:01:39 evcc 0.83 (7a99de71)
[main  ] INFO 2022/03/08 10:01:39 using config file /etc/evcc.yaml
[mqtt  ] INFO 2022/03/08 10:01:39 connecting evcc-1641156788 at tcp://mqtt.xxx.de:1883
c6050aedf1f94672493bc0b7614ae09e16966ae3
[eebus ] ERROR 2022/03/08 10:01:39 eebus listen: listen tcp :4712: bind: address already in use
true, discovered false
[lp-1  ] WARN 2022/03/08 10:02:00 poll interval '30m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
config
======

grid: grid1
-----------
Power:          -2874W
Energy:         8263.3kWh
Current L1..L3: 4.1A 4.9A 4A

pv 0: pv1
---------
Power:          4098W
Energy:         10683.5kWh
Current L1..L3: 5.97A 5.98A 5.96A

loadpoint 1
===========

charger: wallbox1
-----------------
Power:                   offline
Current L1..L3:          offline
....................SoC: timeout
Charge status:           offline
Enabled:                 false
Charged:                 offline
Identifier:              offline

vehicle 0
---------
SoC:            100%
Charge status:  B
Range:          32km
Odometer:       2576km
Finish time:    2022-03-08 09:58:00 +0100 CET
Climate active: false
Outside temp:   20.0°C
Target temp:    20.0°C
Capacity:       17kWh

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>

uri: 0.0.0.0:7070 # uri for ui
interval: 1s # control cycle interval

log: info
levels:
  porsche: debug
  lp-1: debug
  eebus: trace

mqtt:
  broker: mqtt.xxx.de:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.1942 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- type: template
  template: mqtt-meter
  usage: grid
  power: mbmd/solaredge1-1-1/Power
  scalepower: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
  energy: mbmd/solaredge1-1-1/Import
  scaleenergy: 1  # Die Energiemenge wird in kWh erwartet. Falls z.b. Wh geliefert wird, kann dies hier mit dem Wert 0.001 angepasst werden
  currentp1: mbmd/solaredge1-1-1/Current/L1
  currentp2: mbmd/solaredge1-1-1/Current/L2
  currentp3: mbmd/solaredge1-1-1/Current/L3
  timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  name: grid1
- type: template
  template: mqtt-meter
  usage: pv
  power: mbmd/solaredge1-1/Power
  scalepower: 1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
  energy: mbmd/solaredge1-1/Export
  scaleenergy: 1  # Die Energiemenge wird in kWh erwartet. Falls z.b. Wh geliefert wird, kann dies hier mit dem Wert 0.001 angepasst werden
  currentp1: mbmd/solaredge1-1/Current/L1
  currentp2: mbmd/solaredge1-1/Current/L2
  currentp3: mbmd/solaredge1-1/Current/L3
  timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  name: pv1

chargers:
- type: template
  template: pmcc
  ski: bda9-0ea5-fab0-a1a4-211d-b2d3-31b0-eadb-582b-1e83  # Die SKI der Wallbox, ist üblicherweise im Web Interface der Wallbox zu finden
  enforcePVLimits: true  # Verwende PV Laden über eine Ladebegrenzung (wie herkömmliche Wallboxen) bei Fahrzeugen mit Unterstützung von ISO15118
  name: wallbox1

vehicles:
- type: template
  template: porsche
  title: XXX  # Wird in der Benutzeroberfläche angezeigt
  user: XXX  # Benutzername um auf das Gerät zuzugreifen
  password: XXX  # Passwort des Benutzerkontos
  vin: XXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
  capacity: 17.9  # Akku-Kapazität in kWh
  name: ev1

loadpoints:
- title: PMCC
  charger: wallbox1
  vehicles:
  - ev1
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 30m
    min: 0
    target: 100
    estimate: true
  phases: 1
  mincurrent: 2
  maxcurrent: 20
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 5m
  guardduration: 10m

site:
  title: H73
  meters:
    grid: grid1
    pvs:
    - pv1

eebus:
  certificate:

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>

[putty.log](https://github.com/evcc-io/evcc/files/8204402/putty.log)

</pre></details>

andig, 2022-03-08:
> charger out of sync: expected disabled, got enabled
> Mutmaßlich startet deswegen der disable timer nicht.

Das kann gut sein. Irgendwo sind die beiden auseinander gedriftet. Leider haben wir momentan keinen Maintainer für die EEBus Integration- daher bist Du zur Analyse/Behebung auf Dich gestellt, tut mir leid!

MarkusGH, 2022-03-08:
Das issue auf wontfix zu setzen motiviert mich nicht gerade hier tätig zu werden. Analyse kann ich mangels go Erfahrung eventuell leisten, aber ohne Aussicht auf weitere Verarbeitung meines Input lohnt der Aufwand für mich eher nicht. 
Generell würde ich hier das Problem durchaus auch in der übergeordneten Logik sehen, die an der Stelle scheinbar wenig fehlertolerant ist. Denn dass ein Ladevorgang läuft wird ja durchaus erkannt.

MarkusGH, 2022-03-08:
> @MarkusGH Das Problem für `charger out of sync: expected disabled, got enabled`ist hier identisch mit dem Problem im letzten Ticket.
> In EEBUS gibt es nichts das `enable(bool))` (Ladestrom freigeben oder sperren) und `enabled()` 
> (Status der Ladestromfreigabe) abbildet, was evcc aber benötigt. Also müssen die simuliert 
> werden, und das ist nie 100% korrekt. Um das zu lösen müsste evcc auch ohne diese 
> Methoden und deren Implikationen umgehen können, was ein redesign an evcc erfordert.

Ich beginne zu verstehen - auch warum die Reaktion des "ersten Andi" :-) so "abweisend" ist - bei einer 0.85 möchte man nicht alles umwerfen.

Bedeutet das, dass enabled() in der Situation false zurückgibt?

Falls ja müsste das doch leicht zu beheben sein - denn die Meldung "charger out of sync: expected disabled, got enabled" zeigt doch dass der tatsächliche Status korrekt erkannt wird.

DerAndereAndi, 2022-03-08:
Wahrscheinlich müsste man "nur" in der Zeile https://github.com/evcc-io/evcc/blob/master/charger/eebus.go#L214 den Wert `50` auf z.b. `100` hochstellen. Bei dir kommt wird ein ungewöhnlich hoher Ruhestrom von 83W vom Charger geliefert.

MarkusGH, 2022-03-08:
Danke! Habe mir den Code angesehen und beginne zu verstehen. Die Änderung sollte das auftreten des Problems beheben. Habe gerade die Logs angesehen - mein Schätzchen wacht trotz voller Batterie so alle Stunde mal auf und verbraucht dann bis zu 512W.  Hier werden alle Vorurteile gegen PHEV bestätigt...

Ich wage nicht die Änderung einzuchecken weil ich nicht kompilieren kann und weil ich Dir nicht vorgreifen will, aber ich würde das hier vorschlagen:

eebus.go:18:

const idleCurrentFactor = 0.6

eebus.go:215/237:

if data.EVData.Measurements.PowerL1 > data.EVData.LimitsL1.Min * idleCurrentFactor || data.EVData.Measurements.PowerL2 > data.EVData.LimitsL2.Min * idleCurrentFactor || data.EVData.Measurements.PowerL3 > data.EVData.LimitsL3.Min * idleCurrentFactor {

Das wäre sicherlich eine Verbesserung.

Es gibt da aber noch ein Problem, denn eigentlich macht die Logik in loadpoint.go ja das Richtige.

In der Situation wird permanent lp.charger.Enable(false) aufgerufen.

[lp-1  ] WARN 2022/03/08 00:00:45 charger out of sync: expected disabled, got enabled
[eebus ] TRACE 2022/03/08 00:00:45 send: write EV:LoadControl LoadControlLimitListData
[eebus ] TRACE 2022/03/08 00:00:45 send: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":6}]},{"addressDestination":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1,1]},{"feature":1}]},{"msgCounter":849},{"cmdClassifier":"write"},{"ackRequest":true}]},{"payload":[{"cmd":[[{"loadControlLimitListData":[{"loadControlLimitData":[[{"limitId":1},{"isLimitActive":true},{"value":[{"number":0}]}],[{"limitId":2},{"isLimitActive":true},{"value":[{"number":0}]}]]}]}]]}]}]}}]}
[eebus ] TRACE 2022/03/08 00:00:45 recv: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1,1]},{"feature":1}]},{"addressDestination":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":6}]},{"msgCounter":4760},{"msgCounterReference":849},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":1},{"description":"Write failed"}]}]]}]}]}}]}

Aber der PMCC will das irgendwie nicht machen und antwortet mit resultData":[{"errorNumber":1},{"description":"Write failed"}

Danach sieht man dann im Heartbeat resultData":[{"errorNumber":6}:

[eebus ] TRACE 2022/03/08 00:00:45 send: notify EVSE:DeviceDiagnosis DeviceDiagnosisHeartbeatData
Mar  8 00:00:45 homebridge evcc[872]: [eebus ] TRACE 2022/03/08 00:00:45 send: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":5}]},{"addressDestination":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1]},{"feature":5}]},{"msgCounter":850},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"deviceDiagnosisHeartbeatData":[{"heartbeatCounter":471},{"heartbeatTimeout":"PT4S"}]}]]}]}]}}]}
[eebus ] TRACE 2022/03/08 00:00:45 recv: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1]},{"feature":5}]},{"addressDestination":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":5}]},{"msgCounter":4765},{"msgCounterReference":850},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":6}]}]]}]}]}}]}
[eebus ] TRACE 2022/03/08 00:00:45 recv: result CEM:DeviceDiagnosis ResultData
[eebus ] TRACE 2022/03/08 00:00:45 recv: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1,1]},{"feature":3}]},{"addressDestination":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":3}]},{"msgCounter":4767},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"measurementListData":[{"measurementData":[[{"measurementId":1},{"valueType":"value"},{"timestamp":"2022-03-07T23:00:45.799Z"},{"value":[{"number":196000},{"scale":-4}]},{"valueSource":"measuredValue"}],[{"measurementId":4},{"valueType":"value"},{"timestamp":"2022-03-07T23:00:45.799Z"},{"value":[{"number":4319},{"scale":0}]},{"valueSource":"measuredValue"}],[{"measurementId":7},{"valueType":"value"},{"timestamp":"2022-03-07T23:00:45.799Z"},{"value":[{"number":363},{"scale":0}]},{"valueSource":"measuredValue"}]]}]}]]}]}]}}]}
[eebus ] TRACE 2022/03/08 00:00:45 recv: notify CEM:Measurement MeasurementListData

DerAndereAndi, 2022-03-08:
`errorNumber: 6` bedeutet `Command not supported.`, und das auf einen Heartbeat. 🤦 

Und warum er den Loadlimit auf 0 nicht akzeptiert, kann ich leider auch nicht sagen. So ein Verhalten habe ich bei mir noch nie gesehen. Da verschickten Daten sehen korrekt aus. Da kann ich leider auch nicht mehr weiter helfen. Sorry.

MarkusGH, 2022-03-08:
Kein Problem - ich weiß Deine Hilfe sehr zu schätzen.

Was bedeutet errorNumber: 1? Ist das schlicht der Write Error?

Vielleicht überfordere ich den dämlichen PMCC  ja mit cycle interval = 1s.
Hat das cycle interval auch Auswirkung auf die rate mit der die eebus kommunikation erfolgt?

premultiply, 2022-03-08:
Ja. Auf quasi alles was mit Messen, Steuern und Regeln zu tun hat.
Daher würde ich da zunächst mal langsam machen (Default sind 10s) und dann langsam rantasten.
Wenn es zu viel (d.h. zu schnell) ist für die beteiligten Komponenten fängt es an zu schwingen oder macht eben "komische Sachen".

MarkusGH, 2022-03-08:
Nach näherer Betrachtung der Logs würde ich nicht denken dass cycle interval ein Problem ist.

Man sieht, dass gleich der erste Heartbeat nach dem Einstecken mit Fehler zurückkommt:

[eebus ] TRACE 2022/03/07 23:54:29 send: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.2.0"},{"addressSource":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":5}]},{"addressDestination":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1]},{"feature":5}]},{"msgCounter":40},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"deviceDiagnosisHeartbeatData":[{"heartbeatCounter":1},{"heartbeatTimeout":"PT4S"}]}]]}]}]}}]}

[eebus ] TRACE 2022/03/07 23:54:30 recv: #002{"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:19667_PorscheEVSE-00088291"},{"entity":[1]},{"feature":5}]},{"addressDestination":[{"device":"EVCC_HEMS"},{"entity":[1]},{"feature":5}]},{"msgCounter":204},{"msgCounterReference":40},{"cmdClassifier":"result"}]},{"payload":[{"cmd":[[{"resultData":[{"errorNumber":6}]}]]}]}]}}]}

Zu dem Zeitpunkt ist die Kommunikation ohnehin hektisch (ich denke ein handshake findet statt) - unabhängig vom cycle interval.

andig, 2022-03-08:
> Ich beginne zu verstehen - auch warum die Reaktion des "ersten Andi" :-) so "abweisend" ist - bei einer 0.85 möchte man nicht alles umwerfen.

Hallo @MarkusGH das sollte nicht abweisend sein, sondern v.a. Transparent. Wir haben- wie ich geschrieben habe- schlich Niemanden, der sich um EEBus kümmert. 

Da Du nicht kompilieren kannst- ich kann Dir Deinen Fix gerne probehalber einbauen. Welche Plattform?

Wenn aber selbst ein trivialer Heartbeat fehlschlägt fällt mir auch nicht mehr viel ein…

MarkusGH, 2022-03-08:
Alles gut - ich denke wir verstehe uns :-)
Ich würde mich gerne in das Projekt einbringen - mir fehlt schlicht die Zeit.

Ich habe hier ein Ubuntu auf x86.

Den Fix für regelmäßig aufwachende Fahrzeuge mit hohem Verbrauch einzubauen ist sicher kein Fehler, dann kriegt man weniger Fehlermeldungen. Sollte auch unkritisch sein, denn wenn der Charger weniger als den Mindeststrom liefert dann sollte er auch (zumindest logisch betrachtet) aus sein. 

Das eigentliche Problem der fehlerhaften Verbindung löst das jedoch nicht. 

Dieses ließe sich wohl zumindest provisorisch lösen wenn man in dieser Fehlersituation (Heartbeat schlägt mit errorNumber = 6 fehlt) die Verbindung zum Charger trennt und neu aufbaut bzw. aufbauen lässt. Ich denke das passiert bei mir dank WLAN Verbindung ab und zu von selber. Eventuell behebt auch ab- und anstecken des Fahrzeugs das Problem, das habe ich nicht in der Tiefe analysiert.

Ich habe mir gerade eben einen Workaround gebastelt bei dem ein Skript das syslog auf die Fehlermeldung prüft und im Fehlerfall evcc neu startet. Aber das ist doch recht unschön und eine integrierte Lösung die nur die Verbindung zum Charger neu aufbaut wäre sicher besser (und würde anderen helfen die dasselbe Problem haben).

andig, 2022-03-09:
> Den Fix für regelmäßig aufwachende Fahrzeuge mit hohem Verbrauch einzubauen ist sicher kein Fehler, dann kriegt man weniger Fehlermeldungen.

Dein Vorschlag von oben steht unter https://github.com/evcc-io/evcc/pull/2827 zum Test bereit.

> Dieses ließe sich wohl zumindest provisorisch lösen wenn man in dieser Fehlersituation (Heartbeat schlägt mit errorNumber = 6 fehl) die Verbindung zum Charger trennt und neu aufbaut bzw. aufbauen lässt. 

Ich hab grad kurz rein geschaut. Absichtliches Trennen fehlt uns in den Implementierung :(

MarkusGH, 2022-03-09:
> Dein Vorschlag von oben steht unter #2827 zum Test bereit.

Schick - danke!

> Ich hab grad kurz rein geschaut. Absichtliches Trennen fehlt uns in den Implementierung :(

:-((

Kann man die Verbindung nicht einfach sabotieren und den PMCC die Arbeit erledigen lassen?

Der eebus braucht doch einen Heartbeat...
sleep(2000)?

Oder beenden des Prozesses und systemd das Weitere erledigen lassen?
Wenn man an der Stelle landet ist sowieso alles zu spät...

MarkusGH, 2022-03-09:
Dein Projekt, Deine Entscheidung :-)
Wie gesagt, ich hab' mir einen dreckigen Workaround gebastelt, mir ist sowieso erstmal geholfen.
Aber andere Anwender mit einem PMCC werden möglicherweise in dasselbe Problem laufen und sich nicht zu helfen wissen.
Denen würde ggf. selbst ein Beenden des Prozesses helfen.
Ich berichte ob meine Workaround funktioniert, dann kannst Du ja endgültig entscheiden.

MarkusGH, 2022-03-15:
Es scheint als ob ein Neustart des Service Kommunikationsprobleme mit dem PMCC zuverlässig behebt.
Insofern wäre es zu überlegen (als Workaround für nicht behebbare eebus Probleme ) evcc bei schwerwiegenden Fehlern in dem Bereich zu beenden und das weitere unter Unix dem systemd zu überlassen.
Gerne kann ich auch mein Monitor Skript posten, ich weiß allerdings nicht so recht wo und wie.

andig, 2022-03-15:
Ja, das ist im Prinzip äquivalent zum Neuaufbau der Verbindung. Ich verstehe immer noch nicht, warum der PMCC beim Heartbeat rumzickt- das ist ja erstmal unabhängig von der physikalischen Verbindung sondern ein Logikproblem.

MarkusGH, 2022-03-15:
Softwarefehler im PMCC...
Ich denke das Porsche HEMS wird die Ladestrombegrenzung anders machen als evcc (in der Web Anwendung des PMCC wird ja auch der Solarstromanteil angezeigt, der bei Steuerung über evcc immer 0 ist). Wenn das Problem dort nicht auftritt bleibt der Fehler ewig drinnen.  Im Zusammenspiel mit evcc gibt es ja noch mehr Probleme: Wenn evcc den Ladestrom auf 0 begrenzt meldet das Auto beim Anstecken einen Ladefehler, häufige Meldung "climater status UNKNOWN", ...

DerAndereAndi, 2022-03-15:
@MarkusGH schick mir mal bitte ein Trace log (eebus Daten sind wichtig) wo man den Start und das anstecken mit dem Fehler sieht per Email. Die findest in der commit history. Mail at mein Name Punkt de

komisch dass auch bei der IEC Kommunikation dieser Ladefehler auftritt. Du hast einen anderen OBC als ich im Taycan. Aber in letzter Zeit habe ich beim ID3 auch so Meldungen.

Der PMCC ist eigentlich für dein PHEV nicht passend, die guten Features die den PMCC ausmachen kannst nie nutzen. Und nur fürs mitnehmen ist es zu schade. Da würde eine einfache steuerbare Wallbox besser passen.

andig, 2022-03-21:
Ich mache nach dem Fix hier mal zu. Alles Weitere gerne als Discussion.

closed 2022-03-21 by andig