evcc-io/evcc #2815: EVCC berechnet SitePower nicht korrekt (nur PV vorhanden) -> Regelung der Ladesäule mit falschen Werten

bluegaspode, 2022-03-08:
**Describe the bug**
Ich habe EVCC (nur) mit einer PV konfiguriert. 
EVCC berechnet die sitePower basierend auf PV Generierung. Bei der Berechnung werden veraltete Wert der ChargePower abgezogen, dies führt dann zum ungewollten hoch/abregeln der Ladesteuerung.

Beispiel: 
Zyklus1: 
- PV-Ertrag 3kW; 
- chargePower unbekannt = 0A => 0kW
- sitePower = 3kW-0kW =  3kW = 13A verfügbare Leistung
- (Bestimmung der realen ChargePower ... 6A = 1,38kW)
- hochregeln auf 6A+13A = 20A  (heruntergeregelt auf 16)

Zyklus 2:
- PV-Ertrag konstant 3kW
- sitePower wird mit altem chargePower-Wert aus Zyklus1 berechnet
- sitePower = 3kW-1.38kW = 1,62kW = 7A zusätzliche verfügbare Leistung
- (Bestimmung der realen ChargePower ... 16A = 3,68kW)
- hochregeln auf 16A+7A = 24A (heruntergeregelt auf 16A)

Zyklus 3: 
- PV-Ertrag konstant 3kW
- sitePower wird mit altem chargePower-Wert aus Zyklus2 berechnet
- sitePower = 3kW-3,68kW = -0,68Kw = -2,9A zu viel Leistung
- (Bestimmung der realen ChargePower ... 16A = 3,68 kW)
- Ladesäule regelt etwas runter 

Im hier gezeigten Ablauf wird die Ladesäule in Zyklus 1+2 auf 16A gestellt, obwohl das bei einer PV-Leistung von 3kW viel zu hoch ist. Im Zyklus 3 wird korrekt heruntergeregelt, im kommenden Zyklus 4 wird weiter heruntergeregelt, weil evcc weiterhin mit falschen Werten rechnet.

Weiter unten sind vollständige Logs von 3 Zyklen (habe Kommentare ergänzt) 

**To Reproduce**
- volle Logs aktivieren
- Lademodus nur mit PV Messung, nicht mit Grid-Messung
- idealerweise halbswegs konstante PV-Erträge, dann ist die Rechnung leichter nachzuvollziehen.

**Expected behavior**
- PV-only Modus, muss die ChargePower ermittelt werden, bevor die sitePower berechnet wird. Sonst sind die Rechnungen falsche.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.85

```

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 30s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
# sponsortoken:

# log settings
log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
- name: pv
  type: modbus
  model: sunspec
  uri: 192.168.1.108:1502
  id: 71

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/evcc-io/config#chargers
chargers:
- name: go-e
  type: go-e
  uri: http://192.168.1.109 

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/evcc-io/config#vehicles
vehicles:
- name: kia
  type: kia
  title: e-Niro
  capacity: 64
  user: # user
  password: # password

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
#    grid: grid # grid meter
    pvs: pv
  #  battery: battery # battery meter
  # prioritySoC: 60 # give home battery priority up to this soc (0 to disable)
  residualPower: 0

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Carport # display name for UI
  charger: go-e # charger
#  meters:
#    charge: charge # charge meter
#  vehicle: kia
  # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  # - ID.3
  # - e-Up
  mode: minpv
  soc:
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: false # set true to interpolate between api updates
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  onIdentify: # set defaults when vehicle is identified
  - e-Up:
      mode: pv # switch back to pv mode
      targetSoC: 100 # charge to 100%
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

# tariffs are the fixed or variable tariffs
# cheap can be used to define a tariff rate considered cheap enough for charging
tariffs:
  grid:
    # # either
    # type: tibber
    # cheap: 20 # ct/kWh
    # token: "476c477d8a039529478ebd690d35ddd80e3308ffc49b59c65b142321aee963a4" # access token
    # homeid: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c" # optional if multiple homes associated to account

    # # or
    # type: awattar
    # cheap: 20 # ct/kWh
    # region: de # optional, choose at for Austria

# mqtt message broker
mqtt:
  # broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # public: # public key
  # private: # private key

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
[site  ] DEBUG 2022/03/08 12:08:35 ----
[site  ] DEBUG 2022/03/08 12:08:35 pv power: 3123W
[site  ] DEBUG 2022/03/08 12:08:35 site power: -3123W   // Ermittlung chargePower hat noch nicht stattgefunden
[lp-1  ] DEBUG 2022/03/08 12:08:35 charge power: 950W
[lp-1  ] DEBUG 2022/03/08 12:08:35 charge currents: [5.1 0 0]A
[lp-1  ] DEBUG 2022/03/08 12:08:35 charger status: C
[lp-1  ] INFO 2022/03/08 12:08:35 start charging ->
[lp-1  ] DEBUG 2022/03/08 12:08:36 pv charge current: 19.6A = 6A + 13.6A (-3123W @ 1p)   // rechnet mit falscher sitePower
[lp-1  ] DEBUG 2022/03/08 12:08:36 max charge current: 16A
[site  ] DEBUG 2022/03/08 12:09:05 ----
[site  ] DEBUG 2022/03/08 12:09:05 pv power: 3151W
[site  ] DEBUG 2022/03/08 12:09:05 site power: -2201W         // rechnet mit 950W ChargePower aus letztem Zyklus.
[lp-1  ] DEBUG 2022/03/08 12:09:05 charge power: 3580W    // dabei lädt die Box längst mit 3580W
[lp-1  ] DEBUG 2022/03/08 12:09:05 charge currents: [15.2 0 0]A
[lp-1  ] DEBUG 2022/03/08 12:09:05 detected phases: 1p
[lp-1  ] DEBUG 2022/03/08 12:09:05 charger status: C
[lp-1  ] DEBUG 2022/03/08 12:09:05 pv charge current: 25.6A = 16A + 9.57A (-2201W @ 1p)  // will erneut hochregeln
[site  ] DEBUG 2022/03/08 12:09:35 ----
[site  ] DEBUG 2022/03/08 12:09:35 pv power: 3153W
[site  ] DEBUG 2022/03/08 12:09:35 site power: 427W            // rechnet mit 3580 aus letztem Zyklus
[lp-1  ] DEBUG 2022/03/08 12:09:35 charge power: 3570W    // Box ist weiterhin bei +- 3570W / 16A
[lp-1  ] DEBUG 2022/03/08 12:09:35 charge currents: [15.1 0 0]A
[lp-1  ] DEBUG 2022/03/08 12:09:35 detected phases: 1p
[lp-1  ] DEBUG 2022/03/08 12:09:35 charger status: C
[lp-1  ] DEBUG 2022/03/08 12:09:35 pv charge current: 14.1A = 16A + -1.86A (427W @ 1p)   // regelt etwas runter
[lp-1  ] DEBUG 2022/03/08 12:09:35 max charge current: 14A

</pre></details>

andig, 2022-03-08:
> Bei der Berechnung werden veraltete Wert der ChargePower abgezogen, dies führt dann zum ungewollten hoch/abregeln der Ladesteuerung.

Die Frage ist immer wann die ChargePower "korrekt" eingeschwungen ist. Eine Möglichkeit wäre, diese vor jedem Regelintervall neu über alle Ladepunkte auszulesen.

/cc @premultiply

bluegaspode, 2022-03-08:
Ich glaube es hat hier nichts mit Einschwingen von Werten zu tun. 
Sondern mit der korrekten Reihenfolge des Auslesens und Einbeziehen in die Berechnung. 

Siehe Log: das Update-Interval steht auf 30sek für die Zyklen.
EVCC nimmt bei der Berechnung des neuen Wert einen Wert, der (im Beispiel) >30sek alt ist und in einem Zustand gemessen wurde, der VOR der Änderung der Ampere Zahl stattfand. Mithin ist das in jedem Fall ein veralteter und falscher Wert.

bluegaspode, 2022-03-08:
OK - der verlinkte Change hat versucht nach dem Ändern eines Wertes nochmal aktuelle Werte mit etwas Verzögerung zu lesen.

Von der Logik wäre es aber sicherlich besser, VOR Berechnung der sitePower, den aktuellen Wert zu ermitteln. Zumindest im PV-Only Modus, braucht es aktuelle Werte der PV und aktuelle Werte der Wallbox.

VolkerK62, 2022-03-08:
@andig   ist das mit der 0.85 reingekommen?   Wenn ich mir mein Log aus #2784  anschaue, da finde ich diesen Fehler nicht und ich bin mir fast sicher, dass ich da noch die 0.84 drauf hatte.

andig, 2022-03-08:
Müsste ich jetzt schauen… ich würde versuchen, das mal für @bluegaspode zu verbessern und eher den Weg vorwärts einschlagen.

andig, 2022-03-09:
Könnt ihr das mal ausprobieren?

closed 2022-03-09 by andig

bluegaspode, 2022-03-09:
vielen Dank für den Fix. Ich warte auf den Nightly Build und den nächsten Sonnentag und probiere es also idealerweise gleich morgen mal aus!

RTTTC, 2022-03-13:
@bluegaspode So - how does it work ?

bluegaspode, 2022-03-13:
sorry that I did not report back. Nightly is running since 4 days and everything ist fine and smooth.
Given the current good weather there Was also enough situations to test.

I don't have a build environment so didn't test the additional branch