evcc-io/evcc #2834: Fronius: add non-float Template

andig, 2022-03-09:
We could offer a float and non-float config both? I guess most users won‘t have float enabled if thats an additional step?

_Originally posted by @andig in https://github.com/evcc-io/config/issues/96#issuecomment-1062246190_

Or add a float option to the current template.

premultiply, 2022-03-09:
To avoid handling int+sf

andig, 2022-03-09:
> To avoid handling int+sf

Das tut der Sunspec Code ohnehin. Kommt dann vmtl noch aus der Zeit wo die Register von Hand ausgelesen wurden. Im Prinzip könnten wir dann auch komplett auf die Int Register schwenken, oder werden die im float Modus abgeklemmt?

premultiply, 2022-03-09:
Genau. Scheint geschwenkt zu werden. Entweder oder.

closed 2022-03-17 by andig