evcc-io/evcc #2838: Alfen: check C1 state

andig, 2022-03-09:


premultiply, 2022-03-09:
Ich hab da heute schon geschaut.
Du hattest das nachträglich eingefügt und ich konnte aber keinen Hinweis oder Grund dazu finden.
Kannst du dich daran erinnern warum das dort nachträglich so rein kam?

andig, 2022-03-09:
Oh, danke! Das war anscheinend https://github.com/evcc-io/evcc/commit/73e8cb4966802dff3d942d5feffb5179b4f9418c, Grund unklar. Lassen wie es ist da sich bisher niemand über Fehler beschwert hat?

andig, 2022-03-09:
Aha:

[image]

Und nu? Das C1 hat eine andere Bedeutung als bei der Amtron?

premultiply, 2022-03-09:
Hab gerade nochmal nachgeschaut in der Norm. C2 ist der stationäre Ladezustand, C1 ist der Übergangszustand bei dem die EVSE dem Fahrzeug ihrerseits einen Ladeabbruchswunsch (Ladeunterbrechung) signalisiert. Also das was passiert wenn man im Ladezustand enabled auf false setzt.

Das Fahrzeug sollte dann in kurzer Zeit die Energieaufnahme einstellen und mit dem Übergang 6V zu 9V öffnet dann die EVSE das Schütz (stromlos) und der Zustand B1 ist wieder erreicht.

C1 zählt daher auch definitiv noch zum Zustand C, wird aber hier kaum auffallen bzw. aufgefallen sein, da dieser nur ganz kurze Zeit durchlaufen wird bevor wieder B1 anliegt.

So ähnlich ist das ja auch mit Status B2 - wenn das Fahrzeug nicht gerade schläft☝️

premultiply, 2022-03-09:
Ich baue es daher morgen wieder auf das direkte Mapping hier um.

PR #2843 (2022-03-10): Alfen: Fix status mapping
Closes #2838
closed 2022-03-10 by andig